genodelabs/genode #4603: server/vfs: status() calls (potentially costly) num_dirent()

ttcoder, 2022-09-07:
The RPC interface implemented by the vfs server does not have a "num_dirent()", so instead it is status() that plays that role, i.e. instead of having a **num_dirent -> num_dirent -> num_dirent** chain, we have a **num_dirent -> status -> num_dirent** chain:

  https://github.com/genodelabs/genode/blob/c333fc6f0479862eb82a59dce2c34bb200d393d1/repos/os/src/server/vfs/main.cc#L663

That is done for the sole benefit of this client-side user (the others do not need the dirent count calculation from status):

  https://github.com/genodelabs/genode/blob/649647538bfa0dcc2763246df35dafabbc12ba30/repos/os/src/lib/vfs/fs_file_system.h#L761

In some file systems (FUSE, and I think Rump and lx_fs) num_dirent() is a full opendir/readdir loop, which is expensive.

One 'solution' is to use the target file system locally (as a direct vfs plug-in) instead of indirecting through a remote file system and vfs server, then we get spared the above, stat() is dedicated exclusively to stat()ing and does not get slowed down to o(n) by calling opendir/readdir. But in any realistic scenario, the target fs is to be shared between multiple applications, which requires a remote vfs server, so it would be nice if both scenarios (local and remote) were in o(1) instead of o(n).

### Optim ideas

My first approach was to modify the client-side num_dirent() to no longer rely on status(), but create a Dir_vfs_handle and iterate directly on that, but that seems sub-par and I didn't get far that way anyway.

I had success only by adding a parameter to the RPC interface, as in "status( path, bool need_subdir_nument )". It might be cleaner to use a two-state enum. Anyhow, client-side num_dirent() sets the boolean to "true" to signal that it's interested in the dirent count, other users set it to "false" to get spared the expensive operation.

### Context

I'm porting Tracker to Genode. As any desktop, when it displays a folder to the user it stat()s all the entries within it so as to display their size and mdate. Which kind of means the o(n) folder readig turns into o(n²), or at least o(n*m). Plus, Tracker uses filesystem functions like there's no tomorrow   (so it's a "hammer and anvil" situation -- can't modify the hammer much, so I'm working on the anvil :-). What's more, my FUSE back-end seems to bottleneck in block_file. I've discovered and been using GENODE_TSC macros (nice!) to find out who the biggest "offenders" are and optimize them. I started using a block_cache component, added caching in FUSE, now it no longer takes minutes (!) but it still takes seconds to dozens of seconds to browse to a new directory (with e.g. class BFilePanel) so I need to extend the scope of my optimizations to bring the time down to a bearable level. With the patch (which I'll attach shortly) the situation is improved. Though just because that patch works locally for me does not necessarily mean it's the right thing to upstream to Genode, it's just a suggestion. Discuss ! :-)

ttcoder, 2022-09-07:
[vfs_nument_stat.patch.txt](https://github.com/genodelabs/genode/files/9503860/vfs_nument_stat.patch.txt)

One possible way to solve the problem, and only a partial patch (cached_fs_rom etc are not covered by the patch)

nfeske, 2022-09-12:
Thank you for sharing your findings.

I found the following very surprising:
> In some file systems (FUSE, and I think Rump and lx_fs) num_dirent() is a full opendir/readdir loop, which is expensive.

I'd expect a single call of `status` to be cheap. A whole loop over directory entries for a stat'ed directory seems super wasteful. Of course, I'm immediately wondering if the mentioned file systems offer a cheaper way to get hold of the number of directory entries? If it turns out that the file systems offer no cheap way to request the number of directory entries, the merge of `num_dirents` and `status` definitely stands in the way.

Thank you very much for pointing out this issue. Let me have a closer look.

nfeske, 2022-09-12:
@ttcoder could you give https://github.com/nfeske/genode/commit/645b741f25f38e7d85be0b6d5b9e093ad6f1e3f6 a try? The commit adds a dedicated RPC call `num_entries`. It sets the `Status::size` field to 0 when called for a directory.

ttcoder, 2022-09-12:
Looks good, and more elegant, with your patch there is no need to change code relying on status() all over the repo.

I've done before-and-after testing, I'm still in o(1) with this patch applied and mine removed. Also tested against base code : in my reference test (which is complicated by caching I've added and complex client code) I observe it takes almost 50% less time to read a 49 entries folder. Well done @nfeske, appreciated :-)

- ttcoder

P.S. looking around, I spotted this in lx_fs/directory.h (but maybe it's actualy required to be non-zero there):
				.size  = _num_entries() * sizeof(File_system::Directory_entry),

nfeske, 2022-09-13:
Thank you for the super quick response, and for pointing out the missing adaptation of lx_fs. The latter should now be covered by https://github.com/genodelabs/genode/commit/395f1a9c535ca7384b456263d09d2cf729555b2f.

commit 9ef24a92 (2022-09-13): os: add File_system_session::num_entries RPC

This patch splits the querying of the number of directory entries from
the directory's 'status' information. Subsuming the number of directory
entries as part of the status makes 'stat' calls too costly for some
file systems that need to read a directory for determining the number of
entries. So when stat'ing the entries of one directory that contains sub
directories, all entries of each sub directory are visited.

Thanks to Cedric Degea for pointing out this performance bottleneck!

With this change, the 'status' function returns a 'Status::size'  [... cut]
closed 2022-09-21 by chelmuth