genodelabs/genode #4606: Allow passing a session label to `Irq_connection`

jschlatow, 2022-09-09:
I just took my newly implemented pin driver for the Zynq SoC out for a test drive on the Zybo-Z7 board. The board has quite a few buttons, switches and leds. Two of the buttons and one led can be controlled without further ado via the GPIO controller. I therefore wrote a test component that switches the led on when one button has been pressed and switches it off when the other button has been pressed. In doing so, I noticed that there is currently no possiblity to open multiple (different) Irq connections to the pin driver from a single component. The reason for this is that the latter applies label-based policy matching, yet the `Irq_connection` constructor does not allow passing a session label.

nfeske, 2022-09-09:
That's a change I was planning as well for the very same reasons as yours, but somehow I got carried away with other topics. :-) Thank @jschlatow for bringing up this change!

nfeske, 2022-09-15:
Merged as https://github.com/genodelabs/genode/commit/f7b09d11e39691e36d8b00e6ef22161178956273 with a minor white-space tweak applied.

nfeske, 2022-09-21:
Merged to master.

closed 2022-09-21 by nfeske