genodelabs/genode #4610: Dictionary data structure

nfeske, 2022-09-14:
There are several places in the code base where objects are accessed by using a name as key. To avoid the repeated manual crafting of such data structures, I propose to add a basic `Dictionary` data structure to `os/include/util/`.

The data structure should follow the basic design of the `Id_space` and `Registry`. That is, elements are automatically added to the dictionary at construction time, respectively removed at destruction time. There exists an `apply` interface for applying a functor to one element by specifying a name as key, and an `apply_any` interface that can be used to destruct all dictionary items.

chelmuth, 2022-09-15:
https://github.com/genodelabs/genode/blob/master/repos/gems/src/app/trace_recorder/named_registry.h has some subtle differences to the dictionary. @jschlatow could *app/trace_recorder* still be adopted to use the new data structure or does it depend on `for_each()`? I had a look into it because current staging yields a build error in this component.

```
named_registry.h:53:53: error: ambiguous overload for ‘operator>’ (operand types are ‘const Name’ {aka ‘const Genode::String<64>’} and ‘const Name’ {aka ‘const Genode::String<64>’})
```

m-stein, 2022-09-15:
I think in general it's a good idea to have such a structure standardized. Two core functionalities of the NIC router depend on dictionaries.

One feature that would be convenient for the router is that ``apply(...)`` would take two functors, one that is called on finding a matching item and one that is called when no matching item was found (see repos/os/src/server/nic_router/avl_string_tree.h:89). A typical use case would be repos/os/src/server/nic_router/interface.cc:2249 where an interface looks up its new domain during reconfiguration. If there is no new domain (no matching policy or invalid domain name), the interface must react differently. Clearly, in most cases, this behavior can be emulated by introducing a variable that is set true only on a match and checking after the call to ``apply(...)`` but I would prefer forcing the user to consider the case that no match has been found in order to prevent unintentionally ignored errors (same principle as with the ``Attempt`` utility).

Another feature used in the NIC router is a ``for_each(...)`` method template (repos/os/src/server/nic_router/avl_string_tree.h:96). This is used for instance at repos/os/src/server/nic_router/interface.cc:1646 where all interfaces of all domains must be checked for wakeup. I cannot see an easy way for emulating this with the current ``apply``/``apply_any`` method templates.

nfeske, 2022-09-15:
>  I had a look into it because current staging yields a build error in this component.

I fixed this with fixup dfa1eb6ff8783827cdfd42474dc3aebd62487554.

nfeske, 2022-09-15:
@m-stein Thank you for chiming in. I like both of your suggestions.

The two-arguments `apply` is perfectly in line with the spirit of the recent discussion with @jschlatow (https://github.com/genodelabs/genode/issues/4600#issuecomment-1236877442) about `Xml_node::with_sub_node`.

I wonder, should both variants be named differently, expressing the risk that the body might not be called in the single-argument version? If so, what would be a good name for the single-argument apply?

Adding `for_each` is a sensible idea. This would in fact allow the use of the `Dictionary` in core's data structure holding the ROM modules. With this in the back of mind, we might better move it from os/include/util to base/include/util.

jschlatow, 2022-09-15:
@chelmuth I actually don't remember why I added the `Named_registry::for_each(...)` to _app/trace_recorder_. The current implementation does not use it any more. Please consider commit 1a3ff71, which adopts the new `Dictionary` and replaces the former `Named_registry`.

nfeske, 2022-09-15:
Regarding `for_each`, is there a use case for a non-const function or is `for_each` merely used for read access? I'm asking because a non-const `for_each` that allows the modification of the dictionary from within the functor would be much more complicated. I'd prefer to avoid this complexity.

nfeske, 2022-09-15:
> Please consider commit https://github.com/genodelabs/genode/commit/1a3ff714f7e096c8bff7f071a6d4d5498d978534, which adopts the new Dictionary and replaces the former Named_registry.

YES! :-)

chelmuth, 2022-09-15:
> Regarding `for_each`, is there a use case for a non-const function or is `for_each` merely used for read access? I'm asking because a non-const `for_each` that allows the modification of the dictionary from within the functor would be much more complicated. I'd prefer to avoid this complexity.

`for_each()` is effectively const in both cases - in nic_router explicitly implemented that way and in the (now removed) trace_recorder named_registry due to use of `Avl_tree::for_each()`, which is const.

nfeske, 2022-09-15:
Commit https://github.com/genodelabs/genode/commit/5394df5fef7a846fece4b6ea2daabbf0d929c5c0 adds `for_each` as a const method.

Commit https://github.com/genodelabs/genode/commit/8866bb6898255e72070f081184d9a4d87e9454a2 changes `apply` to two functor arguments `match_fn` and `no_match_fn`.

nfeske, 2022-09-15:
Commit https://github.com/genodelabs/genode/commit/5a2aca30ab3055ce1cd6236c4c2c779a3d5bccc4 adds support for returning values from the apply functors.

Commit https://github.com/genodelabs/genode/commit/a00453c7e2747c3b88519392eaef4c5391284b23 adjusts the trace recorder to the use of return values, allowing for improved const hygiene.

m-stein, 2022-09-15:
@nfeske Thanks for applying my suggestions! I must admit that I fail to understand what exactly is going on in the most recent versions of ``apply`` but it sounds exactly like what I wanted :)

nfeske, 2022-09-15:
@m-stein yeah, the code looks cryptic. But this is to the benefit of all users who can use this version with or without a return value. 

In many cases (like in https://github.com/genodelabs/genode/commit/a00453c7e2747c3b88519392eaef4c5391284b23), the support of a return value is handy because it avoids the need for non-const variables to funnel results out of the lamda. But we also want to support the void case.

The trickery in the const version avoids duplicating the code of the non-const version.

m-stein, 2022-09-15:
@nfeske I just stumbled over another thing while adapting the NIC router to use Dictionary: Whenever inserting a NIC client or domain into the tree (repos/os/src/server/nic_router/configuration.cc:210 and 137), the router checks whether the name is unique and, if not, calls a functor instead of inserting the element (repos/os/src/server/nic_router/avl_string_tree.h:125). It's possible to emulate this with ``apply``, constructing in the ``no_match`` functor and handling the name-not-unique-case in the ``match`` functor. 

The current Dictionary implementation allows for the insertion of multiple elements with the same name. On calling ``apply`` with that name, the match functor is called only on the element that was inserted first. My expectation as a user would be that either the Dictionary does not allow for multiple elements with the same name and calling an ``name_not_unique_fn`` in case or call ``match_fn`` on each element with the name. I'd prefer the first option as it simplifies the router code :)

m-stein, 2022-09-15:
@nfeske Sorry, I had to re-write the last sentence in my last post. Please check on github.

nfeske, 2022-09-16:
> My expectation as a user would be that either the Dictionary does not allow for multiple elements with the same name and calls a name_not_unique_fn

The insertion into the dictionary happens at the construction time of a dictionary element. Therefore, the C++ natural way to reflect a name collision would be to throw an exception from the constructor. I'd like to avoid that. However, passing an `name_not_unique_fn` as constructor argument would still leave an ill-constructed object behind. Ultimately, the dictionary cannot enforce the uniqueness of names, only the creator of the elements can.

I'm not sold on adding an error-handling functor argument to the constructor because I cannot think of reasonable way to use this hook instead of merely providing an empty lambda (just to comply with the bureaucracy) or printing an error message.

How about drawing the line as follows?

* The avoidance of name collisions is the responsibility of the caller.
* Whenever an element is inserted, the dictionary checks for a name collision and prints a diagnostic message if there is a conflict. It nevertheless inserts the item into the dictionary. This has the effect that items may become unreachable via `apply`. But all items can still be processed via `apply_any`, which allows for a safe destruction of the dictionary. So the program does not enter an unsafe state but pinpoints to the application's bug at runtime.
* A new method `Dictionary::exists(Name const &) const` gives the caller a convenient way to check for name collisions before creating a dictionary element.

When following this approach, each insertion into the dictionary involves two lookups - one by the caller of the dictionary-element constructor and one by the constructor (for the diagnostic message). But with the expectation that insertions are rare, compared to `apply`, I think that would be fine, We would also avoid introducing an exception for the constructor. 

If we decide to go this way, I'd go forward with adding the `exists` method and diagnostic error message.

m-stein, 2022-09-16:
@nfeske You're right, I didn't think about the problems with the functor.

Alternatively, I thought about making the Element constructor private and introducing a constructing method in Dictionary instead (comparable to Constructible::construct). But it would pull the choice of allocation method into Dictionary which doesn't seem desirable.

So, although I'm not happy with leaving uniqueness unenforced, I agree that your proposal seems to be the best solution.

nfeske, 2022-09-16:
@chelmuth do you think the name `exists` is fitting? Or would `contains` be more suitable?

```
if (dictionary.exists(name)) {
   ....
```
vs.
```
if (dictionary.contains(name)) {
   ....
```
or something else?

m-stein, 2022-09-16:
I forgot to mention, that in the NIC router, such an ``exists`` method would not be of help as the router ensures that _both_ objects that have the clashing name get ignored. I.e., the object that was already in the dictionary must be destroyed again on finding the clash. Therefore the use of ``apply`` fits better.

nfeske, 2022-09-16:
Commit https://github.com/genodelabs/genode/commit/986575b6acd2e97cabec217b2fae981dc11db1f5 adds the check and the `exists` method.

> in the NIC router, such an exists method would not be of help

Thanks for clarifying. However, the dictionary needs the `exists` method internally anyway now that the element constructor performs the check. I think its fine to publicly expose it..

chelmuth, 2022-09-16:
Regarding the naming, `contains` sounds slightly better in my ears but I'm also fine with `exists`.

nfeske, 2022-09-16:
@chelmuth thanks for commenting. I was leaning towards `contains` as well. But in my concrete case of the directory cache in `depot_query`, dictionaries are named `files` (as in list of files) or `listings` (list of listings). In these cases, `files.exists(name)` sounds less forced than `files.contains(name)` but that comes very much down to personal taste. However, should the naming spill to similar data structures (`Registry`, `Id_space`) in the future, `exists` is probably more widely applicable. (I'm looking at the nicely consistent naming in Tcl's `file exists`, `info exists` functions) So in the meanwhile, I gravitate towards `exists`.

m-stein, 2022-09-17:
While we're at naming, I must admit that I find the names ``apply`` and ``apply_any`` not very satisfying. A name like ``apply_to_any`` would make more sense IMO than ``apply_any``. And, taking into account the naming of the two functors ``match_fn`` and ``no_match_fn``, I think that something like ``find``, ``find_by_name``, ``find_match``, ``search_for``, ``search_for_name``, ..., or at least ``apply_to`` would be more descriptive. However, that's just some thoughts and I can live with the current naming as well.

m-stein, 2022-09-17:
I've adapted the NIC router to use Dictionary:
fdd133bd21 nic_router: use the dictionary data structure
43d91defe0 dictionary: avoid overlaying of member 'name'
9dbad4bdc2 dictionary: remove unused template parameter

@nfeske Are the dictionary commits fine with you?

m-stein, 2022-09-17:
Btw. I had to create a non-const for_each version again as the router requires this in quite some places (only for modifications to the managed objects themselves, no destroying). What's the native idea for such a requirement?

m-stein, 2022-09-19:
@nfeske I've updated my commits according to our offline discussion:
b6e4de8766 nic_router: use the dictionary data structure
ba880362a5 fixup "os: add util/dictionary.h utility" (no member overlaying)
69cc4c8cc3 fixup "os: add util/dictionary.h utility" (ARG template param)

m-stein, 2022-09-19:
@nfeske This fix-up is a follow-up of the ``ARG`` fix-up:
857576ebc1 fixup "os: add util/dictionary.h utility" (ARG template param: depot_query)

Furthermore, I found the solution for the problem with the ``ARG`` template parameter in the router. I stumbled over repos/os/include/block_session/connection.h:347, and, indeed, writing
``this->template apply_any<OBJECT_T>(`` in the router fixes the compile error. Anyway, I would stay with the original fix-up as I find ``this->apply_any(`` less ugly. Note that the ``this->`` (or, alternatively ``Genode::Dictionary<OBJECT_T, NAME_T>::``) is required in both cases.

nfeske, 2022-09-19:
> I must admit that I find the names apply and apply_any not very satisfying.

After sleeping over it, I agree. The name `apply` comes from a time when the use of lambda arguments was still novel. With now that the pattern has become routine, we don't really need to highlight that a function is applied. I'm going to rename the methods to `with_element` and `with_any_element`, which are in line with methods like `Xml_node::with_sub_node` or `Attempt::with_result`.

nfeske, 2022-09-19:
I just force-pushed the staging branch with all changes consolidated into https://github.com/genodelabs/genode/commit/3be0136901069a50cf12d21bea49950f5a43dec6

m-stein, 2022-09-19:
I like the new naming! A good name for ``for_each`` would now be ``with_each_element`` but I don't want to be picky :)

m-stein, 2022-09-22:
I've had to add a fix-up:
f92ed1356d fixup "nic_router: use the dictionary data structure" (new method naming)

m-stein, 2022-09-22:
@chelmuth: I just found out about f384bd05bb. Thanks for your work! But, please let me suggest a different approach for such situations: When there is a re-base problem on merging a commit of mine, I'd prefer that, in general, my commit is ignored until I create a fix-up. What do you think?

m-stein, 2022-09-22:
Here's the re-based version:
ab3acef245 fixup "nic_router: use the dictionary data structure" (lisp antidote)

chelmuth, 2022-09-22:
> @chelmuth: I just found out about [f384bd0](https://github.com/genodelabs/genode/commit/f384bd05bb0bd477d2cbe8963b5ea163e020ea61). Thanks for your work! But, please let me suggest a different approach for such situations: When there is a re-base problem on merging a commit of mine, I'd prefer that, in general, my commit is ignored until I create a fix-up. What do you think?

In general, I support this procedure, but in this case it was the rebase on your side that failed and I wanted an immediate fixup of the merged commit.

m-stein, 2022-09-22:
@chelmuth Sorry, I have the feeling that I don't understand yet. Was it critical for you to merge 5e983b29d09be635277bb0b73f8a6348e1f7cd1f?

chelmuth, 2022-09-22:
> @chelmuth Sorry, I have the feeling that I don't understand yet. Was it critical for you to merge [5e983b2](https://github.com/genodelabs/genode/commit/5e983b29d09be635277bb0b73f8a6348e1f7cd1f)?

No, it was not critical to merge the commit. I just did yesterday in good faith (and without conflicts), but it became apparent that it did not work in our nightly CI run and, thus, I fixed it this morning.

nfeske, 2023-01-11:
I just noticed that the `Dictionary::for_each` visits the entries in reverse alphabetic order. Even though we left the ordering of dictionary entries unspecified, it would be sensible to reverse that order I think.

chelmuth, 2023-01-26:
Can this issue be closed?

nfeske, 2023-01-26:
Sure.

commit 7d77c5f3 (2022-09-14): os: add util/dictionary.h utility

The new 'Dictionary' provides an easy way to access objects using
strings as key. The 'String' received the 'operator >' to simplify the
organization of strings in an AVL tree.

The patch removes the former definition of the 'operator >' from the
platform driver because it would be ambigious now.

Fixes #4610
closed 2023-01-26 by nfeske