genodelabs/genode #4612: nic_router: bind link state to remote DNS config

m-stein, 2022-09-14:
The NIC router README claims that the 'dns_config_from' attribute in a DHCP
    server configuration binds the propagated link state of all interfaces at the
    domain of the server to the validity of the IP config of the domain that is
    given through 'dns_config_from'.
    
    However, this was not true. The router missed to implement this detail which
    led to clients of such a DHCP server sending DHCP DISCOVER packets too early.
    These early DHCP DISCOVER packets were dropped by the router potentially
    causing a big delay until the client started a new attempt. Unnecessary long
    network boot-up delays were observed with at least the lwip run script and
    Sculpt on the PinePhone and could be tracked down to this former
    inconsistency in the router.

m-stein, 2022-09-14:
@nfeske @chelmuth This is an attempt to describe your issue correctly and fix it:
1f924e12c6 nic_router: bind link state to remote DNS config

Can you please give it a try respectively review?

m-stein, 2022-09-15:
@chelmuth I've added two fixups:
4cc0b813ea fixup "nic_router: bind link state to remote DNS config" (README)
9ef6d80648 fixup "nic_router: bind link state to remote DNS config" (get rid of the notion of an "interface ready state")

After offline discussions with @chelmuth and @nfeske I conclude that there are two additional ideas regarding the ``dns_config_from`` that would be worth a try:
* The mechanism could be made more conformant to how remote DNS configs work with a manager component (see nic_router_dhcp_managed.run). This means that a domain that depends on a remote DNS config could go up directly simply applying an empty DNS config in case the remote one is not available yet. Dynamic propagation of changes to the remote DNS config must be implemented anyway.
* The DHCP protocol could actually - against my assumption so far - provide means for doing dynamic propagation of config updates (see https://en.wikipedia.org/wiki/Dynamic_Host_Configuration_Protocol#DHCP_message_types). If these means are also generally considered by client applications, this would allow us to move away from misusing link states for it. The benefit would be that it would work on all supported session types likewise and also across intermediate network components (given they forward it).

chelmuth, 2022-09-15:
@m-stein would you mind to rebase your merge_to_staging branch on current staging? 

Also, I'm afraid using the DHCP-protocol extension for config updates is a dead end according to discussions on Stackoverflow. These include references to postings of the ISC DHCP server lead developer voicing resistance to adopt DHCPFORCERENEW without authentication.

m-stein, 2022-09-15:
@nfeske I'm happyto hear that :)

m-stein, 2022-09-19:
@nfeske @chelmuth I've re-based my branch and added a fix-up:
 8f25368759 fixup "nic_router: bind link state to remote DNS config" (is_ready -> ready)

commit acfba5f8 (2022-09-15): nic_router: bind link state to remote DNS config

The NIC router README claims that the 'dns_config_from' attribute in a DHCP
server configuration binds the propagated link state of all interfaces at the
domain of the server to the validity of the IP config of the domain that is
given through 'dns_config_from'.

However, this was not true. The router missed to implement this detail which
led to clients of such a DHCP server sending DHCP DISCOVER packets too early.
These early DHCP DISCOVER packets were dropped by the router potentially
causing a big delay until the client started a new attempt [... cut]
commit 99254b4d (2022-09-21): nic_router: bind link state to remote DNS config

The NIC router README claims that the 'dns_config_from' attribute in a DHCP
server configuration binds the propagated link state of all interfaces at the
domain of the server to the validity of the IP config of the domain that is
given through 'dns_config_from'.

However, this was not true. The router missed to implement this detail which
led to clients of such a DHCP server sending DHCP DISCOVER packets too early.
These early DHCP DISCOVER packets were dropped by the router potentially
causing a big delay until the client started a new attempt [... cut]
closed 2022-09-21 by chelmuth