genodelabs/genode #4615: platform API: simplify _wait_for_device

skalk, 2022-09-15:
The current implementation of `_wait_for_device` in the `Platform::Connection` class is a bit too complicated, and although correct it can lead to nasty kernel warnings when a signal to an invalid signal handler is raised. We can supress this warning, and thereby simplify the waiting for a valid devices ROM.

nfeske, 2022-09-15:
When looking at the patch, I'm wondering if there is any chance that a signal may get lost?

When the platform driver tries to submit a signal just after the client called `sigh(Signal_context_capability())`, would the platform driver send the signal on the next call of `sigh(_handler)`?

skalk, 2022-09-16:
After rethinking it, I think you're right, but actually the problem existed already before. In general, we have to register the signal handler at construction time, and need to overwrite it, when the client requests it. Thereby, we cannot loose a signal, nor accidentally overwrite an explicitly set signal handler of the client. I'll sketch a solution in a follow-up commit.

skalk, 2022-09-16:
@nfeske I've replaced the original commit by commit a2e15db

nfeske, 2022-09-16:
The new version looks good to me. I merged the commit to staging.

closed 2022-09-21 by chelmuth