genodelabs/genode #4618: missing port check ineffective in some libraries

jschlatow, 2022-09-26:
Last week I stumbled upon the issue that the build system did not detect that I missed preparing a certain port. The error occured in a header-only library (empty _lib/mk/\<libname\>.mk_). Although I used `select_from_ports` in _lib/import/import-\<libname\>.mk_, the missing port was not detected. As it turned out, the reason for this was that the `select_from_ports` was used within an `INC_DIR +=` assignment that was evaluated lazily. In order to prevent these situations, we should make a habit of using the `:=` assignment when calling `select_from_ports`.

jschlatow, 2022-09-26:
Fixed by 46c259f.

chelmuth, 2022-10-18:
@jschlatow do you have any idea how we could prevent (hidden) slips like those I fixed in ec3f2ddea49188bb86f73603c32bea9af34611bc and 28b4d302681365771dc4c5415df0ae6a8536e9a3 in context of the changes in this issue?

jschlatow, 2022-10-19:
AFAIK there is no way to render a variable immutable in make or at least output a warning if it gets re-assigned, which would certainly help with detecting those issues. Another option could be to make `INC_DIR` an immediate variable but I'm not sure whether it is intentionally a deferred variable.

chelmuth, 2022-10-19:
Then, we have to live with the potential aliasing. I suggest to reduce the chance for this issue by applying the practice of naming the variable for the plain port directory `<PORT_NAME>_PORT_DIR` in future changes consistently.

closed 2022-11-17 by chelmuth