genodelabs/genode #4621: qt5: fix build errors with GCC 11 host tool chain

cproc, 2022-09-27:
With GCC 11 as host tool chain (for example on Ubuntu 22.04), the `limits` header file needs to be included in some of the Qt5 source files. [1][2]

[1] https://bugreports.qt.io/browse/QTBUG-93452
[2] https://github.com/OpenMandrivaAssociation/qt5-qtbase/blob/b91146253cbf8caabc90ee438050e077d4c4b7e9/qt5-qtbase-gcc11.patch

cproc, 2022-09-27:
Fixed by ff4bc28.

chelmuth, 2022-10-11:
@cproc do you have any idea how to fix the following tool-chain build error?

```
building ali2dep...
gprconfig: can't find a toolchain for the following configuration:
gprconfig: language 'Ada', target 'all', runtime 'default'
make: *** [tool/tool_chain:462: .../genode.git/build/tool_chain-21.05/ali2dep/ali2dep] Error 1
```

cproc, 2022-10-11:
@chelmuth: not really. I think gprconfig is looking for the bootstrap gnat installation in `build/tool_chain-21.05/bootstrap/install`. Does that exist?

commit ff4bc282 (2022-09-28): qt5: fix build errors with GCC 11 host tool chain

Fixes #4621
closed 2022-10-12 by chelmuth