genodelabs/genode #4623: libports: update expat and fatfs

rite, 2022-10-03:
- The source archive of the Expat XML Parser 2.4.6 hosted on https://sourceforge.net/projects/expat/ has been renamed to `expat-2.4.6-RENAMED-VULNERABLE-PLEASE-USE-2.4.9-INSTEAD.tar.gz`.

- It looks like http://www.elm-chan.org is blocking certain IP because downloading the FatFs source archive on certain machines fails with "403 Forbidden".

rite, 2022-10-03:
- a68b6cb4b01b39f3617d793e8ddf910cc57485bb updates the version of the Expat XML parser without any other changes/patches. Our tests that require expat as a transitive dependency didn't break, however we don't explicitly test the expat port in depth.

- f79a4e1f7bc51716a75257af9545e80de5f3560d uses a GitHub mirror for FatFs. Unfortunately, the sources downloaded from this mirror differ slightly from the upstream host. Please take this into account when considering cherry-picking this commit to master.

chelmuth, 2022-10-04:
Thanks @rite,

> * [a68b6cb](https://github.com/genodelabs/genode/commit/a68b6cb4b01b39f3617d793e8ddf910cc57485bb) updates the version of the Expat XML parser without any other changes/patches. Our tests that require expat as a transitive dependency didn't break, however we don't explicitly test the expat port in depth and I didn't review the changes from 2.4.6 to 2.4.9.

Please consider 717dfca385b17456e3eccd7923b3826a3f0ed9cc, which already hit staging and will be mastered soon.

>     * [f79a4e1](https://github.com/genodelabs/genode/commit/f79a4e1f7bc51716a75257af9545e80de5f3560d) uses a GitHub mirror for FatFs. Unfortunately, the sources downloaded from this mirror differ slightly from the upstream host. Please take this into account when considering cherry-picking this commit to master.

I'll have a look at the differences to our current port.

rite, 2022-10-04:
> Please consider https://github.com/genodelabs/genode/commit/717dfca385b17456e3eccd7923b3826a3f0ed9cc, which already hit staging and will be mastered soon.

Oh, thanks for the heads-up, I wasn't aware of this. Please just ignore my commit a68b6cb4b01b39f3617d793e8ddf910cc57485bb then.

chelmuth, 2022-11-17:
Just had a look into the fatfs commit and discovered only small noncritical upstream source changes. Merged to staging and marked as fix commit.

closed 2022-11-29 by chelmuth