genodelabs/genode #4625: dde_linux: provide __phys_to_pfn for arm64

cnuke, 2022-10-05:


cnuke, 2022-10-05:
Commit [09d709e](https://github.com/cnuke/genode/commit/09d709e83578b041ab5194643c8b68e2e98ccaf1) address this issue.

chelmuth, 2022-10-13:
@cnuke the commit would close this issue as fixed but contains lines with `XXX`. Please clarify the state and mark the issue as fixed appropriately.

cnuke, 2022-10-14:
I marked the additions to indicate that they might have to be double checked or maybe could be provided in another way. They are good enough for now for the one driver though.

skalk, 2022-11-01:
I checked @cnuke analysis, and can confirm it. But I think a:
``
#include <linux/pfn.h>
``
at the beginning of memory.h, like in the original asm-generic/memroy_model.h would be good, to prevent potential compiler errors.

cnuke, 2022-11-04:
Commit [62f410d](https://github.com/cnuke/genode/commit/62f410da3d96a61978bd2508a2b472bb65a9c32e) contains the cleaned up version.

commit 64bde7f9 (2022-11-07): dde_linux: provide __phys_to_pfn for arm64

This macro is needed by the 'pinephone_camera_drv' driver port.

Fixes #4625.
closed 2022-11-17 by chelmuth