genodelabs/genode #4629: dde_bsd: update audio driver to OpenBSD 7.1

cnuke, 2022-10-05:


cnuke, 2022-10-05:
Commit [c84535d](https://github.com/cnuke/genode/commit/c84535dc148bfbc899faffa13d950ca973025d3d) adds *disable no snoop* handling for AMD/ATI HDA devices, while commit [5fa10c4](https://github.com/cnuke/genode/commit/5fa10c471c6d5e9e60b7fd201fbac53fbdb9e610) contains the update. The new driver was successfully tested by running `audio_out.run` on a X201, X250, Fuji4 and Fuji5c. In addition `audio_in.run` works on X201, X250 and Fuji4 (also using a headset).

nfeske, 2022-10-05:
Thank you for the commit series, which I just merged to staging.

cnuke, 2022-10-06:
Please merge commit [519e33e](https://github.com/cnuke/genode/commit/519e33e558e3b43e27840ddd4597d894c701cd34). It prevents a page-fault in the driver component when it is restarted (see #4631).

TL;DR when the driver component is removed in Sculpt the device is still running and that will be detected by the driver the next time it is started. During its attempt to power down the already running codec it will access an uninitialized object. The patch moves the access after the allocation.

With the patch in place the driver behaves like the old one, i.e., you may have to restart it twice(!) to properly configure the device.

nfeske, 2022-10-06:
Thanks for the welcome fixup, which I merged to staging. I very much appreciate the little cleanups left and right, like the simplification of the port file.

commit b5d0d545 (2022-10-05): dde_bsd: update audio driver to OpenBSD 7.1

This commit updates the driver from version 6.6 to 7.1. In contrast
to the old driver the new one will now probe all available HDA devices
and will drive the first usable one, e.g.:

```
[init -> audio_drv] azalia0 [8086:160c]
[init -> audio_drv] :
[init -> audio_drv] azalia0: no supported codecs
[init -> audio_drv] azalia1 [8086:9ca0]
[init -> audio_drv] :
[init -> audio_drv] azalia1: codecs: Realtek ALC292
[init -> audio_drv] audio0 at azalia1
```

Fixes #4629.
closed 2022-10-12 by chelmuth