genodelabs/genode #4630: Intel framebuffer driver fails on x250 + Dell UltraSharp 4K display

nfeske, 2022-10-05:
Once when connecting the external display, the intel framebuffer driver apparently enters a new code path:

```
[drivers -> dynamic -> intel_fb_drv] i915 00:02.0: [drm] *ERROR* AUX C/DDI C/PHY C: did not complete or timeout within 10ms (status 0xad4003ff)
[drivers -> dynamic -> intel_fb_drv] i915 00:02.0: [drm] fb0: i915 frame buffer device
[drivers -> dynamic -> intel_fb_drv] eDP-1:  enable name='1920x1080' id=1 1920x1080@60 - no mode match: 0x0
[drivers -> dynamic -> intel_fb_drv] [31mError: Function kvfree_call_rcu not implemented yet!
[drivers -> dynamic -> intel_fb_drv] Backtrace follows:
[drivers -> dynamic -> intel_fb_drv] 0x115b3cb
[drivers -> dynamic -> intel_fb_drv] 0x1084f35
[drivers -> dynamic -> intel_fb_drv] 0x1059881
[drivers -> dynamic -> intel_fb_drv] 0x1012bb4
[drivers -> dynamic -> intel_fb_drv] 0x105963d
[drivers -> dynamic -> intel_fb_drv] 0x1012d87
[drivers -> dynamic -> intel_fb_drv] 0x1083472
[drivers -> dynamic -> intel_fb_drv] 0x11772a1
[drivers -> dynamic -> intel_fb_drv] 0x11776f7
[drivers -> dynamic -> intel_fb_drv] 0x1162226
[drivers -> dynamic -> intel_fb_drv] 0x11ac5f4
[drivers -> dynamic -> intel_fb_drv] Will sleep forever...
```

Sculpt's driver manager tries to respawn the dead driver. However, the new instance fails to (re-)initialize the hardware at that point:
```
[drivers -> dynamic -> intel_fb_drv] --- Intel framebuffer driver started ---
[drivers -> dynamic -> intel_fb_drv] NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 256
[drivers -> dynamic -> intel_fb_drv] clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[drivers -> dynamic -> intel_fb_drv] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[drivers -> dynamic -> intel_fb_drv] clocksource: Switched to clocksource dde_counter
[drivers -> dynamic -> intel_fb_drv] disabling PPGTT to avoid GPU code paths
[drivers -> dynamic -> intel_fb_drv] pci_iomap: request for dev: 00:02.0 bar: 0
[drivers -> dynamic -> intel_fb_drv] [31mError: could not read I/O port resource 0x3cc
[drivers -> dynamic -> intel_fb_drv] [31mError: could not write I/O port resource 0x3c2
[drivers -> dynamic -> intel_fb_drv] i915 00:02.0: [drm] Failed to load DMC firmware i915/skl_dmc_ver1_27.bin. Disabling runtime power management.
[drivers -> dynamic -> intel_fb_drv] [31mError: could not write I/O port resource 0x3c4
[drivers -> dynamic -> intel_fb_drv] [31mError: could not read I/O port resource 0x3c5
[drivers -> dynamic -> intel_fb_drv] [31mError: could not write I/O port resource 0x3c5
[drivers -> dynamic -> intel_fb_drv] i915 00:02.0: [drm] DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915[drm] Initialized i915 1.6.0 20201103 for 00:02.0 on minor 0
[drivers -> dynamic -> intel_fb_drv] sched_clock: Marking stable (121920000, 9386000)->(130849000, 457000)
[drivers -> dynamic -> intel_fb_drv] i915 00:02.0: [drm] *ERROR* AUX C/DDI C/PHY C: did not complete or timeout within 10ms (status 0xad4003ff)
[drivers -> dynamic -> intel_fb_drv] i915 00:02.0: [drm] *ERROR* AUX C/DDI C/PHY C: did not complete or timeout within 10ms (status 0xad4003ff)
[drivers -> dynamic -> intel_fb_drv] i915 00:02.0: [drm] *ERROR* AUX A/DDI A/PHY A: did not complete or timeout within 10ms (status 0xac2003ff)
[drivers -> dynamic -> intel_fb_drv] i915 00:02.0: [drm] *ERROR* AUX C/DDI C/PHY C: did not complete or timeout within 10ms (status 0xad4003ff)
```

nfeske, 2022-10-05:
By taking the implementation of `kvfree_call_rcu`, we get a step further, hitting another dummy.

```
[drivers -> dynamic -> intel_fb_drv] [31mError: Function drm_scdc_set_high_tmds_clock_ratio not implemented yet!
```

chelmuth, 2022-10-06:
If I'm not mistaken we tried to enable this subsystem in the past and had issues with S938. @alex-ab do you recollect more details or (even better) test the current staging version with the device?

nfeske, 2022-10-06:
Interesting. As far as I could see, the added file merely adds a few functions that were previously handled by `lx_emul_trace_and_stop`. Since those weren't called previously, I figured that adding `drm_scdc_helper.c` does not pose risks for regressions. Given your remark, I'd very much appreciate @alex-ab giving the latest version a spin.

commit 5ce8e5cd (2022-10-05): intel_fb_drv: add gpu/drm/drm_scdc_helper.c

With this patch, the Dell UltraSharp 4K display works in principle when
connected via HDMI to a Lenovo x260 laptop.

Fixes #4630
closed 2022-10-12 by chelmuth