genodelabs/genode #4631: Sculpt 22.10: Minor issues

ssumpf, 2022-10-05:
A place to put minor issues while testing the upcoming Sculpt release.

cnuke, 2022-10-06:
FWIW, while fetching a large pkg I noticed what seems to be reoccuring diagnostic messages, e.g.

```
[init -> runtime -> depot_query] Warning: directory 'alex-ab/raw/vm-tc-firefox/105.0.1-tc-13' does not exist
[init -> runtime -> depot_query] Warning: directory 'alex-ab/bin/x86_64/vfs_block/2022-10-05' does not exist
[init -> runtime -> depot_query] Warning: directory 'alex-ab/bin/x86_64/vfs_import/2022-10-05' does not exist
[init -> runtime -> depot_query] Warning: directory 'alex-ab/bin/x86_64/seoul/2022-10-05' does not exist
[init -> runtime -> depot_query] Warning: directory 'alex-ab/raw/seoul-config/2022-06-06' does not exist
[init -> runtime -> depot_query] Warning: directory 'alex-ab/raw/vm-tc-firefox/105.0.1-tc-13' does not exist
[init -> runtime -> depot_query] Warning: directory 'alex-ab/bin/x86_64/vfs_block/2022-10-05' does not exist
[init -> runtime -> depot_query] Warning: directory 'alex-ab/bin/x86_64/vfs_import/2022-10-05' does not exist
[init -> runtime -> depot_query] Warning: directory 'alex-ab/bin/x86_64/seoul/2022-10-05' does not exist
[init -> runtime -> depot_query] Warning: directory 'alex-ab/raw/seoul-config/2022-06-06' does not exist
[…]
```

Commit [f0044cf](https://github.com/cnuke/genode/commit/f0044cfa89230c3965c2bc0e21a390f8fe4b9e50) removes the message but it is probably put in place for a reason. (Since I use a somewhat modified Sculpt branch it is likely I introduced the problem.)

cnuke, 2022-10-06:
Restarting the updated `dde_bsd` driver leads a page-fault, e.g.
```
[init -> runtime -> bsd_audio_drv] --- probe drivers ---
[init -> runtime -> bsd_audio_drv] azalia0 [8086:160c]
[init -> runtime -> bsd_audio_drv] : 
no RM attachment (WRITE pf_addr=0x4 pf_ip=0x101d490 from pager_object: pd='init -> runtime -> bsd_audio_drv' thread='ep') 
page fault, pd='init -> runtime -> bsd_audio_drv' thread='ep' cpu=0 ip=0x101d490 address=0x4 stack pointer=0x404fe2d0 qualifiers=0x6 irUWp reason=1
```

I'll look into that.

nfeske, 2022-10-06:
I'd like to remove vbox5 from the default Sculpt index and the sculpt_distribution. By now, we should all have migrated to vbox6. What do you think?

nfeske, 2022-10-10:
> I think that I´ll drop these intermediate attempts and merely issue a new query once the installation is settled.

@cnuke, commit c6900821d7a6835f21d8c7e08b52ff3c1cda18eb + fixup 79471f04fe4a9083d5ffdff2178a00aedd4a3503 should remove the annoying warnings.

cnuke, 2022-10-10:
@nfeske thanks, those commit indeed address the warnings.

However (even without the fixup commits), as I noticed now on the X250, the download does not succeed for larger pkgs when using the ethernet nic. At some point it stops and I cannot even ping the machine after that. The problem does not occur on Fuji3.

nfeske, 2022-10-10:
> as I noticed now on the X250, the download does not succeed for larger pkgs when using the ethernet nic

This symptom sounds unrelated unrelated to the Sculpt manager's change though, doesn't it? At least, I cannot draw any connection. I wonder, may the former pressure on the file system have masked a networking issue?

cnuke, 2022-10-10:
Yes it is unrelated. I just noticed it now as it was the first time I used the X250 for testing.

cnuke, 2022-10-10:
@chelmuth booting from an USB stick makes all the difference. So it points to an PXE related problem.

ssumpf, 2022-10-11:
@chelmuth, @nfeske: When issuing a `poweroff` I experienced a RAM resource request from `acpica`. c04a0e8 should fix that.

nfeske, 2022-10-12:
All fixes went to master. Thanks everyone for testing and refining.

closed 2022-10-12 by nfeske

alex-ab, 2022-10-13:
With ae84eb1d9bcca247f758d05af1fe3832741f95ce the battery app works now with the current acpica port again.

nfeske, 2022-10-13:
> With https://github.com/genodelabs/genode/commit/ae84eb1d9bcca247f758d05af1fe3832741f95ce the battery app works now with the current acpica port again.

That is super cool! Thank you for solving this long-standing riddle. :-)