genodelabs/genode #4632: Sculpt + Vbox5: Guest fails to get IP config

m-stein, 2022-10-06:
After updating Sculpt, @ssumpf reported that his VBox guest (gentoo) doesn't get an IP config via DHCP on startup. However, once he reconfigured the NIC router, the guest was able to complete DHCP and use the network just fine. We found that in the bad case, there appear no packets on the fresh NIC session. This is because the NIC device model in VBox thinks that the session is still "down" (synchronized link state) while the internal link state (real link state) of the session at the router is already "up". This is very likely related to issue https://github.com/genodelabs/genode/issues/4612 that changed the initial link state of a NIC session component in the router to be "down" and not become before it gets attached to a domain with none or a ready-to-use DHCP server.

ssumpf, 2022-10-07:
@m-stein: Here we go:
```
### vm_vbox6 -> vbox -> 0: ack=0 tls=DOWN              sigh=0 submit=0
### vm_vbox6 -> vbox -> 0: ack=0 tls=DOWN              sigh=0 submit=1
### vm_vbox6 -> vbox -> 0: ack=0 tls=DOWN_UP           sigh=0 submit=0
### vm_vbox6 -> vbox -> 0: ack=1 tls=UP                sigh=1 submit=1
### vm_vbox6 -> vbox -> 0: ack=1 tls=UP_ACKNOWLEDGED   sigh=1 submit=0
```

ssumpf, 2022-10-07:
@m-stein: I also found when switching from a working Wifi connection to Wired, the wired connection does not receive an IP address. May this be related to this issue as well?

m-stein, 2022-10-07:
@ssumpf Thanks for the log! I don't get yet. Your NIC client (vbox dev model) reads out the "down" state first and then also the following "up" state. The signal sumbissions happen just as expected. I'm wondering why after the above cascade your dev model still thinks it's down. The last call to ``Nic_session::link_state`` should return true AFAICS.

m-stein, 2022-10-07:
@ssumpf Regarding your other observation, I'd require more information. Could you share a ``<config verbose_packets="yes" verbose_packet_drop="yes" verbose_domain_state="yes" verbose="yes">`` log while doing the switch?

ssumpf, 2022-10-10:
@chelmuth, @m-stein: 1ed6b69 fixes VirtualBox IP configuration.

m-stein, 2022-10-10:
@ssumpf ... and therefore this issue. Could you please add ``Fixes #4632`` to the commit message?

ssumpf, 2022-10-10:
> @ssumpf ... and therefore this issue. Could you please add `Fixes #4632` to the commit message?

Did already.

nfeske, 2022-10-10:
According to my tests, the fix works well. Thanks @m-stein and @ssumpf!

m-stein, 2022-10-10:
@nfeske Thanks for testing it so fast :)

nfeske, 2022-10-12:
Fixed in master.

closed 2022-10-12 by nfeske