genodelabs/genode #4635: ipxe nic: sporadic dma fault on restart

m-stein, 2022-10-12:
At b447c2cb91, which is based on a preliminary state of Sculpt 22.10, when switching between "Wired" and other options in the network manager, I sometimes observe a DMA fault when the NIC driver gets re-started. After this DMA fault, I fail to get an IP config with the NIC driver. Note that this is not due to unusual DHCP delays - the router continuously fails in doing DHCP DISCOVER in my case. Here is the log of such an attempt where the fault occurred during the first driver restart (I commented out some stuff in order to keep it digestible):
```
[kernel] [ 0] TSC:2112000 kHz BUS:0 kHz DL
[kernel] [ 0] CORE:0:0:0 6:8e:c:7 [ca] Intel(R) Core(TM) i7-8665U CPU @ 1.90GHz
[kernel] [ 4] CORE:0:0:1 6:8e:c:7 [ca] Intel(R) Core(TM) i7-8665U CPU @ 1.90GHz
[kernel] [ 5] CORE:0:1:1 6:8e:c:7 [ca] Intel(R) Core(TM) i7-8665U CPU @ 1.90GHz
[kernel] [ 1] CORE:0:1:0 6:8e:c:7 [ca] Intel(R) Core(TM) i7-8665U CPU @ 1.90GHz
[kernel] [ 2] CORE:0:2:0 6:8e:c:7 [ca] Intel(R) Core(TM) i7-8665U CPU @ 1.90GHz
[kernel] [ 6] CORE:0:2:1 6:8e:c:7 [ca] Intel(R) Core(TM) i7-8665U CPU @ 1.90GHz
[kernel] [ 3] CORE:0:3:0 6:8e:c:7 [ca] Intel(R) Core(TM) i7-8665U CPU @ 1.90GHz
[kernel] [ 7] CORE:0:3:1 6:8e:c:7 [ca] Intel(R) Core(TM) i7-8665U CPU @ 1.90GHz
[core] Hypervisor features VMX
[core] Hypervisor reports 4x2 CPUs
[core] mapping: affinity space -> kernel cpu id - package:core:thread
[core]  remap (0x0) -> 0 - 0:0:0) boot cpu
[core]  remap (0x1) -> 4 - 0:0:1) 
[core]  remap (1x0) -> 1 - 0:1:0) 
[core]  remap (1x1) -> 5 - 0:1:1) 
[core]  remap (2x0) -> 2 - 0:2:0) 
[core]  remap (2x1) -> 6 - 0:2:1) 
[core]  remap (3x0) -> 3 - 0:3:0) 
[core]  remap (3x1) -> 7 - 0:3:1) 
[core] ROM modules:
[core]  ROM: ...
[core] 168M kernel memory
[core] Genode 22.08-184-g9430586292 <local changes>
[core] 15769 MiB RAM and 63253 caps assigned to init
[core] [init] Warning: runtime: configured RAM exceeds available RAM, proceed with 16112193472
[runtime] parent provides
[runtime]   service "ROM"
[runtime]   service "PD"
[runtime]   service "RM"
[runtime]   service "CPU"
[runtime]   service "LOG"
[runtime]   service "IO_MEM"
[runtime]   service "IO_PORT"
[runtime]   service "IRQ"
[runtime]   service "Gui"
[runtime]   service "Timer"
[runtime]   service "Block"
[runtime]   service "Report"
[runtime]   service "Platform"
[drivers] Warning: dynamic: configured RAM exceeds available RAM, proceed with 157583364
[drivers] Warning: dynamic: configured caps exceed available caps, proceed with 2385
[drivers -> usb_hid_drv] --- USB HID input driver ---
[drivers -> usb_hid_drv] Warning: use compatibility mode: will claim all HID devices from USB report
[drivers -> usb_hid_drv] Configured HID screen with 0x0 (multitouch=false)
[drivers -> ps2_drv] Using keyboard with scan code set 1 (xlate)
[drivers -> acpi_drv] ...
[drivers -> usb_drv] ...
[drivers -> dynamic -> intel_gpu_drv] ...
[drivers -> dynamic -> intel_fb_drv] ...
[drivers -> usb_drv] usb 3-2: new high-speed USB device number 2 using xhci_hcd
[drivers -> dynamic -> intel_fb_drv] ...
[runtime] parent provides
[runtime]   service "VM"
[runtime]   service "Usb"
[runtime]   service "File_system"
[runtime]   service "Rtc"
[runtime]   service "TRACE"
[runtime]   service "Event"
[runtime]   service "Capture"
[runtime]   service "Gpu"
[runtime] child "runtime_view" ...
[runtime] child "panel_view" ...
[runtime] child "menu_view" ...
[runtime] child "settings_view" ...
[runtime] child "network_view" ...
[runtime] child "popup_view" ...
[runtime] child "file_browser_view" ...
[... 219 lines cut ...]
[runtime -> inspect -> terminal] Warning: Application Cursor Keys not implemented
[runtime -> inspect -> terminal] Warning: Alternate Screen (new xterm code) not implemented
[runtime -> nic_router] [uplink] snd ETH 00:2b:67:41:c3:67 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP DISCOVER 00:2b:67:41:c3:67 > 0.0.0.0
[runtime -> nic_router] [uplink] snd ETH 00:2b:67:41:c3:67 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP DISCOVER 00:2b:67:41:c3:67 > 0.0.0.0
[runtime -> inspect -> bash -> 3] Warning: unsupported ioctl (request=0x802c7414)
[runtime -> inspect -> terminal] Warning: Alternate Screen (new xterm code) not implemented
[runtime -> inspect -> terminal] Warning: Application Cursor Keys not implemented
[runtime -> inspect -> bash -> 3] Warning: unsupported ioctl (request=0x802c7414)
[runtime -> inspect -> bash -> 3] Warning: unsupported ioctl (request=0x802c7414)
[runtime -> inspect -> terminal] Warning: Application Cursor Keys not implemented
[runtime -> inspect -> terminal] Warning: Alternate Screen (new xterm code) not implemented
[runtime -> nic_router] [uplink] snd ETH 00:2b:67:41:c3:67 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP DISCOVER 00:2b:67:41:c3:67 > 0.0.0.0
[runtime -> nic_router] [uplink] snd ETH 00:2b:67:41:c3:67 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP DISCOVER 00:2b:67:41:c3:67 > 0.0.0.0
[runtime -> inspect -> bash -> 4] Warning: unsupported ioctl (request=0x802c7414)
[runtime -> inspect -> terminal] Warning: Alternate Screen (new xterm code) not implemented
[runtime -> inspect -> terminal] Warning: Application Cursor Keys not implemented
[runtime -> nic_router] [uplink] snd ETH 00:2b:67:41:c3:67 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP DISCOVER 00:2b:67:41:c3:67 > 0.0.0.0
[runtime -> nic_router] [uplink] snd ETH 00:2b:67:41:c3:67 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP DISCOVER 00:2b:67:41:c3:67 > 0.0.0.0
[runtime -> nic_router] [uplink] snd ETH 00:2b:67:41:c3:67 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP DISCOVER 00:2b:67:41:c3:67 > 0.0.0.0
[runtime -> inspect -> bash -> 4] Warning: unsupported ioctl (request=0x802c7414)
[runtime -> inspect -> bash -> 4] Warning: unsupported ioctl (request=0x802c7414)
[runtime -> inspect -> terminal] Warning: Application Cursor Keys not implemented
[runtime -> inspect -> terminal] Warning: Alternate Screen (new xterm code) not implemented
[runtime -> nic_router] [uplink] snd ETH 00:2b:67:41:c3:67 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP DISCOVER 00:2b:67:41:c3:67 > 0.0.0.0
[runtime -> nic_router] [uplink] snd ETH 00:2b:67:41:c3:67 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP DISCOVER 00:2b:67:41:c3:67 > 0.0.0.0

m-stein, 2022-10-12:
I should add that I observed the issue with my ThinkPad T490 with an Intel I219 Rev 30 NIC.

m-stein, 2022-10-12:
@skalk and I created d57a7f3912, that shows that the DMA fault (varying address) is always in a region [0x00a00000..0x00b00000) that was attached as DMA by the platform driver beforehand during an RPC ``acquire_single_device`` by the NIC driver. This region is freed again at the platform driver in ``_release_device`` on NIC driver shutdown. Tha commit also contains outcommented debugging log calls in the NIC driver. I commented them out as we noticed that the issue disappears with them.

chelmuth, 2022-10-12:
Could you please check the state of the LEDs on the ethernet ports of your notebook and switch?

m-stein, 2022-10-12:
I have found no LEDs at my Fritzbox, an at the T490 there seem to be only the two of the RJ45 socket. In my native Linux, the green one is always on while the plug is plugged and only shortly goes off when switching network mode (Off <-> Wired). As long as the green one is on, the orange one blinks regularly. In preliminary Sculpt, it's almost the same with the difference that the green one is not going off even when switching net mode.

I observed further details. The fault normally occurs about 1 second after the first log message of the NIC driver and it does not occur when only shutting down the NIC driver and waiting without a restart. Normally, the fault occurs at the first restart. And, even after the fault, the NIC driver keeps noticing when I remove and re-insert the plug, because the NIC router reports the drivers Uplink session disconnecting and connecting.

nfeske, 2022-10-12:
This issue seems to be related to #4538.

alex-ab, 2022-10-14:
We tested now your setup on several machines, which does not trigger. What would be interesting to know:
- Did this work before for you, e.g. with Sculpt 22.04 ?
- Does disabling the iommu support helps (by removal of the parameter on the kernel commandline) ?

m-stein, 2022-10-24:
@alex-ab Thanks for looking into this!

AFAIK, I never had problems with IPXE NIC in Sculpt 22.04.

I removed IOMMU support as you suggested and then the router always received an IP config with the IPXE NIC. There was a kernel message ``sys_assign_pci: Invalid Hint (0xfe)`` though every time the driver was restarted and I don't know what to make of it:

```
[runtime] child "nic_drv"
[runtime]   RAM quota:  20232K
[runtime]   cap quota:  266
[runtime]   ELF binary: nic_drv
[runtime]   priority:   2
[runtime] child "nic_router"
[runtime]   RAM quota:  9992K
[runtime]   cap quota:  266
[runtime]   ELF binary: nic_router
[runtime]   priority:   2
[runtime]   provides service Nic
[runtime]   provides service Uplink
[runtime -> nic_drv] --- iPXE NIC driver started ---
[runtime -> nic_router] [uplink] dynamic IP config: none
[runtime -> nic_router] [uplink] NIC sessions: 0
[runtime -> nic_router] [default] static IP config: interface 10.0.1.1/24, gateway 0.0.0.0, P2P 0
[runtime -> nic_router] [default] NIC sessions: 0
[runtime] child "nic_router" announces service "Nic"
[runtime] child "nic_router" announces service "Uplink"
[drivers -> platform_drv] Assignment of PCI device 00:1f.6 to device PD failed, no IOMMU?!
[runtime -> nic_drv] Found:  8086:15bd (rev 30)
[runtime -> nic_drv] using driver i219lm-6
[runtime -> nic_drv] PCI device 00:01.0 latency timer is unreasonably low at 0. Setting to 32.
[runtime -> nic_drv] bus_addr = 160000 len = 20000
[runtime -> nic_drv] snprintf not implemented
[runtime -> nic_drv] MAC address 00:2b:67:41:c3:67
[runtime -> nic_router] [uplink] NIC sessions: 1
[runtime -> nic_router] [uplink] snd ETH 00:2b:67:41:c3:67 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP DISCOVER 00:2b:67:41:c3:67 > 0.0.0.0
[runtime -> nic_router] [uplink] rcv ETH 00:fd:45:fc:6f:80 > 00:2b:67:41:c3:67 IPV4 10.0.0.2 > 10.0.0.37 UDP 67 > 68 DHCP OFFER 00:2b:67:41:c3:67 > 10.0.0.2
[runtime -> nic_router] [uplink] snd ETH 00:2b:67:41:c3:67 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP REQUEST 00:2b:67:41:c3:67 > 0.0.0.0
[runtime -> nic_router] [uplink] rcv ETH 00:fd:45:fc:6f:80 > 00:2b:67:41:c3:67 IPV4 10.0.0.2 > 10.0.0.37 UDP 67 > 68 DHCP ACK 00:2b:67:41:c3:67 > 10.0.0.2
[runtime -> nic_router] [uplink] dynamic IP config: interface 10.0.0.37/24, gateway 10.0.0.1, P2P 0, DNS server 10.0.0.2
[kernel] [ 0] sys_assign_pci: Invalid Hint (0xfe)
[runtime -> nic_router] [uplink] rcv ETH 00:fd:45:fc:6f:80 > ff:ff:ff:ff:ff:ff ARP 00:fd:45:fc:6f:80 10.0.0.2 > 00:00:00:00:00:00 10.0.0.5 cmd 1
[runtime] child "nic_drv"
[runtime]   RAM quota:  20232K
[runtime]   cap quota:  266
[runtime]   ELF binary: nic_drv
[runtime]   priority:   2
[runtime] child "nic_router"
[runtime]   RAM quota:  9992K
[runtime]   cap quota:  266
[runtime]   ELF binary: nic_router
[runtime]   priority:   2
[runtime]   provides service Nic
[runtime]   provides service Uplink
[runtime -> nic_drv] --- iPXE NIC driver started ---
[runtime -> nic_router] [uplink] dynamic IP config: none
[runtime -> nic_router] [uplink] NIC sessions: 0
[runtime -> nic_router] [default] static IP config: interface 10.0.1.1/24, gateway 0.0.0.0, P2P 0
[runtime -> nic_router] [default] NIC sessions: 0
[runtime] child "nic_router" announces service "Nic"
[runtime] child "nic_router" announces service "Uplink"
[drivers -> platform_drv] Assignment of PCI device 00:1f.6 to device PD failed, no IOMMU?!
[runtime -> nic_drv] Found:  8086:15bd (rev 30)
[runtime -> nic_drv] using driver i219lm-6
[runtime -> nic_drv] PCI device 00:01.0 latency timer is unreasonably low at 0. Setting to 32.
[runtime -> nic_drv] bus_addr = 160000 len = 20000
[runtime -> nic_drv] snprintf not implemented
[runtime -> nic_drv] MAC address 00:2b:67:41:c3:67
[runtime -> nic_router] [uplink] NIC sessions: 1
[runtime -> nic_router] [uplink] snd ETH 00:2b:67:41:c3:67 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP DISCOVER 00:2b:67:41:c3:67 > 0.0.0.0
[runtime -> nic_router] [uplink] rcv ETH 00:fd:45:fc:6f:80 > 00:2b:67:41:c3:67 IPV4 10.0.0.2 > 10.0.0.37 UDP 67 > 68 DHCP OFFER 00:2b:67:41:c3:67 > 10.0.0.2
[runtime -> nic_router] [uplink] snd ETH 00:2b:67:41:c3:67 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP REQUEST 00:2b:67:41:c3:67 > 0.0.0.0
[runtime -> nic_router] [uplink] rcv ETH 00:fd:45:fc:6f:80 > 00:2b:67:41:c3:67 IPV4 10.0.0.2 > 10.0.0.37 UDP 67 > 68 DHCP ACK 00:2b:67:41:c3:67 > 10.0.0.2
[runtime -> nic_router] [uplink] dynamic IP config: interface 10.0.0.37/24, gateway 10.0.0.1, P2P 0, DNS server 10.0.0.2
[runtime -> nic_router] [uplink] rcv ETH c8:3a:35:12:24:28 > 01:00:5e:7f:ff:fa IPV4 10.0.0.1 > 239.255.255.250 UDP 6454 > 1900 
[kernel] [ 0] sys_assign_pci: Invalid Hint (0xfe)
```

alex-ab, 2022-10-24:
> AFAIK, I never had problems with IPXE NIC in Sculpt 22.04.
> I removed IOMMU support as you suggested and then the router always received an IP config with the IPXE NIC. There 
was a kernel message `sys_assign_pci: Invalid Hint (0xfe)` though every time the driver was restarted and I don't know what to make of it:

The message by the kernel appears if you have the iommu disabled, which you did.

Since it works without IOMMU and worked before 22.10 with the old platform driver, it sounds that some inaccuracies and/or different usage pattern with the setup of the iommu pagetable (device pd code) in the new platform driver causes this issue. I fear without debugging on your hardware and comparing/understanding the subtle difference of the old/new platform driver is necessary.

m-stein, 2022-10-24:
Thanks for the explanation. This is not a very pressing issue for me and I'm not at all familiar with the platform driver or iommu pagetables. So, I'd let it rest for now if no one is eager to debug it with my machine. However, if someone wants to investigate further, please feel free to approach me ;)

m-stein, 2023-04-20:
I cannot observe this anymore with preliminary Sculpt 23.04. So, I'd close the issue.