genodelabs/genode #4642: genode block C API: update config ROM by signal

skalk, 2022-10-17:
Currently, the config ROM in the C++ backend of the Genode C API for C-based, ported block drivers doesn't get updated when the configuration has changed.

nfeske, 2022-10-17:
I think the right approach would be removing the `Attached_rom_dataspace` from `block.cc` and add a C++ API function like this:

```
void genode_block_apply_config(Xml_node const &);
```

This approach would lay the configuration in the hands of the component's `Main`, which is then free to supply a static config, or take it from a ROM, and update it dynamically as it sees fit.

skalk, 2022-10-19:
@nfeske: yes, I recognized the design issue of having the config ROM and signal handler inside the C++ backend too, when investigating the missing configuration update. I just wanted to solve this problem first. But I fully understand when you do not want to repeatedly change the os API. Therefore, I will provide an alternative solution.

closed 2022-10-19 by skalk