genodelabs/genode #4646: idea: report_rom should output warning() diagnosis msg on missing policy

ttcoder, 2022-10-19:
Context: I was wondering why platform_drv was failing to provide devices to the ahci driver, ps2 driver etc via rom_filter, which itself accesses report_rom.

Turned out, rom_filter was being *silently* denied access to report_rom, until I added this missing bit in report_rom's config:

				<policy label="rom_filter -> pci_devices" report="pci_decode -> devices"/>

It took me a bit of time to realize where the problem was and how to fix it.

Suggestion:
I noticed that in some cases, some other components output a Genode::warning like "warning: denying access so-and-so on missing policy, requested by so and so client", in that kind of situation.
With that kind of information, it becomes very quick and easy to realize you mis-configured your report_rom component... But if the problem is not reported, as both client and server keep quiet about it, it takes more time (supposedly even experienced genodians could benefit from the message?). My 2 cents :-)

P.S. that warning-message feature could possibly be enabled by a special attribute acting as a 'switch' in the xml config, in case the message is not desirable in some cases.

PPS. or alternatively, maybe it should actually be up to the client (i.e. rom_filter here) to log a message when it fails to connect to a report_rom

chelmuth, 2022-10-20:
I tried this and if I remove the report_rom policy for rom_filter in *repos/libports/run/acpica.run* I get the following diagnostics.

```
[init -> drivers_reports] Warning: no policy defined for label 'rom_filter -> pci_devices'
[init -> drivers_reports] Warning: no valid policy for ROM request 'rom_filter -> pci_devices'
[init -> rom_filter] Error: ROM-session creation failed (ram_quota=6144, cap_quota=3, label="pci_devices")
[init -> rom_filter] Error: Could not open ROM session for "pci_devices"
[init -> rom_filter] Error: Uncaught exception of type 'Genode::Rom_connection::Rom_connection_failed'
[init -> rom_filter] Warning: abort called - thread: ep
```

Looks pretty explicit in my eyes or did get your case wrong?

ttcoder, 2022-10-20:
Indeed!
I hadn't used that scenario for testing due to another problem (I'll file a ticket soon re. building usb_host on a non-linux host). I circumvented that problem by commenting out the usb_host dependancy, and was able to replicate the correct behavior indeed.

However when using my own run scenario, with that same policy change, it gets stuck. I've added the app/top component to see if someone's churning out CPU cycles (gone into an infinite loop), but it seems not.

I'll close this (guess that makes sense) as I don't know what angle to use to attack this yet, but will revisit if I can collect useful info.

PS -- log:
´´´
Genode sculpt-22.10 <local changes>
677 MiB RAM and 63253 caps assigned to init
[init -> basic-layers]   0x1000000 .. 0x10ffffff: linker area
[init -> ahci_drv] --- Starting AHCI driver ---
[init -> acpi_drv] Found MADT
[init -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
[init -> acpi_drv] MADT IRQ 5 -> GSI 5 flags: 13
[init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13
[init -> acpi_drv] MADT IRQ 10 -> GSI 10 flags: 13
[init -> acpi_drv] MADT IRQ 11 -> GSI 11 flags: 13
[init -> acpi_drv] Found MCFG
[init -> acpi_drv] MCFG BASE 0xb0000000 seg 0x0 bus 0x0-0xff
[init -> acpi_drv] RSDT OEM 'BOCHS ', table id 'BXPCRSDT', revision 1, creator 'BXPC' (1)
[init -> top] sorting based on execution context (ec) [other option is scheduling context (sc)]
[init -> acpi_drv] SMBIOS table (entry point: 0x155700 structures: 0xf5720)
[init -> top] cpu=0.0   0     0  28.22%   0.00% thread='entrypoint'              label='core'
[init -> top] cpu=0.0  64 10000  17.61%   0.00% thread='ep'                      label='init'
[init -> top] cpu=0.0   0     0  15.64%   0.00% thread='pager'                   label='core'
[init -> top] cpu=0.0   0     0   5.08%   7.56% thread='root'                    label='core'
[init -> top] cpu=0.0   0     0   2.23%   0.00% thread='ep'                      label='core'
[init -> top] cpu=0.0  64 10000   2.02%  17.16% thread='init'                    label='init'
[init -> top] 
[init -> ps2_drv] Using keyboard with scan code set 1 (xlate)
[init -> ps2_drv] detected ExPS/2 mouse - activating scroll-wheel and 5-button support
[init -> basic-layers]   0x40000000 .. 0x4fffffff: stack area
[init -> basic-layers]   0x30000 .. 0x14cfff: ld.lib.so
[init -> basic-layers]   0x10e17000 .. 0x10ffffff: libc.lib.so
[init -> basic-layers]   0x10d6e000 .. 0x10e16fff: vfs.lib.so
[init -> basic-layers]   0x10d2c000 .. 0x10d6dfff: libm.lib.so
[init -> basic-layers]   0x1017000 .. 0x11d4fff: stdcxx.lib.so
[init -> basic-layers]   0x11d5000 .. 0x11ebfff: zlib.lib.so
[init -> basic-layers]   0x11ec000 .. 0x1f61fff: haiku.lib.so
[init -> basic-layers]   0x1f62000 .. 0x1f9ffff: jpeg.lib.so
[init -> basic-layers]   0x1fa0000 .. 0x2bc5fff: avcodec.lib.so
[init -> basic-layers]   0x2bc6000 .. 0x2c0bfff: avfilter.lib.so
[init -> basic-layers]   0x2c0c000 .. 0x2d71fff: avformat.lib.so
[init -> basic-layers]   0x2d72000 .. 0x2d9dfff: avutil.lib.so
[init -> basic-layers]   0x2d9e000 .. 0x2db1fff: avresample.lib.so
[init -> basic-layers]   0x2db2000 .. 0x2de4fff: swscale.lib.so
[init -> top] cpu=0.0   0     0  51.45%  51.35% thread='idle'                    label='kernel'
[init -> top] cpu=0.0   0     0  12.31%   0.00% thread='pager'                   label='core'
[init -> top] cpu=0.0  64 10000   7.18%   0.00% thread='ep'                      label='init -> basic-layers'
[init -> top] cpu=0.0   0     0   5.74%   0.00% thread='entrypoint'              label='core'
[init -> top] cpu=0.0  64 10000   3.66%   0.00% thread='ep'                      label='init -> nitpicker'
[init -> top] cpu=0.0  64 10000   2.96%   0.00% thread='ep'                      label='init -> platform_drv'
[init -> top] 
[init -> top] cpu=0.0   0     0  86.78%  86.63% thread='idle'                    label='kernel'
[init -> top] cpu=0.0  64 10000   3.78%   0.00% thread='ep'                      label='init -> timer'
[init -> top] cpu=0.0  64 10000   2.97%   0.00% thread='ep'                      label='init -> nitpicker'
[init -> top] cpu=0.0  64 10000   2.38%   6.33% thread='threaded_time_source'    label='init -> timer'
[init -> top] cpu=0.0  64 10000   1.96%   1.96% thread='signal handler'          label='init -> nitpicker'
[init -> top] cpu=0.0  64 10000   1.73%   4.74% thread='nitpicker'               label='init -> nitpicker'
[init -> top] 
[init -> top] cpu=0.0   0     0  86.58%  86.34% thread='idle'                    label='kernel'
[init -> top] cpu=0.0  64 10000   3.84%   0.00% thread='ep'                      label='init -> timer'
[init -> top] cpu=0.0  64 10000   3.02%   0.00% thread='ep'                      label='init -> nitpicker'
[init -> top] cpu=0.0  64 10000   2.44%   6.45% thread='threaded_time_source'    label='init -> timer'
[init -> top] cpu=0.0  64 10000   1.98%   1.97% thread='signal handler'          label='init -> nitpicker'
[init -> top] cpu=0.0  64 10000   1.75%   4.82% thread='nitpicker'               label='init -> nitpicker'
[init -> top] 
[init -> top] cpu=0.0   0     0  86.55%  86.38% thread='idle'                    label='kernel'
[init -> top] cpu=0.0  64 10000   3.85%   0.00% thread='ep'                      label='init -> timer'
[init -> top] cpu=0.0  64 10000   3.01%   0.00% thread='ep'                      label='init -> nitpicker'
[init -> top] cpu=0.0  64 10000   2.46%   6.48% thread='threaded_time_source'    label='init -> timer'
[init -> top] cpu=0.0  64 10000   2.01%   1.98% thread='signal handler'          label='init -> nitpicker'
[init -> top] cpu=0.0  64 10000   1.74%   4.82% thread='nitpicker'               label='init -> nitpicker'

´´´

closed 2022-10-20 by ttcoder

chelmuth, 2022-10-21:
> However when using my own run scenario, with that same policy change, it gets stuck. I've added the app/top component to see if someone's churning out CPU cycles (gone into an infinite loop), but it seems not, the CPU is 86% idle. (timer seems a little high at 6% though).
> 
> I'll close this (guess that makes sense) as I don't know what angle to use to attack this yet, but will revisit if I can collect useful info.

Unfortunately, the log shows nothing interesting to hunt down the issue. In such situations, my next step is to use a visual diff tool to compare the log and the config XML of the good and the bad case. Sometimes reducing the the scenario to a minimum (e.g., removing *basic-layers*) helps much.