genodelabs/genode #4647: GRUB loading... "incompatible license" (on real hw)

ttcoder, 2022-10-21:
This is with 22.10 but I think the new grub was introduced in 22.08 or before

Since the grub upgrade, I can no longer boot Genode on real hardware (it still works as before in Qemu though). Tried on a few TP's, similar results. Either the thinkpad gets stuck on "GRUB loading...", or that message is followed by "incompatible license - aborted - press any key to exit", or (on the T530) I get a spontaneous reset (so the computer gets into a reset - boot - reset - boot etc loop). This is even with the simple "run/log" scenario.

	make -C build/x86_64  KERNEL=nova  run/log

Which I "burn" to usb as usual, with

	dd if=log.iso of=/dev/disk/usb/0/0/raw bs=1M conv=fsync

I can successfully boot again, if I hack a reversal to the previous grub (grub2-856f9946a1482cbebc519818e3e4736bcd11e5e4 intead of the new grub2-f15e84afbb47b892ed26a5ae56f5bb038777a3c0)

I'm having trouble with the HDA driver and meant to file a ticket with a simple reproducible case, but gotta fix this boot problem before I can talk about the HDA problem :-)

Thanks as always, Team Genode!

chelmuth, 2022-10-25:
Some questions to shed some light on the scenario.

- Is this legacy or UEFI boot?
- Is secure boot enabled?
- Did you try network boot (either with run/tftp, run/iso+ipxe, run/uefi+ipxe)?
- Could you list the failing devices?
- Did you boot any device successfully (esp. more recent models like T450)?
-

ttcoder, 2022-10-25:
Some data:

- that should be legacy BIOS boot (I assume UEFI and secure-boot is mostly on recent machines)
- I'm not setup for network boot no
- Here's my hardware pool : R61, T410, T430, T530, M73

And an Asus F2A55 which I discount completely (I'm not trying to attain compatibility on it). I didn't test on the M73 yet due to not  having usb mouse support yet (still working on it), but come to think of it, I don't need kbd/mouse to test if it boots up! Should try that out.

I think you guys have a thinkpad T430 in the lab ? I see it listed in the unofficial HCL.

Interestingly, the contrib/grub2(revnumberhere)/boot/ folder is much smaller than in the previous grub package, all the .mod files have disappeared and now there are just 5 files in there! :
```
drwxrwxr-x krishna/krishna   0 2022-02-17 09:08 grub2/
-rw-rw-r-- krishna/krishna 569344 2022-03-16 09:57 grub2/grub2_64net.efi
-rw-rw-r-- krishna/krishna 790528 2022-03-16 09:57 grub2/grub2_64.efi
-rw-rw-r-- krishna/krishna 548864 2022-03-16 09:57 grub2/grub2_32.efi
-rw-rw-r-- krishna/krishna 156416 2022-03-16 09:57 grub2/embedded.img
-rw-rw-r-- krishna/krishna 157952 2022-03-16 09:57 grub2/eltorito.img
```

chelmuth, 2022-10-25:
@alex-ab can you give the iso usb a shot on x201/t430?

alex-ab, 2022-10-25:
I can reproduce the issue, iso images on X201 does not boot anymore, with the older grub2 (before 2.06) it works. I'm going to look into this.

alex-ab, 2022-10-26:
Just as note, if you don`t need actually a ISO format, but just but from USB, you may use ```--include image/disk``` instead of ```--include image/iso``` (The iso image type is the default in new generated build/arch/etc/build.conf), e.g. some more [infos](https://genode.org/documentation/release-notes/18.05#Creating_GPT-based_disk_images_by_default)

ttcoder, 2022-10-26:
That would make sense indeed, I only intend to ever boot from USB or from an HDD.

I installed the package for sgdisk as directed by the run script when trying out image/disk. My OS seems to have poor support for sgdisk (see below), not that I'm surprised given the general state of packaging in haiku-os :-). But if I can get it to work I'll retool to use it indeed. Maybe I just need to "silence" sgdisk errors so that the script can ignore them and proceed. When I look around, I see that the xorriso-based script mentions something like that too.

```

$  make -C build/x86_64  KERNEL=nova  run/log
make: Entering directory '/GenCharlie/nimble_TMP_XPND/_d_hog_gcc10/genode/build/x86_64'
including /GenCharlie/nimble_TMP_XPND/_d_hog_gcc10/genode/tool/run/power_on/qemu
including /GenCharlie/nimble_TMP_XPND/_d_hog_gcc10/genode/tool/run/log/qemu
including /GenCharlie/nimble_TMP_XPND/_d_hog_gcc10/genode/tool/run/image/disk
including /GenCharlie/nimble_TMP_XPND/_d_hog_gcc10/genode/tool/run/boot_dir/nova
including /GenCharlie/nimble_TMP_XPND/_d_hog_gcc10/genode/repos/base/run/log.run
building targets: core init test/log
genode build completed
checking configuration syntax
  CHECK init
building targets: kernel core/nova lib/ld-nova
genode build completed
using 'core-nova.a' as 'core.a'
using 'ld-nova.lib.so' as 'ld.lib.so'
Disk device is var/run/log.header
Disk device is var/run/log.header
Disk device is var/run/log.header
.......
Disk device is var/run/log.header
Disk device is var/run/log.header
The operation has completed successfully.
Warning! We seem to be running on no known platform!
Error -2147454938 when determining sector size! Setting sector size to 512
Error -2147454938 when determining sector size! Setting sector size to 512
Error -2147454938 when determining sector size! Setting sector size to 512
Warning! We seem to be running on no known platform!
Error -2147454938 when determining sector size! Setting sector size to 512
Warning! We seem to be running on no known platform!
Error -2147454938 when determining sector size! Setting sector size to 512
Error -2147454938 when determining sector size! Setting sector size to 512
........
Error -2147454938 when determining sector size! Setting sector size to 512
Error -2147454938 when determining sector size! Setting sector size to 512
Warning: Platform not recognized!
    while executing
"exec $sgdisk --delete=3 [run_dir].header"
    (procedure "run_image" line 9)
    invoked from within
"run_image"
    (procedure "run_boot_dir" line 130)
    invoked from within
.......

```

alex-ab, 2022-10-26:
882a916da68d4774d3866c44972ea466cef09ca4 fixes the issue for me, @ttcoder please give it a spin on your machines.

ttcoder, 2022-10-28:
@alex-ab Confirming, I can now make run/log, or make run/audio_out etc and it boots up on my R61, thanks!

commit 4a2def34 (2022-10-26): grub2: make ISO bootable on hardware

With "grub2: update to newer grub2 2.06 version" a regression slipped in
which makes the ISO not bootable on real hardware. The commit reverts to
the previous behaviour, to load the GRUB2 modules after boot later one by
one in the ISO format case.

Fixes #4647
commit 7a5eab95 (2022-11-17): grub2: make ISO bootable on hardware

With "grub2: update to newer grub2 2.06 version" a regression slipped in
which makes the ISO not bootable on real hardware. The commit reverts to
the previous behaviour, to load the GRUB2 modules after boot later one by
one in the ISO format case.

Fixes #4647
closed 2022-11-17 by chelmuth