genodelabs/genode #4648: [usb_host] build: missing <byteswap.h> header etc (on non-Linux host)

ttcoder, 2022-10-25:
Seems the new USB-host driver's build process is not quite self contained, it needs to have (a small amount of) headers _outside_ of the contrib/ folder, specifically, Linux headers.
This is probably of little importance for the 100-ish % of genodians who do build Genode code on Linux instead of beos/hai* like me :-). Even if the host linux and contrib folder's linux have slightly different version numbers, the headers should be compatible I guess.

But thought I'd document this in a ticket anyway, even if to close the ticket quickly after it's solved.

So...
The first error is easy to work-around.
The build script complains about "byteswap.h". That header file is not part of contrib, not in the `drwxrwxr-x krishna/krishna   0 2021-06-25 13:22 /usr/local/genode/tool/21.05/` toolchain folder nor in my own build of the toolchain, and nowhere on my system. But I simply work-around that with:

```
cd (genode-contrib-linux)/src/linux/tools/include
touch byteswap.h
```

(it seems the #include is spurious, the code does not actually need to include that header).

The next error is a little more hairy, but seems I should be able to sort out the conflict:

```
      [Linux]  /GenCharlie/nimble_TMP_XPND/_d_hog_gcc10/genode/contrib/linux-340e0ebc16b91aab3ff33d3d03611dcddd02d450/src/linux/arch/x86/tools/relocs.c:10: warning: "Elf_Phdr" redefined
      [Linux]   #define Elf_Phdr  ElfW(Phdr)
      [Linux]  /boot/system/develop/headers/os/kernel/elf.h:37: note: this is the location of the previous definition
      [Linux]    #define Elf_Addr Elf64_Addr
       [Linux]  /GenCharlie/nimble_TMP_XPND/_d_hog_gcc10/genode/contrib/linux-340e0ebc16b91aab3ff33d3d03611dcddd02d450/src/linux/arch/x86/tools/relocs.c:150:12: error: 'STT_COMMON' undeclared (first use in this function); did you mean 'SHN_COMMON'?
      [Linux]     SYM_TYPE(STT_COMMON),
```

Will keep working on it and call for help if I'm in over my head

ttcoder, 2022-11-03:
Well it's more involved than I thought, I seem to be fighting a losing battle.

I'm tempted to ask someone to email me the finished usb_host binary so that I don't have to download and install a Linux distro just to build this one driver, when everything else works well for me :-)

To wit, my hacks seem to be of increasing complexity as I make my way through the usb_host build(?) or config(?) process:

Reloc utility:
- relocs.h: replaced #include <elf.h> with #include "(genode)/depot/genodelabs/api/pc_linux/2022-10-11/src/linux/include/uapi/linux/elf.h"
- relocs.h: also commented out #include <byteswap.h>
- relocs.c: commented out everything (!) and created a dummy process_64/process_32 function
- tools/include/linux/types.h: replace #include <asm/types.h> with hand crafted typedef definitions (typedef unsigned long __u64; and so on)
- that takes me to linking stage:
- ```[Linux]    HOSTLD  arch/x86/tools/relocs```

Sorttable util:
- symbol "SHN_XINDEX" is not defined anywhere in the linux DDE or in my host OS, hence:
- sorttable.c:  I remed out those parts of the code that refer to it
- sorttable.h: same as above, with the addition of the "SHT_SYMTAB_SHNDX" symbol.
- bitops.h: commented out #include <asm/types.h>

DESCEND objtool:
- on and on it goes...

What is that stuff used for anyway, those seem to be host tools that must be generated in order to cross-build the actual linux-based driver, hmmm. Linux is an odd beast.

nfeske, 2022-11-04:
> I'm tempted to ask someone to email me the finished usb_host binary [...]

Here you go: https://depot.genode.org/genodelabs/bin/x86_64/usb_host_drv/ :-)

> To wit, my hacks seem to be of increasing complexity as I make my way through the usb_host build(?) or config(?) process:

Those hacks are a dead end. Please don't get caught up in this spiderweb.

Let me try to explain what all this complexity is all about:

The Linux USB driver code includes tons of Linux kernel headers. Not all of these headers are present in the Linux kernel's source tree. Quite a few headers are generated by the kernel's build system. Some of those depend on the Linux kernel configuration and specifics of the chosen CPU architecture.

For compiling the driver code, we need those headers. We considered two options:

1. Generating the headers manually using one known kernel configuration and hosting those generated headers in our source tree, or

3. Using the knowledge embedded in the Linux kernel's build system to generate these headers as part of our build process before we build a Linux-based driver (like the USB driver).

The first option has the advantage to avoiding the (host) dependencies from all the tools used by the Linux kernel's build system. But it inflates the source tree, and it requires high discipline on our side to update those generated headers whenever the Linux kernel configuration is changed or the Linux kernels is updated to a new version. Furthermore, the steps to update the headers need to be documented and manually followed. The latter point bears two risks. First, it makes kernel config changes rather inconvenient so that a developer may be tempted to evade the pain of such changes even in cases where such a change would objectively be the right way. Second, if our discipline slips or when making mistakes during the manual steps, the generated headers may become inconsistent with the kernel configuration, introducing subtle bugs.

In contrast, the second option keeps our source tree tidy and clean, reduces maintenance costs, avoids the need to document the manual steps (because there are no such steps), and the lowers the risk for subtle bugs. This is the road we took.

Our underlying assumption is that the host system is able to build a Linux kernel. In your case, using Haiku, this assumption seems to not hold.

I'm not quite sure what would be the best way to solve this issue. The most sustainable approach would be improving Haiku to become able to build the Linux kernel, making all the tools happy on which the kernel's build system depends. I think that's nice because it has the potential to make Haiku better as a collateral effect of solving this issue. On the other hand, this would of course be quite distracting from what you actually want to achieve (using Genode's USB host driver).

A more pragmatic (but less sustainable) way would be working around the pitfalls by patching the offending parts of the Linux build system. This would at least require you to analyze and understand the Linux build system and some of the tools invoked.

A dirty approach would be to generate the headers on a Linux system, copying them to your host, and pointing the build of the USB host driver to this copy of the headers.

ttcoder, 2022-11-04:
I had forgotten about the depot and pre-built binaries!

When Genode 22.11 is released later this month I'll whole-heartedly jump into that opportunity for sure.

The current binaries in genodelabs/bin/ are from last year (2021) so while downloading the most recent one I expected it wouldn't like the new platform_drv and pci_decode paradigm, and indeed it won't run (for another reason though):

```
Genode sculpt-22.10 <local changes>
665 MiB RAM and 63253 caps assigned to init
[init -> acpica]   0x1000000 .. 0x10ffffff: linker area
[init -> usb_host_drv] Error: LD: symbol not found: 'Genode::Slab::alloc(unsigned long, void**)'
[init -> acpica]   0x40000000 .. 0x4fffffff: stack area
[init -> acpica]   0x30000 .. 0x14cfff: ld.lib.so
[init -> usb_host_drv] Error: Uncaught exception of type 'Linker::Not_found'
[init -> usb_host_drv] Warning: abort called - thread: ep
[init -> acpi_drv] Found MADT
[init -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
[init -> acpi_drv] MADT IRQ 5 -> GSI 5 flags: 13
[init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13
```

Will keep an eye out on the mtime timestamp of that online directory, when it reaches "2022-nov" that'll be a life-saver.

P.S. I'm keeping your post bookmarked and not ruling out the approaches you detailed with such care ; I tell myself the usb_host build process just involves some pain during the native (haiku) toolchain phase-one, but after that the build presumably returns to using the Genode toolchain as usual during a second phase. So who knows if I can pull it off, work on that first phase and make my host Elf headers etc compatible with linux.

nfeske, 2022-11-04:
> When Genode 22.11 is released later this month I'll whole-heartedly jump into that opportunity for sure.

This made me realize that I pointed you to the outdated USB driver. The real (modern) one is here: https://depot.genode.org/genodelabs/bin/x86_64/pc_usb_host_drv/

> I tell myself the usb_host build process just involves some pain during the native (haiku) toolchain phase-one, but after that the build presumably returns to using the Genode toolchain as usual during a second phase.

Yup.

ttcoder, 2022-11-05:
Well I just finished testing that new USB driver on my set of guinea pigs here. Not only does it solve my problem (no need to build it, I can download it from the depot), but on top of that, it works on everything I tried ! Congrats on code well done :-). The previous 2021'ish variant I was utilizing until now had compatibility issues on a couple laptops, including my T430, forcing me to use the ps/2 touchpad.

Closing, I ought to revisit the above for my benefit, but as far as Genode Labs is concerned this ticket should no longer be part of the visible tickets.

closed 2022-11-05 by ttcoder

nfeske, 2022-11-07:
Thank you for the encouraging feedback! :-)