genodelabs/genode #4649: libc/vfs: emulate a "mute" stdin

jschlatow, 2022-10-26:
Yesterday, I was running a piece of contrib code on genode that tries to read user input from stdin as follows:

```
int number;
while (1) {
  std::cout << "Please type a number: ";
  std::cin >> number;
  switch (number) {
    /* ... */
  }
}
```

I ran this component with the _log_ and _null_ file system from the vfs as stdin since I wanted to mute the user input. In both cases, the program passes the `>>` operator and outputs a bogus number despite the fact that there is no data from stdin. The expected behaviour for `std::cin >> ...` is to wait for any user input, though. The same effect occurs if I use `fscanf()` instead of `std::cin`.

I therefore looked into the implementations. Both, _log_ and _null_ complete the read operation but do not return any data, i.e. they basically return an EOF. Although the EOF is reflected in the return value of `fscanf` I have rarely seen any code that checks the return value. It is actually possible to wrap the `std::cin >> ...` in an `if` statement to check for errors but I have never seen this in use before. Instead, it appears to be assumed that the read from stdin blocks until any data becomes available.

I think there are four scenarios when it comes to stdin emulation:
1. full emulation: This is achieved with the _terminal_ file system.
2. static input: This is achieved with an _inline_ file system.
3. empty input: This is currently implemented by the _null_ (and _log_) file system.
4. no input (blocking indefinitely)

I currently see no option for the 4th scenario. However, I tried changing the return value of the _log_ file system's `read()` method from `READ_OK` to `READ_QUEUED`, which did the trick. I'd therefore like to consider this as a permanent change to the _log_ file system. Since _log_ is naturally used as a replacement for stderr and stdout, it would imo make sense to be able to use it as (mute) stdin as well.

chelmuth, 2022-10-26:
I tested this on Linux with */dev/null* and */dev/stdout*. Our `<null>` already mimics Linux behavior and your suggested change corresponds to what I see on `strace cat < /dev/stdout`. Thus, I support your proposal.

jschlatow, 2022-10-27:
Commit 78414a2 implements the suggested change.

commit 78414a2f (2022-10-27): vfs/log: block indefinitely on read access

With this change, the log file system mimics the behaviour of
/dev/stdin.

Fixes #4649
commit b2189ae8 (2022-11-17): vfs/log: block indefinitely on read access

With this change, the log file system mimics the behaviour of
/dev/stdin.

Fixes #4649
closed 2022-11-17 by chelmuth