genodelabs/genode #4650: core: DMA allocation failes in 4GB range for Intel GPU

ssumpf, 2022-10-26:
```
Error: out of physical memory while allocating 58720256 bytes in range [0x1000-0xffffffff]
[init -> drivers -> dynamic -> intel_gpu_drv] Warning: unable to obtain DMA address from platform driver
[init -> drivers -> dynamic -> intel_gpu_drv] Error: Uncaught exception of type 'Genode::Ipc_error'
[init -> drivers -> dynamic -> intel_gpu_drv] Warning: abort called - thread: ep
```
There is still enough memory above 4GB available and usable by the GPU. This was introduced with genodelabs/genode#4335 This restriction is only required for devices that expect allocations below 4GB (e.g., UHCI).

ssumpf, 2022-10-28:
Some investigation showed that my test machine has less than 1GB of physical RAM below 4GB. This combined with relatively large DMA requests that have to be continuous (~56MB by VBOX6 guests with 3D for VRAM) + allocator fragmentation cause the problem.

chelmuth, 2022-11-15:
Addressed by e81ee57e8b6540ad1d682c4c2cd823c6fa2aa15a (somewhat)?

ssumpf, 2022-11-15:
@chelmuth: Yes. Acutally by #4665

chelmuth, 2022-11-15:
Pursued and eventually fixed in #4665.

closed 2022-11-15 by chelmuth