genodelabs/genode #4651: i.MX FEC NIC driver: remove legacy driver from dde_linux

skalk, 2022-10-27:
The deprecated lx_emul/lx_kit is still used by the FEC i.MX NIC driver. We should remove it, and replace it by an updated version using the new DDE Linux porting approach, and the recent lx_emul/lx_kit. Moreover, the new driver should reside inside the genode-imx repository.

chelmuth, 2022-10-28:
It seems there are some fixups required, for example in *arm_v7a.imx7d_sabre.hw.fetchurl_lwip*.

```
[2022-10-28 07:55:50] [init -> drivers] Warning: nic_drv: cap upgrade of Platform failed
[2022-10-28 07:55:50] [init -> drivers] child "nic_drv" requests resources: cap_quota=2
```

chelmuth, 2022-10-28:
*arm_v7a.imx7d_sabre.sel4.fetchurl_lwip*

```
[2022-10-28 07:58:16] [init -> drivers -> nic_drv] Error: cache_clean_invalidate_data not implemented for this kernel!
```

skalk, 2022-10-31:
@chelmuth thank you for the investigation. I've added commit f3ffe5d to address the missing cache_clean_invalidate_data call on sel4. It still misses cache_invalidate_data on this platform (which is not available), but also without that it, the driver should be operational.

nfeske, 2022-11-17:
Fixed in master.

closed 2022-11-17 by nfeske