genodelabs/genode #4655: IMX8 USB networking: Data corruption on socket write

ponast, 2022-10-27:
When sending packets on the IMX8mq_evk board via the net_usb_drv data corruption sometimes is observed on socket writes. Corruption typically occurs after having written several (more than three) packets in a quick sequence. If a delay of two seconds is added before every write to the socket no corruption occurs. It is always the final 5 to 16 bytes of the TCP payload that are corrupted and the bytes are from the prior packet sent. Providing additional RAM to the components has been tried but did not help. No dedicated test application that triggers the error exists. Attached are two files, an execution log and a config except from a larger system that triggers the error and that might be of help in analyzing the issue. The excerpts include relevant logs and configs from the components in the USB networking chain, i.e. platform_drv,  usb_host_drv, usb_net_drv and the component executing the libc socket writes (net_client).
[usb_net_drv_bugreport.log](https://github.com/genodelabs/genode/files/9879854/usb_net_drv_bugreport.log)
[usb_net_drv_bugreport_config.txt](https://github.com/genodelabs/genode/files/9879877/usb_net_drv_bugreport_config.txt)

### Platform
genode: sculpt-22.10,
genode-imx: current HEAD (ffddc1329c4d8b95967583ad615e843db904476a)

skalk, 2022-10-31:
I could successfully reproduce that data is send corrupted via usb_net_drv link in contrast to FEC NIC driver link. To re-produce the issue, I've adapted the lighttpd.run script to provide a 2M binary blob of random data via HTTP. The original run-script uses the FEC NIC device, and I can successfully, repeatedly download the binary blob, which has the correct SHA512 checksum. Alternativly the imx8q_evk_usb_net_lighttpd.run script uses the USB NIC (ASIX AX88179) dongle as uplink. In that case, I've observed different checksums for every download.

ponast, 2022-11-01:
@skalk Thanks for the quick confirmation of the bug. Do you have any idea about the origin of the bug (which subsystem) and how difficult it is to fix?

skalk, 2022-11-01:
@ponast not yet

skalk, 2022-11-01:
@ponast I think we found the root cause of this issue. Formerly, implementations of the USB host controllers copied data from the packet stream of USB device clients to dedicated memory. Now, we use the packet stream buffer itself to DMA from/to the USB host controller, which is fine, but the allocation for distinct packets was not cache-line aligned. Therefore, the memory preparation/post-processing (cache cleaning, invalidation) of independent DMA transfers got disturbed when they shared a cache line.
I will provide a solution tomorrow.

skalk, 2022-11-02:
@ponast commit 123f53e should solve the issue. Would you please give it a try?

skalk, 2022-11-03:
@ssumpf and @cnuke gave me the hint that there are other USB session clients left, which might share the potential problem of mis-aligned packets. Therefore, I reopen the issue, and will provide a more general solution.

skalk, 2022-11-03:
commit 0x92703 implements a more general solution for cache-line aligned USB packet allocations.

nfeske, 2022-11-07:
Merged to staging.

closed 2022-11-17 by chelmuth