genodelabs/genode #4656: No HDA audio since update

ttcoder, 2022-10-28:
With #4647 fixed (muchas gracias) I can post a couple reproducible cases for the hda regression on my set of TP laptops and ThinkCenter here.

make run/audio_out:

- no audio playback
- nothing interesting in the serial feedback (in qemu), stuck mid-way

make run/audio_player:

- no audio playback
- the serial output (when forced to run in qemu) shows a pci_audio_drv crash:

```
[init -> audio_drv] Error: Uncaught exception of type 'Genode::Ipc_error'
[init -> audio_drv] Warning: abort called - thread: ep
```

I've seen that before, not sure if it was when I misconfigured platform_drv or whatnot...

I see the "config" file between the above two scenarios have a different report_rom policy:

- platform_drv -> devices (for run/audio_out)
- platform_drv -> acpi (for run/audio_player)

Oh -- the latter does NOT yet include pci_decode, needs to be sync'ed with recent changes ? That one still refers to "legacy_pc_platform_drv" too.

Was first thinking of building pci_audio_drv with -DDEBUG but now seems like it's pci/platform related instead. Should be reproducible with the T430 etc in the lab.

alex-ab, 2022-10-28:
I think the platform session behaves nowadays differently between old and new platform driver. The driver got adjusted all to the new platform driver, so any run script with the legacy_platform_driver you should consider not working anymore.

chelmuth, 2022-10-28:
> ```
> [init -> audio_drv] Error: Uncaught exception of type 'Genode::Ipc_error'
> [init -> audio_drv] Warning: abort called - thread: ep
> ```
> 
> I've seen that before, not sure if it was when I misconfigured platform_drv or whatnot...

This is the direct result of the following.

> That one still refers to "legacy_pc_platform_drv" too.

The *audio_drv* tries to use the current Platform session (speak RPC interface) to communicate with the legacy platform driver that implements the deprecated legacy interface, which is incompatible. Thus, the client ends up in an IPC (RPC) error.

Additionally to what @alex-ab wrote: Please look out for `source ${genode_dir}/repos/base/run/platform_drv.inc` in run scripts. This means the script was not adapted and will not work as *platform_drv.inc* is deprecated in conjunction with the legacy platform driver and will be removed soon.

cnuke, 2022-11-01:
> make run/audio_out:
> 
>     * no audio playback

Could you please provide the full log of the `audio_out.run` test.

>     * nothing interesting in the serial feedback (in qemu), stuck mid-way

With the update I dropped the *eap* and *auich* driver, which leaves *azalia* as the only driver. AFAICT the Qemu HDA codec does not work with the Qemu device model, so the driver is only test on hardware.

> make run/audio_player:
> 
>     * no audio playback

My [sculpt-22.10-2022-10-13](https://github.com/cnuke/genode-world/commits/sculpt-22.10-2022-10-13) branch in world contains fixes for the `audio_player` component that are required for it to work better (there is still an issue with using `alloc()` to acquire the next *Audio_out::Packet* where it will not make any progress after the revert done in #4609).

ttcoder, 2022-11-01:
> Could you please provide the full log of the audio_out.run test.

Here goes, but it's in qemu and I now understand it's problematic, regardless of which HDA scenario I run.
That m ight be the reason why the scenario gets "stuck" after the smbios related line (no more output to the qemu window). Hence I should use qemu only to debug basic issues like missing pci_decode component or wrong IPC/RPC usage, and then switch to real hardware.

I looked at your sculpt branch, and there is indeed one revision that relates to "modified runtime", but I'm not sure they deal with the pci_decode stuff (since the audio driver does not even start).

Anyway I can certainly wait until Genode 22.11, which I guess will consolidate drivers, since currently I'm stuck with my NTFS driver code (it won't readdir() some directories for some reason), will take me some time to debug and keep me busy, away from audio matters, i.e. I'll work with "Black hole" audio for a while  :-)

***

```

Welcome to the Haiku shell.

/GenCharlie/nimble_TMP_XPND/_d_hog_gcc10/genode/build/x86_64> qemu-system-x86_64 -no-kvm -display sdl -cpu core2duo  -serial mon:stdio -m 800 -cdrom var/run/audio_out.iso -machine q35 -net nic,model=e1000,netdev=net0 -netdev user,id=net0
Running QEMU with SDL 1.2 is deprecated, and will be removed
in a future release. Please switch to SDL 2.0 instead

Bender Version 0.9-beta7-49-g5086245-dirty
Patching BDA with I/O port 0x3f8.
Bender: Hello World.
hwp config: eeo=na irq=na hwp=na epp=na epb=na

NOVA Microhypervisor v8-a34076e (x86_64): Oct 17 2022 14:59:54 [gcc 10.3.0] [MBI2]

[ 0] TSC:2618768 kHz BUS:1009500 kHz (measured)
[ 0] CORE:0:0:0 6:f:b:0 [0] Intel(R) Core(TM)2 Duo CPU     T7700  @ 2.40GHz

Hypervisor NOVA (API v8)
core     image  [0000000000100000,000000000061a000)
binaries region [0000000000287000,000000000061a000) free for reuse
detected physical memory: 0x0000000000000000 - size: 0x000000000009fc00
use      physical memory: 0x0000000000000000 - size: 0x000000000009f000
detected physical memory: 0x0000000000100000 - size: 0x0000000031edf000
use      physical memory: 0x0000000000100000 - size: 0x0000000031edf000
reserved memory: 0x31ac3000 - size: 0x51b2f0 type=-2
reserved memory: 0x9fc00 - size: 0x400 type=2
reserved memory: 0xf0000 - size: 0x10000 type=2
reserved memory: 0x31fdf000 - size: 0x21000 type=2
reserved memory: 0xb0000000 - size: 0x10000000 type=2
reserved memory: 0xfed1c000 - size: 0x4000 type=2
reserved memory: 0xfffc0000 - size: 0x40000 type=2
reserved memory: 0xfd000000 - size: 0x12c000 type=-5
reserved memory: 0x1000000 - size: 0x800000 type=-1
reserved memory: 0x1800000 - size: 0x1c00000 type=-1
reserved memory: 0x31fe20b8 - size: 0x0 type=-3
Hypervisor reports 1x1 CPU 
Warning: CPU has no invariant TSC.
mapping: affinity space -> kernel cpu id - package:core:thread
 remap (0x0) -> 0 - 0:0:0) boot cpu
ROM modules:
 ROM: [0000000031c4b000,0000000031c70bd8) acpi_drv
 ROM: [0000000031c71000,0000000031c71ba0) config
 ROM: [000000000000c000,0000000000010000) core_log
 ROM: [0000000031c72000,0000000031cda708) init
 ROM: [0000000031cdb000,0000000031dc42f0) ld.lib.so
 ROM: [0000000031dc5000,0000000031e2bd50) pci_audio_drv
 ROM: [0000000031e2c000,0000000031e48790) pci_decode
 ROM: [0000000031e49000,0000000031ebbb98) platform_drv
 ROM: [000000000000a000,000000000000b000) platform_info
 ROM: [0000000031ebc000,0000000031eddd10) report_rom
 ROM: [0000000031ede000,0000000031fad332) sample.raw
 ROM: [0000000031fae000,0000000031fc1958) test-audio_out
 ROM: [0000000031fc2000,0000000031fddb40) timer

36M kernel memory

Genode sculpt-22.10 <local changes>
757 MiB RAM and 63253 caps assigned to init
[init] parent provides
[init]   service "ROM"
[init]   service "IRQ"
[init]   service "IO_MEM"
[init]   service "PD"
[init]   service "RM"
[init]   service "CPU"
[init]   service "LOG"
[init] child "timer"
[init]   RAM quota:  776K
[init]   cap quota:  66
[init]   ELF binary: timer
[init]   priority:   0
[init]   provides service Timer
[init] child "report_rom"
[init]   RAM quota:  1800K
[init]   cap quota:  66
[init]   ELF binary: report_rom
[init]   priority:   0
[init]   provides service Report
[init]   provides service ROM
[init] child "acpi_drv"
[init]   RAM quota:  3848K
[init]   cap quota:  316
[init]   ELF binary: acpi_drv
[init]   priority:   0
[init] child "pci_decode"
[init]   RAM quota:  776K
[init]   cap quota:  316
[init]   ELF binary: pci_decode
[init]   priority:   0
[init] child "platform_drv"
[init]   RAM quota:  776K
[init]   cap quota:  66
[init]   ELF binary: platform_drv
[init]   priority:   0
[init]   provides service Platform
[init] child "audio_drv"
[init]   RAM quota:  9992K
[init]   cap quota:  116
[init]   ELF binary: pci_audio_drv
[init]   priority:   0
[init]   provides service Audio_out
[init] child "test-audio_out"
[init]   RAM quota:  3848K
[init]   cap quota:  66
[init]   ELF binary: test-audio_out
[init]   priority:   0
[init -> test-audio_out] --- Audio_out test ---
[init] child "report_rom" announces service "Report"
[init] child "report_rom" announces service "ROM"
[init] child "timer" announces service "Timer"
[init -> acpi_drv] Found MADT
[init -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
[init -> acpi_drv] MADT IRQ 5 -> GSI 5 flags: 13
[init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13
[init -> acpi_drv] MADT IRQ 10 -> GSI 10 flags: 13
[init -> audio_drv] --- probe drivers ---
[init -> acpi_drv] MADT IRQ 11 -> GSI 11 flags: 13
[init -> acpi_drv] Found MCFG
[init -> acpi_drv] MCFG BASE 0xb0000000 seg 0x0 bus 0x0-0xff
[init -> acpi_drv] RSDT OEM 'BOCHS ', table id 'BXPCRSDT', revision 1, creator 'BXPC' (1)
[init] child "platform_drv" announces service "Platform"
[init -> acpi_drv] SMBIOS table (entry point: 0x155710 structures: 0xf5730)
```

cnuke, 2022-11-04:
> Here goes, but it's in qemu and I now understand it's problematic, regardless of which HDA scenario I run. That m ight be the reason why the scenario gets "stuck" after the smbios related line (no [...]

Normally the driver should complain that it does not find any suitable codec when running in QEMU. Since I do not plan to investigate that shortcoming any further I added the check https://github.com/genodelabs/genode/blob/9079a083d28bb2d47c14684a9a56c5bfca6fcf2f/repos/dde_bsd/run/audio_out.run#L3-L6 to the run script.

> ```
>
> Welcome to the Haiku shell.
>
> /GenCharlie/nimble_TMP_XPND/_d_hog_gcc10/genode/build/x86_64> qemu-system-x86_64 -no-kvm -display sdl -cpu core2duo  -serial mon:stdio -m 800 -cdrom [...]

If this is indeed the complete QEMU commandline it seems the audio device option is missing, which could explain why it appears to get stuck.

For all intents and purposes we need the output from one of your non-working test machines to diagnose the issue any further ☺.

ttcoder, 2022-11-08:
Ok, I'm back attending to this ticket! So here's what I did:

- I reviewed e.g. c7e74ac433cde , comparing with my "distro" scenario setup (differences: I have a mixer, I use rom_filter on top of report_rom, but anyhow it seems ok)
- I made a run without debug tracing (see attached) on an old R61 which used to work well
- I made a run on a different machine (T430, used to work too), with all 3 tracing flags enabled (-DDEBUG, -DAZALIA_DEBUG, -DDIAGNOSTIC I think)

It should be reproducible with the T430 (or X201?) at your end I guess. But if needed, tell me what my next debugging steps should be and I'll take care of it :-)

[image]

[image]

ttcoder, 2022-11-09:
So,

Turns out audio-out still works indeed, with one exception : it no longer works on my old R61, but that's no big deal as I can use that test rig for reviewing other aspects of my distro (file system, etc). The other computers here get audio with `run/mixer` indeed.

The remaining trouble -- about my own apps being silent -- is in my code, which leverages libav for decoding mp3 files : handle_progress() is no longer called ! My code has not changed, it calls `output.sigh( handle_progress )` etc. Fiddling with the classes, I found out I need to "prime the pump" with a (hardcoded) call to handle_progress(), and then it starts getting called by signals from then on, at least when testing against `black_hole`. Possibly, it's because the new audio server must receive at least one submit()ted packet before it starts sending signals to the client ?

Testing against `pci_audio_drv` on my T410, handle_progress() is called a half-dozen times, submitting 2 packets each time, for a total of ca. 12 packets (so not enough to cover the length of the mp3 file, which is about 10 seconds long). I'll look into that next, might be a math error somewhere since my code was fragile in that area, even before upgrading to the newer Genode. Or likely, I need to apply https://github.com/cnuke/genode-world/commits/sculpt-22.10-2022-10-13 libav tweaks as mentionned by Josef.

So I might end up with a work-around (the new Genode audio chain needs "pump priming") and a minor regression (I'll do audio tests on one less laptop than before, no big deal), which would resolve this ticket. Will keep you posted.

ttcoder, 2022-11-15:
Ended up using the `vfs_oss` plug-in. Learned quite a bit about various parts of Genode's audio in the past few days, and realized the OSS plug-in is quite simple to use, I basically just need to make open(), close(), write() calls and that's it. Well I do have to poll @ circa 100 Hz to advance the decoding/feed/sink pipeline as that API does not give access to signals it seems, but audio works and I'm a happy camper. Might revisit this later and work on the native API (audio_out session) for optimization purposes, but with the OSS API I'm operational already.

Also had to activate audio-in (recording) otherwise the plug-in won't load ; in real-world use it means pci_audio_drv denies service to a second application coming on the tail of the first application that uses the OSS plug-in (since recording is 'exclusive'), but that should be solvable if I re-route audio-in to the mixer instead of the bare driver. Only one app is supposed to record at any given time, anyway.

I suppose that's all here, so closing.

P.S. reviewing black_hole's audio_out.h file, quickly realized there is a line that goes "if( queue.empty() ) return", i.e. it returns before sending a wake-up signal to the client indeed ; not sure if that's worth operning a new ticket for though.

closed 2022-11-15 by ttcoder

ttcoder, 2022-11-16:
> I think it is very good that you settled on the vfs_oss plugin because we will change the audio-session interfaces sooner or later

That reminds me, I suspect that part of the changes you'll introduce in the future might be related to "joining" audio channels together, maybe in the same packet (interleaved float streams) : in some places in the Genode code I see things like "synchronize p[LEFT] and p[RIGHT] in case they get out of sync" : indeed since left-channel packets are distinct -- also in their processing -- from right channels one, loss of sync seems possible, and preventing that occurence complexifies the client code a wee bit (though changing the API just to simplify that small issue,  might introduce complexity upstream?).

My experience lacks the breadth and depth to support that suggestion as the only media-related apps I developed since 1999 have dealt with audio (sans video) and it was mainly on one platform (Be-style API), so that's the sole data point I have to offer : IIRC in the Be media API, all channels (be it two as in stereo, or 5+1 channels) are in the same "BBuffer" (equivalent of a packet) so they can never get out of sync. I wonder how are things with MS's DirectX or Linux's JACK and stuff.

Re black hole, to save you guys a bit of time if you revisit this, here's a direct link to the 'submit' which it seems only get executed once the audio chain gets going (non-empty() queue):
https://github.com/genodelabs/genode/blob/9079a083d28bb2d47c14684a9a56c5bfca6fcf2f/repos/os/src/server/black_hole/audio_out.h#L81

Most of the time that is not a problem as client code (OSS etc) tends to be complex enough that it ends up calling handle-progress() as part of its initialization and thus ends up doing an initial submit() before the various callbacks and timeouts are called.