genodelabs/genode #4658: libc: build 'libc-string' library with -O3

cproc, 2022-11-01:
Building the `libc-string` library with -O3 improves the performance of functions like `memcpy()` and `memset()`, which are important, for example, for graphical updates in the Qt-based web browsers. Here are some results of the `memcpy.run` test with -O2 and -O3 in comparison:

```
PinePhone (hw) O2:

bytewise:                copied 8 GB in 23474 milliseconds (348 MiB/sec)
memcpy:                  copied 8 GB in 23464 milliseconds (349 MiB/sec)
memset:                  copied 8 GB in 2827 milliseconds (2897 MiB/sec)
libc memcpy:             copied 8 GB in 9955 milliseconds (822 MiB/sec)
libc memset:             copied 8 GB in 5310 milliseconds (1542 MiB/sec)
memcpy (page offset):    copied 8 GB in 23198 milliseconds (353 MiB/sec)
memcpy (uncached write): copied 8 GB in 23333 milliseconds (351 MiB/sec)
memcpy (uncached read):  copied 8 GB in 1726927 milliseconds (4 MiB/sec)
```
```
PinePhone (hw) O3:

bytewise:                copied 8 GB in 23462 milliseconds (349 MiB/sec)
memcpy:                  copied 8 GB in 23492 milliseconds (348 MiB/sec)
memset:                  copied 8 GB in 2829 milliseconds (2895 MiB/sec)
libc memcpy:             copied 8 GB in 9407 milliseconds (870 MiB/sec)
libc memset:             copied 8 GB in 2836 milliseconds (2888 MiB/sec)
memcpy (page offset):    copied 8 GB in 23200 milliseconds (353 MiB/sec)
memcpy (uncached write): copied 8 GB in 23337 milliseconds (351 MiB/sec)
memcpy (uncached read):  copied 8 GB in 1725877 milliseconds (4 MiB/sec)
```
```
x260 (nova) O2:

bytewise:                copied 8 GB in 3508 milliseconds (2335 MiB/sec)
memcpy:                  copied 8 GB in 1641 milliseconds (4992 MiB/sec)
memset:                  copied 8 GB in 856 milliseconds (9570 MiB/sec)
libc memcpy:             copied 8 GB in 1813 milliseconds (4518 MiB/sec)
libc memset:             copied 8 GB in 1017 milliseconds (8055 MiB/sec)
memcpy (page offset):    copied 8 GB in 1670 milliseconds (4905 MiB/sec)
memcpy (uncached write): copied 8 GB in 1589 milliseconds (5155 MiB/sec)
memcpy (uncached read):  copied 8 GB in 2987 milliseconds (2742 MiB/sec)
```
```
x260 (nova) O3:

bytewise:                copied 8 GB in 3508 milliseconds (2335 MiB/sec)
memcpy:                  copied 8 GB in 1641 milliseconds (4992 MiB/sec)
memset:                  copied 8 GB in 860 milliseconds (9525 MiB/sec)
libc memcpy:             copied 8 GB in 1572 milliseconds (5211 MiB/sec)
libc memset:             copied 8 GB in 835 milliseconds (9810 MiB/sec)
memcpy (page offset):    copied 8 GB in 1674 milliseconds (4893 MiB/sec)
memcpy (uncached write): copied 8 GB in 1587 milliseconds (5161 MiB/sec)
memcpy (uncached read):  copied 8 GB in 2987 milliseconds (2742 MiB/sec)
```
```
x260 (hw) O2:

bytewise:                copied 8 GB in 8456 milliseconds (968 MiB/sec)
memcpy:                  copied 8 GB in 2176 milliseconds (3764 MiB/sec)
memset:                  copied 8 GB in 1243 milliseconds (6590 MiB/sec)
libc memcpy:             copied 8 GB in 2886 milliseconds (2838 MiB/sec)
libc memset:             copied 8 GB in 2130 milliseconds (3846 MiB/sec)
memcpy (page offset):    copied 8 GB in 2180 milliseconds (3757 MiB/sec)
memcpy (uncached write): copied 8 GB in 2178 milliseconds (3761 MiB/sec)
memcpy (uncached read):  copied 8 GB in 7200 milliseconds (1137 MiB/sec)
```
```
x260 (hw) O3:

bytewise:                copied 8 GB in 8977 milliseconds (912 MiB/sec)
memcpy:                  copied 8 GB in 2174 milliseconds (3768 MiB/sec)
memset:                  copied 8 GB in 1238 milliseconds (6617 MiB/sec)
libc memcpy:             copied 8 GB in 1944 milliseconds (4213 MiB/sec)
libc memset:             copied 8 GB in 986 milliseconds (8308 MiB/sec)
memcpy (page offset):    copied 8 GB in 2176 milliseconds (3764 MiB/sec)
memcpy (uncached write): copied 8 GB in 2174 milliseconds (3768 MiB/sec)
memcpy (uncached read):  copied 8 GB in 7818 milliseconds (1047 MiB/sec)
```

cproc, 2022-11-01:
The uncached read test was really slow on the PinePhone and the x260 tests were slower with hw than with nova, not sure if this is to be expected. Nevertheless, building with -O3 improved the performance of the libc functions to some degree.

cproc, 2022-11-01:
Fixed by 5af88ee.

cproc, 2022-11-02:
Some more test results on the PinePhone and PINE A64 board. Even with the -O3 optimization level on Genode, libc `memcpy()` was still faster on native Linux and FreeBSD. It was also a bit faster on the A64 board than on the PinePhone in general, but for some reason noticeably faster on Linux on the A64 board while `memset()` was slower on Linux on the A64 board. The CPU frequency was set to 816 MHz on both platforms.

```
PinePhone (base-linux) O2:

bytewise:                copied 8 GB in 23655 milliseconds (346 MiB/sec)
memcpy:                  copied 8 GB in 23532 milliseconds (348 MiB/sec)
memset:                  copied 8 GB in 2863 milliseconds (2861 MiB/sec)
libc memcpy:             copied 8 GB in 9624 milliseconds (851 MiB/sec)
libc memset:             copied 8 GB in 5335 milliseconds (1535 MiB/sec)
memcpy (page offset):    copied 8 GB in 23666 milliseconds (346 MiB/sec)
memcpy (uncached write): copied 8 GB in 23605 milliseconds (347 MiB/sec)
memcpy (uncached read):  copied 8 GB in 23893 milliseconds (342 MiB/sec)
```
```
PinePhone (base-linux) O3:

bytewise:                copied 8 GB in 23575 milliseconds (347 MiB/sec)
memcpy:                  copied 8 GB in 23558 milliseconds (347 MiB/sec)
memset:                  copied 8 GB in 2854 milliseconds (2870 MiB/sec)
libc memcpy:             copied 8 GB in 8896 milliseconds (920 MiB/sec)
libc memset:             copied 8 GB in 2859 milliseconds (2865 MiB/sec)
memcpy (page offset):    copied 8 GB in 23608 milliseconds (347 MiB/sec)
memcpy (uncached write): copied 8 GB in 23688 milliseconds (345 MiB/sec)
memcpy (uncached read):  copied 8 GB in 23656 milliseconds (346 MiB/sec)
```

Results with the portable test from the `linux` subdirectory (run script in commit 58bcfa4):

```
PinePhone (Ubuntu 20.04, Linux 5.6)

bytewise memcpy test:
copied 8388608 KiB in 22358370 usecs (366 MiB/sec)
libc memcpy test:
copied 8388608 KiB in 7088170 usecs (1155 MiB/sec)
libc memset test:
copied 8388608 KiB in 2975785 usecs (2752 MiB/sec)
```
```
PINE A64 (Ubuntu 20.04, Linux 5.15)

bytewise memcpy test:
copied 8388608 KiB in 21256330 usecs (385 MiB/sec)
libc memcpy test:
copied 8388608 KiB in 4121211 usecs (1987 MiB/sec)
libc memset test:
copied 8388608 KiB in 4160345 usecs (1969 MiB/sec)
```
```
PINE A64 (FreeBSD 13.1)

bytewise memcpy test:
copied 8388608 KiB in 23228630 usecs (352 MiB/sec)
libc memcpy test:
copied 8388608 KiB in 6191592 usecs (1323 MiB/sec)
libc memset test:
copied 8388608 KiB in 2972678 usecs (2755 MiB/sec)
```
```
PinePhone (base-linux) O2:

[init -> test-libc_memcpy] bytewise memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 23510000 usecs (348 MiB/sec)
[init -> test-libc_memcpy] libc memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 9549000 usecs (857 MiB/sec)
[init -> test-libc_memcpy] libc memset test:
[init -> test-libc_memcpy] copied 8388608 KiB in 5356000 usecs (1529 MiB/sec)
```
```
PinePhone (base-linux) O3:

[init -> test-libc_memcpy] bytewise memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 23662000 usecs (346 MiB/sec)
[init -> test-libc_memcpy] libc memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 8523000 usecs (961 MiB/sec)
[init -> test-libc_memcpy] libc memset test:
[init -> test-libc_memcpy] copied 8388608 KiB in 2890000 usecs (2834 MiB/sec))
```
```
PINE A64 (base-linux) O2:

[init -> test-libc_memcpy] bytewise memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 23537000 usecs (348 MiB/sec)
[init -> test-libc_memcpy] libc memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 9484000 usecs (863 MiB/sec)
[init -> test-libc_memcpy] libc memset test:
[init -> test-libc_memcpy] copied 8388608 KiB in 5381000 usecs (1522 MiB/sec)
```
```
PINE A64 (base-linux) O3:

[init -> test-libc_memcpy] bytewise memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 23516000 usecs (348 MiB/sec)
[init -> test-libc_memcpy] libc memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 8031000 usecs (1020 MiB/sec)
[init -> test-libc_memcpy] libc memset test:
[init -> test-libc_memcpy] copied 8388608 KiB in 4014000 usecs (2040 MiB/sec)
```
```
PinePhone (hw) O2:

[init -> test-libc_memcpy] bytewise memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 23177000 usecs (353 MiB/sec)
[init -> test-libc_memcpy] libc memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 10073000 usecs (813 MiB/sec)
[init -> test-libc_memcpy] libc memset test:
[init -> test-libc_memcpy] copied 8388608 KiB in 5301000 usecs (1545 MiB/sec)
```
```
PinePhone (hw) O3:

[init -> test-libc_memcpy] bytewise memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 23409000 usecs (349 MiB/sec)
[init -> test-libc_memcpy] libc memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 9513000 usecs (861 MiB/sec)
[init -> test-libc_memcpy] libc memset test:
[init -> test-libc_memcpy] copied 8388608 KiB in 2816000 usecs (2909 MiB/sec)
```
```
PINE A64 (hw) O2:

[init -> test-libc_memcpy] bytewise memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 23331000 usecs (351 MiB/sec)
[init -> test-libc_memcpy] libc memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 8963000 usecs (913 MiB/sec)
[init -> test-libc_memcpy] libc memset test:
[init -> test-libc_memcpy] copied 8388608 KiB in 5306000 usecs (1543 MiB/sec)
```
```
PINE A64 (hw) O3:

[init -> test-libc_memcpy] bytewise memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 23320000 usecs (351 MiB/sec)
[init -> test-libc_memcpy] libc memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 7437000 usecs (1101 MiB/sec)
[init -> test-libc_memcpy] libc memset test:
[init -> test-libc_memcpy] copied 8388608 KiB in 2817000 usecs (2908 MiB/sec)
```

chelmuth, 2022-11-02:
> The [...] x260 tests were slower with hw than with nova, not sure if this is to be expected.

I'd not expect any difference in results for the benchmark. Could this be temporary effects? :thinking: How many iterations of the scenario did you run?

cproc, 2022-11-02:
> > The [...] x260 tests were slower with hw than with nova, not sure if this is to be expected.
> 
> I'd not expect any difference in results for the benchmark. Could this be temporary effects? How many iterations of the scenario did you run?

When I let the component repeat the tests in a loop the values don't change much:

```

x260 (hw) O2:

[init -> test-libc_memcpy] bytewise memcpy test:                                                                                                             
[init -> test-libc_memcpy] copied 8388608 KiB in 9071000 usecs (903 MiB/sec)
[init -> test-libc_memcpy] libc memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 2924000 usecs (2801 MiB/sec)
[init -> test-libc_memcpy] libc memset test:
[init -> test-libc_memcpy] copied 8388608 KiB in 2107000 usecs (3887 MiB/sec)
[init -> test-libc_memcpy] bytewise memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 9074000 usecs (902 MiB/sec)
[init -> test-libc_memcpy] libc memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 2929000 usecs (2796 MiB/sec)
[init -> test-libc_memcpy] libc memset test:
[init -> test-libc_memcpy] copied 8388608 KiB in 2107000 usecs (3887 MiB/sec)
[init -> test-libc_memcpy] bytewise memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 9035000 usecs (906 MiB/sec)
[init -> test-libc_memcpy] libc memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 2921000 usecs (2804 MiB/sec)
[init -> test-libc_memcpy] libc memset test:
[init -> test-libc_memcpy] copied 8388608 KiB in 2107000 usecs (3887 MiB/sec)
```

nfeske, 2022-11-02:
Regarding the difference between base-hw and NOVA, here is my guess: The memcpy test accesses a buffer of 8 MiB, which dictates the working set of virtual memory. On base-hw the RAM consumed for page tables is capped per PD. (in contrast, NOVA allocates the page tables from a global allocator) The base-hw kernel flushes all mappings (and frees the page tables) whenever the PD-specific RAM is exhausted so that new mappings can be established. So once, the program's virtual-memory footprint exceeds a certain threshold, we see thrashing effects. This may be confirmed by an instrumentation in the base-hw variant of core.

cproc, 2022-11-02:
I added some instrumentation and there are some flushes in the kernel, but they seem to be only related to `malloc()` and `free()`, no `Out_of_tables` exception:

```
[init -> test-libc_memcpy] memcpy_test()                                                                                                                     
[init -> test-libc_memcpy] memcpy_test(): calling malloc()                                                                                                   
Kernel: Hw::Address_space::flush(): virt: 0xffffffd0003d6000, size: 8392704                                                                                  
Kernel: Hw::Address_space::flush(): virt: 0xffffffd0003d6000, size: 8392704                                                                                  
[init -> test-libc_memcpy] memcpy_test(): malloc() done                                                                                                      
[init -> test-libc_memcpy] memcpy_test(): initializing buffers with memset()                                                                                 
[init -> test-libc_memcpy] memcpy_test(): initialization with memset() done
[init -> test-libc_memcpy] bytewise memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 9069000 usecs (903 MiB/sec)
[init -> test-libc_memcpy] memcpy_test(): calling free()
Kernel: Hw::Address_space::flush(): virt: 0x11000000, size: 8392704
Kernel: Hw::Address_space::flush(): virt: 0x12000000, size: 8392704
[init -> test-libc_memcpy] memcpy_test(): free() done
[init -> test-libc_memcpy] memcpy_test() finished

[init -> test-libc_memcpy] memcpy_test()
[init -> test-libc_memcpy] memcpy_test(): calling malloc()
Kernel: Hw::Address_space::flush(): virt: 0xffffffd0003d6000, size: 8392704
Kernel: Hw::Address_space::flush(): virt: 0xffffffd0003d6000, size: 8392704
[init -> test-libc_memcpy] memcpy_test(): malloc() done
[init -> test-libc_memcpy] memcpy_test(): initializing buffers with memset()
[init -> test-libc_memcpy] memcpy_test(): initialization with memset() done
[init -> test-libc_memcpy] libc memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 2921000 usecs (2804 MiB/sec)
[init -> test-libc_memcpy] memcpy_test(): calling free()
Kernel: Hw::Address_space::flush(): virt: 0x11000000, size: 8392704
Kernel: Hw::Address_space::flush(): virt: 0x12000000, size: 8392704
[init -> test-libc_memcpy] memcpy_test(): free() done
[init -> test-libc_memcpy] memcpy_test() finished

[init -> test-libc_memcpy] memcpy_test()
[init -> test-libc_memcpy] memcpy_test(): calling malloc()
Kernel: Hw::Address_space::flush(): virt: 0xffffffd0003d6000, size: 8392704
[init -> test-libc_memcpy] memcpy_test(): malloc() done
[init -> test-libc_memcpy] memcpy_test(): initializing buffers with memset()
[init -> test-libc_memcpy] memcpy_test(): initialization with memset() done
[init -> test-libc_memcpy] libc memset test:
[init -> test-libc_memcpy] copied 8388608 KiB in 2105000 usecs (3891 MiB/sec)
[init -> test-libc_memcpy] memcpy_test(): calling free()
Kernel: Hw::Address_space::flush(): virt: 0x11000000, size: 8392704
[init -> test-libc_memcpy] memcpy_test(): free() done
[init -> test-libc_memcpy] memcpy_test() finished
```

nfeske, 2022-11-02:
That's all we need to know. The test is definitely influenced by the page-fault handling. I wonder, how does it change with commit https://github.com/nfeske/genode/commit/0a9076e102ae292234eff7bd8d625509d26364ed?

cproc, 2022-11-02:
> That's all we need to know. The test is definitely influenced by the page-fault handling. I wonder, how does it change with commit [nfeske@0a9076e](https://github.com/nfeske/genode/commit/0a9076e102ae292234eff7bd8d625509d26364ed)?

Unfortunately it doesn't change with the commit. I added more instrumentation around the flushes and they are related to unmapping of dataspaces, either from core in `_clear_ds()` or from the test component during `detach()`. Before the measurement of the copy operations the allocated memory gets cleared with `memset()` in the test component and there are no flush messages in that part, so in theory everything should be mapped before the measurement starts.

```
[init -> test-libc_memcpy] memcpy_test()
[init -> test-libc_memcpy] memcpy_test(): calling malloc()
[init -> test-libc_memcpy] Pd_session_client::try_alloc(): size: 8392704
Ram_dataspace_factory::_clear_ds()
Ram_dataspace_factory::_clear_ds(): calling map_local()
Ram_dataspace_factory::_clear_ds(): map_local() returned
Ram_dataspace_factory::_clear_ds(): calling unmap_local()
Kernel: Hw::Address_space::flush(): virt: 0xffffffd0041d7000, size: 8392704
Ram_dataspace_factory::_clear_ds(): unmap_local() returned
Ram_dataspace_factory::_clear_ds() finished
[init -> test-libc_memcpy] Pd_session_client::try_alloc() finished
[init -> test-libc_memcpy] Region_map_client::attach()
[init -> test-libc_memcpy] Region_map_client::attach() finished: addr: 0x11000000
[init -> test-libc_memcpy] Pd_session_client::try_alloc(): size: 8392704
Ram_dataspace_factory::_clear_ds()
Ram_dataspace_factory::_clear_ds(): calling map_local()
Ram_dataspace_factory::_clear_ds(): map_local() returned
Ram_dataspace_factory::_clear_ds(): calling unmap_local()
Kernel: Hw::Address_space::flush(): virt: 0xffffffd0041d7000, size: 8392704
Ram_dataspace_factory::_clear_ds(): unmap_local() returned
Ram_dataspace_factory::_clear_ds() finished
[init -> test-libc_memcpy] Pd_session_client::try_alloc() finished
[init -> test-libc_memcpy] Region_map_client::attach()
[init -> test-libc_memcpy] Region_map_client::attach() finished: addr: 0x12000000
[init -> test-libc_memcpy] memcpy_test(): malloc() done
[init -> test-libc_memcpy] memcpy_test(): initializing buffers with memset()
[init -> test-libc_memcpy] memcpy_test(): initialization with memset() done
[init -> test-libc_memcpy] bytewise memcpy test
[init -> test-libc_memcpy] copied 8388608 KiB in 9084000 usecs (901 MiB/sec)
[init -> test-libc_memcpy] memcpy_test(): calling free()
[init -> test-libc_memcpy] Region_map_client::detach(): local_addr: 0x11000000
Kernel: Hw::Address_space::flush(): virt: 0x11000000, size: 8392704
[init -> test-libc_memcpy] Region_map_client::detach() finished
[init -> test-libc_memcpy] Region_map_client::detach(): local_addr: 0x12000000
Kernel: Hw::Address_space::flush(): virt: 0x12000000, size: 8392704
[init -> test-libc_memcpy] Region_map_client::detach() finished
[init -> test-libc_memcpy] memcpy_test(): free() done
[init -> test-libc_memcpy] memcpy_test() finished
```

chelmuth, 2022-11-02:
I ran the tests on our nightly x86_64 test machine (macho: i7-4770)  just now and got the same results for hw and nova like follows.

**hw**
```
bytewise:                copied 8 GB in 3113 milliseconds (2631 MiB/sec)
memcpy:                  copied 8 GB in 1557 milliseconds (5261 MiB/sec)
memset:                  copied 8 GB in 411 milliseconds (19931 MiB/sec)
libc memcpy:             copied 8 GB in 1602 milliseconds (5113 MiB/sec)
libc memset:             copied 8 GB in 750 milliseconds (10922 MiB/sec)
memcpy (page offset):    copied 8 GB in 1490 milliseconds (5497 MiB/sec)
memcpy (uncached write): copied 8 GB in 1552 milliseconds (5278 MiB/sec)
memcpy (uncached read):  copied 8 GB in 2445 milliseconds (3350 MiB/sec)
```

**nova**

```
[ 0] TSC:3400000 kHz BUS:0 kHz DL
[ 0] CORE:0:0:0 6:3c:3:1 [19] Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz
bytewise:                copied 8 GB in 3081 milliseconds (2658 MiB/sec)
memcpy:                  copied 8 GB in 1540 milliseconds (5319 MiB/sec)
memset:                  copied 8 GB in 401 milliseconds (20428 MiB/sec)
libc memcpy:             copied 8 GB in 1587 milliseconds (5161 MiB/sec)
libc memset:             copied 8 GB in 762 milliseconds (10750 MiB/sec)
memcpy (page offset):    copied 8 GB in 1472 milliseconds (5565 MiB/sec)
memcpy (uncached write): copied 8 GB in 1537 milliseconds (5329 MiB/sec)
memcpy (uncached read):  copied 8 GB in 2446 milliseconds (3349 MiB/sec)

```

cproc, 2022-11-02:
When I disable the SpeedStep feature I get similar results on nova and hw, too:

```
x260 (nova, no SpeedStep) O2:

[init -> test-libc_memcpy] bytewise memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 4207000 usecs (1947 MiB/sec)
[init -> test-libc_memcpy] libc memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 1763000 usecs (4646 MiB/sec)
[init -> test-libc_memcpy] libc memset test:
[init -> test-libc_memcpy] copied 8388608 KiB in 1098000 usecs (7460 MiB/sec)
```
```
x260 (hw, no SpeedStep) O2:

[init -> test-libc_memcpy] bytewise memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 4224000 usecs (1939 MiB/sec)
[init -> test-libc_memcpy] libc memcpy test:
[init -> test-libc_memcpy] copied 8388608 KiB in 1795000 usecs (4563 MiB/sec)
[init -> test-libc_memcpy] libc memset test:
[init -> test-libc_memcpy] copied 8388608 KiB in 1099000 usecs (7454 MiB/sec)
```

cnuke, 2022-11-03:
FWIW, the `intel_hwp` bender plugin is currently only used on `NOVA`, which could explain your findings.

commit 03352d44 (2022-11-14): libc: build 'libc-string' library with -O3

Fixes #4658
closed 2022-11-17 by chelmuth