genodelabs/genode #4663: enable Intel AX211 wifi

atopia, 2022-11-07:
Enable the AX211 intel wifi card as found e.g. in the Lenovo Thinkpad X1 Nano G2.

atopia, 2022-11-07:
@cnuke could you please add the files referenced in [d5368e5](https://github.com/atopia/genode/commit/d5368e5a03617263cadec8253ad8e755162341be) (`iwlwifi-so-a0-gf-a0-64.ucode` and `iwlwifi-so-a0-gf-a0.pnvm`) to your dde_linux_firmware repo and somehow communicate the new hashes so I can add them to the commit?

atopia, 2022-11-07:
I'm not sure if the error message when looking up `iwl-debug-yoyo.bin` introduced by [6dda99e](atopia/genode/commit/6dda99e7af2b7eadbef3e8db0034f910bd853131) is acceptable, let me know if `firmware_request_nowarn()` shouldn't cause any error messages in the driver runtime either.

SunriseJungle, 2022-11-08:
Hello sir I’m sorry to bother you and I know this is not really a communication line. But I wonder, when will Genode be ready for most computers?

cnuke, 2022-11-08:
@atopia There you go: https://github.com/cnuke/dde_linux_firmware/commit/cbd51e5ee647ec834c30a3987225cae2341753b0.

> I'm not sure if the error message when looking up `iwl-debug-yoyo.bin` introduced by [6dda99e](atopia/genode/commit/6dda99e7af2b7eadbef3e8db0034f910bd853131) is acceptable, let me know if `firmware_request_nowarn()` shouldn't cause any error messages in the driver runtime either.

If `iwl-debug-yoyo.bin` is not needed and keeping the previous dummy implementation does not introduce any other issues at run time I would prefer to keep it that way.

atopia, 2022-11-08:
Thanks for adding the firmware!
The problem is that the PNVM file is loaded using `firmware_request_nowarn()`. Now, I added the PNVM blob before discovering the IRQ problem, meaning I haven't checked if the driver works with the IRQ workaround but without the PNVM file. The way I see it there is a set of options:

1. While omitting the PNVM file will cause its own set of warnings in the driver, the card might work without it, although my understanding is that down the road this may cause a reduced set of functionality related to regulation  such as available frequencies for wifi 6.

2. Add a special-case in `firmware_request_nowarn()` to restore the old behavior for `iwl-debug-yoyo.bin`. A bit ugly but certainly straightforward.

3. Factor out common functionality for `request_firmware()` and `firmware_request_nowarn()` and pass up an argument to `lx_emul_request_firmware_nowait()` to suppress the warning. I think this is the cleanest option and did consider this solution but then first wanted to hear if it was considered over engineered just to avoid the (correct) error message `Error: firmware 'iwl-debug-yoyo.bin' is not in the firmware white list`.

4. Leave the code as-is and accept the error message.

I'm happy to do 3. if you don't consider it overkill, although the other options would work for me too.

cnuke, 2022-11-10:
@atopia I see, thanks for the clarification. In this case I agree and option *3* would be my preferred one as well (although *2* is tempting ;-)). From a practical point-of-view the intention is to encountering the `Error` only when a mandatory FW image is missing.

I fear there will be more ported drivers in the future that require firmware blobs and from that perspective spending time on the `lx_emul` interface and tweaking the implementation is fine (in the long-term it would be better to get rid of the hardcoded list but that's for another day).

chelmuth, 2022-11-10:
> I fear there will be more ported drivers in the future that require firmware blobs and from that perspective spending time on the `lx_emul` interface and tweaking the implementation is fine (in the long-term it would be better to get rid of the hardcoded list but that's for another day).

Looking at the changes in @atopia's commit this step seems long overdue (even more another day). What speaks against putting the firmware into a tar archive that's part of the wifi_drv package, request this as a single ROM in a local vfs server, and use the specific firmware ROMs via fs_rom? The hardcoded list in *lib/wifi* should also become a part of the configuration as it does not belong into the binary and needlessly complicates testing new platforms.

cnuke, 2022-11-11:
> What speaks against putting the firmware into a tar archive that's part of the wifi_drv package, request this as a single ROM in a local vfs server, and use the specific firmware ROMs via fs_rom? [...]

Nothing as this is basically the approach that was briefly discussed to address this shortcoming (without featuring an extra  `vfs` and `fs_rom` and omitting the firmware list but deploying an `firmware.tar` nonetheless though). But either solution is fine with me.

atopia, 2022-11-15:
@chelmuth @cnuke as (safe for the last one) the commits in this series are orthogonal to this discussion, can we open a new issue for this and - if there are no outstanding issues to address - merge the series into staging?

chelmuth, 2022-11-15:
Merged the series to staging.

cnuke, 2022-11-16:
Please merge [b1fa90c](https://github.com/cnuke/genode/commit/b1fa90cacd90d208bff4b067d467a05da5f0c9f0).

chelmuth, 2022-11-17:
> Merged the commit to staging but still wonder if _lib/wifi_firmware_ could be dropped in favor of _raw/wifi_firmware_, which is sufficient.

Any opinions?

chelmuth, 2022-11-17:
Beside the note above the issue is fixed in master.

closed 2022-11-17 by chelmuth