genodelabs/genode #4665: platform_drv: allow n:m mappings in device pd

ssumpf, 2022-11-10:
Some newer devices have little physical RAM below 4GB. Therefore, the current approach to allocate DMA buffers in core for "managing_system" below 4GB can lead to DMA memory exhaustion. If we allow DMA allocations above 4GB, map physical RAM not 1 to 1 within the platform driver's device PD if an IOMMU is present, while still mapping below 4GB there, we can minimize this error. Note: The below 4GB mapping is necessary for devices that support 32Bit DMA addresses only.

ssumpf, 2022-11-11:
@skalk: Could you have a look into the commits above (especially the platform driver related) and let me know if you approve this approach and what to change?

ssumpf, 2022-11-14:
After staff meeting, we decided to access the iommu information from the platform driver directly again.

ssumpf, 2022-11-14:
@skalk, @nfeske, @chelmuth: [f995aa0](https://github.com/ssumpf/genode/commit/f995aa028d95078a7f57b17a12e05b847945f6e1), [1eb2a8a](https://github.com/ssumpf/genode/commit/1eb2a8a71a4f6b93e5439fb7c31cb9c020eee6c3), and [5c96383](https://github.com/ssumpf/genode/commit/5c963839a5dbca50913fda3e68b141872255953d) implement the feature as discussed.

chelmuth, 2022-11-15:
Thanks, merged the series.

alex-ab, 2022-11-21:
The vesa driver on hardware does not work anymore. A lot of DMA faults happen and garbage is on screen. Reverting commit "platfrom_drv: map DMA memory non-natural when iommu is present" make the driver working again.

alex-ab, 2022-11-22:
c8402cbf21e91cbbfd17c588207ce03d94b62d4e makes the 1:1 mapping working (when requested for reserved memory) and let the vesa driver work again.

chelmuth, 2022-11-29:
Fixed in master.

closed 2022-11-29 by chelmuth