genodelabs/genode #4666: virtualbox6: bump version from 6.1.26 to 6.1.40

ssumpf, 2022-11-10:
This is a side effect of testing Ubuntu 22.04.

ssumpf, 2022-11-10:
FYI: [0ec6d38](https://github.com/ssumpf/genode/commit/0ec6d38a00ed08cdac0a67261896d55ed8f6030e) is not well tested. [6be14d2](https://github.com/ssumpf/genode/commit/6be14d2ddc0a461caf23dfa1e7f1240cd0ddd5b3) forwards errors during machine power up of VirtualBox6 to Genode's console.

ssumpf, 2022-11-17:
@chelmuth: [9c30ac2](https://github.com/ssumpf/genode/commit/9c30ac2166298bccd2e7f31144e55b60b760d987) marks a workaround for the clipboard problems described above. I am very open to any more insights of why this is happening.

chelmuth, 2022-11-17:
Oh, I forgot to mention `AssertPtrReturn(expr, error)` becomes just `if (!expr) return error` in case `RT_STRICT` is not enabled.

chelmuth, 2023-02-09:
> FYI: [0ec6d38](https://github.com/ssumpf/genode/commit/0ec6d38a00ed08cdac0a67261896d55ed8f6030e) is not well tested.

Are you using the commit yourself for the last weeks? Is it ready for a merge to staging?

ssumpf, 2023-02-09:
> > FYI: [0ec6d38](https://github.com/ssumpf/genode/commit/0ec6d38a00ed08cdac0a67261896d55ed8f6030e) is not well tested.
> 
> Are you using the commit yourself for the last weeks? Is it ready for a merge to staging?

No, hasn't made into my Sculpt 22.10.

chelmuth, 2023-05-03:
Could we finalize the bump until 23.05 is released?

ssumpf, 2023-05-15:
@chelmuth: [ba214db](https://github.com/ssumpf/genode/commit/ba214db577aead1db4f08189e3b183c172d3f951) bumps the version to 6.1.44,is rebased to staging, and tested.

chelmuth, 2023-05-31:
Version update was merged into Genode 23.05 :tada:

closed 2023-05-31 by chelmuth