genodelabs/genode #4667: platform_drv: remove 'alloc_dma_buffer' from connection

ssumpf, 2022-11-10:
`alloc_dma_buffer` does perform an auto cap/ram upgrade within the `connection.h` file, which in turn calls the `client.h` function. This makes it impossible for clients to handle resource upgrades themselves.

nfeske, 2022-11-10:
> This makes it impossible for clients to handle resource upgrades themselves.

I wonder, wouldn't an explicit call of `platform.Client::alloc_dma_buffer(...)` do the trick in cases where the automatic upgrade stands in the way?

For general use by most drivers (which are not resource multiplexers), the transparent upgrade is a useful default.

ssumpf, 2022-11-10:
@nfeske: I think the way to go is to make `Platform::Dma_buffer` work as intended and let every driver use this implementation. Right now the `retry_with_upgrade` there is a nop.

nfeske, 2022-11-10:
I agree, using the `Platform::Dma_buffer` is certainly the best way to go when going forward. So In principle that would allow us to cut back the convenience of the plain `Platform::Connection`. That said, when grepping the source tree for calls of `alloc_dma_buffer`, there are many drivers that would call for adjustment.

> Right now the `retry_with_upgrade` there is a nop.

I guess you lost me here. ;-) What does that mean?

It would be nice if you give the explicit `Client::` prefix a try. Just curious.

ssumpf, 2022-11-10:
> I agree, using the `Platform::Dma_buffer` is certainly the best way to go when going forward. So In principle that would allow us to cut back the convenience of the plain `Platform::Connection`. [...]

It means `retry_with_upgrade` in `dma_buffer.h` calls `alloc_dma_buffer` in `connection.h` of the platform session, which in turn calls `retry_with_upgrade`.
> 
> It would be nice if you give the explicit `Client::` prefix a try. Just curious.

From my side there is no need for it, we can add it, but it will not be called. I have adjusted the drivers using it to use `Dma_buffer` and the the legacy ones (think dde_linux < 5.14) to call `retry_with_upgrade`. Hence the issue. So, please wait for the commits to inspect yourselves and then let me know.

ssumpf, 2022-11-11:
@nfeske: Could you have a look at the commits above? In case you still want to keep `alloc_dma_buffer` within the `connection.h` after, I can reintroduce it with little effort and at `Client::alloc_dma_buffer` at the relevant places (`dma_buffer.h` and gpu/intel).

nfeske, 2022-11-11:
Thanks to your commits and your latest comment, the clouds are lifting.

I _very_ much appreciate your transitioning to the consistent use of `Dma_buffer` (https://github.com/ssumpf/genode/commit/a83ffd89a7bbb213f9032c4791a0a2d102aca55f, https://github.com/ssumpf/genode/commit/b7312b77d0ade3c893ff6dc177e9cd6de2ee429a). It leads to much nicer (and more robust) driver code. Thank you!

Regarding https://github.com/ssumpf/genode/commit/12ed6884c740e13c19f38229a0ec946118be852a, I'd think we should keep the current version because the automatic upgrade is consistent with the other connection classes (e.g., taking a look at https://github.com/genodelabs/genode/blob/master/repos/os/include/file_system_session/connection.h or https://github.com/genodelabs/genode/blob/master/repos/base/include/cpu_session/connection.h). It is a useful and established pattern.

I think the doubled gemoppelted `retry_with_upgrade` (in both `Dma_buffer` and `Platform::Connection`) is certainly not good. (even though it is not an actual problem) To avoid the redundancy, I'm for removing the `retry_with_upgrade` from the `Dma_buffer`.

ssumpf, 2022-11-11:
@nfeske: Ok, will do.

ssumpf, 2022-11-11:
@nfeske: The commits above apply your suggestions and can be found at my https://github.com/ssumpf/genode/tree/issue_gpu12_vbox_11-10 branch which also includes #4664, #4665, and #4666.

nfeske, 2022-11-11:
Thank you very much! I merged the commits to staging.

To help me keeping track of the remaining commits of your branch, could you rebase your topic branch to the new staging?

nfeske, 2022-11-17:
Fixed in master.

closed 2022-11-17 by nfeske