genodelabs/genode #4669: x86: add basic support for ACPI suspend and resume

alex-ab, 2022-11-14:


alex-ab, 2022-11-17:
The commits are ready for review and testing.

alex-ab, 2022-11-18:
I added the commits d47c861f2dbbef6b9f89ad62007e8647d3a7d3ea and 172efca94583ae5967f494aab8de6fda3fa0b64d addressing you suggestions.

alex-ab, 2022-11-18:
> Just out of curiosity, is there a limit of how many syscalls NOVA can provide, which prompted you to group different features under a single (misc) umbrella? I'm not complaining, just being interested in the line of thinking.

The last time I tried to add another syscall it did not work instantly, but I did not investigate.

nfeske, 2022-11-18:
Thanks a lot for the quick response! I merged the commit series (with the fixups squashed) to staging.

chelmuth, 2022-12-07:
I discovered strange behavior of my VM with the changes in https://github.com/alex-ab/NOVA/commit/0a54bd120ee15672089a0fcacef3b02cc0db523b#diff-376cfce0fe4e0376e8f455b1111da1205d047831ac7cdc2eb6b0e016115901c7R47 Fast double mouse clicks lead to broken mouse motion, which could be fixed by changing virtual desktops inside the VM only. I tested with Xubuntu and i3 window manager.

I bisected the issue to the NOVA commit above and also suspect the initial write to TSC is not quite correct resulting in out-of-sync TSCs. The original implementation ensured here that all ECs (CPUs?) reached the synchronization loop before writing `0` to the TSC. When I disable the TSC write like in commit 3b92051208c27320de19f0c26ec9e6a019ab7edf the strange behavior disappears.

alex-ab, 2022-12-08:
Thanks for the bisecting and pinpointing to the actual reason. May you please try https://github.com/alex-ab/NOVA/commit/7dbce0657f337ffba6130fc91ff9881bfc7b22cd, which should resolve the issue.

chelmuth, 2022-12-08:
Great, with commit [112f041](https://github.com/chelmuth/genode/commit/112f041ba79023977de75cd64f7e6e93f15a19f8) the issue disappears.

alex-ab, 2022-12-08:
You can merge, the offending commit got already mastered with Genode 22.11 as r10 nova branch.

alex-ab, 2023-01-25:
The issue_4669 contains the poc commits to get suspend/resume working for base-hw/x86. I structured the commits in two parts, mainly for review reasons. The first part of commits consists of extensions of interfaces, which effect all platforms (with empty implementations), the second part then adds the actual x86 functionality.

skalk, 2023-02-23:
@alex-ab Apart from the two minor suggestions I've commented inline your commit series, I'm absolutely happy with your commit series! Great how the halt/suspend job now fits into the overall design. Thank you very much for your rework!

skalk, 2023-02-23:
@chelmuth from my point of view, @alex-ab commits can be merged to staging, as soon as he gives a sign.

alex-ab, 2023-02-23:
@skalk: thanks, i added a fixup addressing the comments on my merge_to_staging

alex-ab, 2023-02-23:
I added commits for genode-rpi, genode-allwinner and genode-zynq on merge_to_staging branches to avoid compilation/linking failures as soon as the base-hw commits enter staging.

alex-ab, 2023-02-24:
I added a rpi3 fixup commit to merge_to_staging, which adds a missing method which caused tonight trouble.

alex-ab, 2023-12-11:
The basic support for suspend/resume was added to the kernels NOVA with Genode release 22.11 and hw with Genode release 23.02. Further works should be handled in separated issues per driver, eg. intel GPU, ahci/nvme, USB, etc.

closed 2023-12-11 by alex-ab