genodelabs/genode #4682: pc_usb_host_drv: unresponsive devices due to dynamic use in combination with usb_reset_configuration()

chelmuth, 2022-11-17:
In fairly dynamic USB-passthrough cases the use of `usb_reset_configuration()` in *repos/dde_linux/src/lib/lx_emul/usb.c* is not sufficient to ensure responsiveness of the USB device for further use. A full `usb_reset_device()` on the other hand solidly resets the device in all tested scenarios. As documented in #4566 `usb_reset_device()` has the drawback of significant delay until it finishes.

The test case for this issue switched the use of the Fujitsu S938 internal smartcard reader between vbox6 guests and the host system.

(see also https://github.com/genodelabs/genode/commit/4612ea3a8f0bb40935b7e68dad5e88bddab43241#r89512816)

Closely linked to issue #4566.

chelmuth, 2023-09-28:
Can we close this issue?

skalk, 2023-10-04:
This issue got implicitely solved by the solution of issue #4969 and therefore can get closed.

closed 2023-10-04 by skalk