genodelabs/genode #4693: run/image/uboot: make the use of gzip's `--best` mode configurable

nfeske, 2022-11-30:
By default, the run/image/uboot and run/image/uboot_fit plugins call `gzip` with the `--best` argument, which causes a surprisingly large delay in the build-test cycle for Sculpt OS for the PinePhone. Changing the argument from `--best` to `--fast` reduces the system-image integration time from 15 to 10 seconds. Hence, I propose using `--fast` by default and providing a new option `--image-uboot-gzip-best` for explicitely enabling the `--best` mode. It shaves off around 10% of the final image, which is nice for release builds but is not worth the waiting time during the regular development workflow.

nfeske, 2022-11-30:
Fixed by commit https://github.com/nfeske/genode/commit/ffb55455b8a70e2dd1aa166de9bcf33592ed5354.

chelmuth, 2022-12-01:
@nfeske @jschlatow please also consider 9ed38653767696be115e62afb1717717746bea28 as it saves much redundancy.

nfeske, 2022-12-02:
Thanks @chelmuth for the very welcome consolidation. I merged both commits to staging now.

commit a9383dfa (2022-12-02): run/image/uboot: use gzip --fast by default

By making the use of gzip's '--best' option configurable and disabling it
by default, this patch noticably reduces the built-test cycle from 15 to
10 seconds when integrating the Sculpt system image for the PinePhone.

Fixes #4693