genodelabs/genode #4694: Support to distinguish Intel E and P  for 12th Gen+ CPUs

alex-ab, 2022-12-01:


alex-ab, 2022-12-01:
I added a first version and the output of the test notebook on my desk. The information about the cores is reported by the kernel to core, which uses it to order them depending on the type. P-core and hyperthreads first, E-core last, e.g. an affinity space would look like:

CPU of the run (4P and 8E):
```
   1  2  3  4  5  6  7  8             x-axis
 1 P\ P\ P\ P\ E  E  E  E
 2 P/ P/ P/ P/ E  E  E  E

y-axis: hyperthreads \ / of same core
```

On more unsymmetric setup, i think the Framework notebook (6 P and 8 E), it would look like:
```
   1  2  3  4  5  6  7  8  9  10              x-axis
 1 P\ P\ P\ P\ P\ P\ E  E  E  E 
 2 P/ P/ P/ P/ P/ P/ E  E  E  E

y-axis: hyperthreads \ / of same core
```

The information about the E/P cores is available by the kernel, core and in the platform info in the commit.

@chelmuth: may you please check whether this is what you had in mind

chelmuth, 2022-12-01:
Yes, this looks excellent to me.

commit a8706126 (2022-12-05): nova: support detection of Intel P & E CPUs

Fixes #4694
commit 377f2166 (2023-01-24): nova: support detection of Intel P & E CPUs

Fixes #4694
closed 2023-01-24 by chelmuth