genodelabs/genode #4695: sculpt_manager: fix router label lookup

m-stein, 2022-12-01:
The commit "sculpt_manager: relax nic_drv policy label" introduced the use of
    the "label_prefix" attribute instead of "label" for the uplink policy in the
    NIC router. However, it missed an appropriate adaption of the lookup of that
    attribute when the Sculpt manager has to decide which uplink is used in a
    manually managed router config. This causes the uplink to disappear whenever a
    user creates a manually managed router config.

m-stein, 2022-12-01:
@nfeske This should fix the issue:
b89d764ab8 sculpt_manager: fix router label lookup

nfeske, 2022-12-02:
Related to #4660.

Thanks @m-stein for uncovering and fixing this regression. I merged your commit to staging.

commit f10c4709 (2022-12-02): sculpt_manager: fix router label lookup

The commit "sculpt_manager: relax nic_drv policy label" introduced the use of
the "label_prefix" attribute instead of "label" for the uplink policy in the
NIC router. However, it missed an appropriate adaption of the lookup of that
attribute when the Sculpt manager has to decide which uplink is used in a
manually managed router config. This caused the uplink to disappear whenever a
user created a manually managed router config. This commit fixes the problem.

Issue #4660
Fixes #4695
closed 2022-12-05 by chelmuth