genodelabs/genode #4696: nic_router: fix obsolete link states on IP change

m-stein, 2022-12-02:
If, at a domain with active link states that are somehow dependent on the router's local IP config (eg., the source domain of the link gets NATed by that domain), the router's IP config changes without reconfiguration involved, then, these link states become obsolete and the router doesn't handle that. In other words, the router might keep link states with outdated information. For instance, imagine domain Uplink that applies NAT for all other domains and has received a dynamic IP X, and domain Downlink that routes all traffic to Uplink. Now, a client at Downlink continuously pings an internet IP through Uplink. At some point, the DHCP server behind Uplink is restarted and Uplink receives a new IP Y. The router will continue to set the source address of the ping packets from Downlink to X instead of Y because the corresponding link state still references X. This eventually causes the router to drop the ping responses because they target X, an IP not known to the router anymore.

m-stein, 2022-12-15:
Fixed:
ca65178c20 nic_router: fix bad connection states on IP change
f72742b22c nic_router: futile ip config updates on dhcp renew

m-stein, 2022-12-15:
@chelmuth Could you please use these updated and re-based versions of the commits?

chelmuth, 2022-12-16:
I merged both commits into staging and my sculpt testing branch. Now I'm curious if the connection aborts disappear.

m-stein, 2022-12-16:
It would be a coincidence but I'll approach your problem in a dedicated issue.

closed 2023-01-24 by chelmuth