genodelabs/genode #4697: VFS: facilitate the batching of I/O operations

nfeske, 2022-12-07:
Even though various VFS plugins work asynchronously (e.g., fs, lwip), the batching of I/O operation is not consistently effective across different kernels. It particularly depends on the kernel's scheduling decision upon the delivery of asynchronous notifications. Kernels that eagerly switch to the signal receiver may thereby prevent the batching of consecutive write operations.

The influence of the kernel's scheduler can in principle be mitigated by scheduling asynchronous notifications less eagerly at the application (or libc) level. By waking up a remove peer not before the application stalls for I/O, all scheduled operations would appear at the remote side an one batch.

nfeske, 2022-12-07:
The commit series above implements this idea for the fs and lwip VFS plugins. The new run scripts libports/libc_many_writes.run and ports/run/lighttpd_fetchurl.run served as testing ground.

A few more VFS plugins could benefit from the new approach, in particular I'm thinking of lxip, rump, pipe, and fatfs.

jschlatow, 2022-12-09:
@nfeske That looks great. I was wondering how this commit series plays out on the Zynq and made a few tests. 

```
kernel    Eth speed  TCP_STREAM (Mbit/s)          TCP_MAERTS (Mbit/s)
Zybo Z720              old         new              old          new
------------------------------------------------------------------------
Base-HW     1Gbit     104.51      139.31           90.98        122.25
Base-HW   100Mbit      55.39       42.84           78.90         80.81
```

The results show a significant speedup on this platform as well. Yet, if I connect the board to a 100Mbit switch port, the TCP_STREAM throughput decreases with the new implementation. Do you have an explanation?

nfeske, 2022-12-09:
Thanks for the feedback!

> Do you have an explanation?

Unfortunately I have no goot idea.

To investigate, you may try adding `top` or the `trace logger` to the scenario to get hold of the activities of the various components. Just to see if some component is unreasonably active.

Then, the cross correlation of the socket-fs patterns between both NICs may be insightful. In the new ports/run/lighttpd_fetchurl.run script, I'm using the vfs_audit plugin to get hold of those patterns (observing the reads/writes of the socket's data file). Maybe it is worth getting a sample of those interactions for both NICs (possibly replacing the vfs_audit's use of log by using tracing).

nfeske, 2023-02-21:
Fixed in master.

closed 2023-02-21 by nfeske