genodelabs/genode #4701: legacy/lx_emul/usb: handle saturated submit queue

cnuke, 2022-12-13:


cnuke, 2022-12-13:
Commit [53a040f](https://github.com/cnuke/genode/commit/53a040f73482f1e58a17ad3ee832a8d5ea1af8ef) addresses the issue.

cnuke, 2022-12-16:
@chelmuth please see commit [abf72e3](https://github.com/cnuke/genode/commit/abf72e38f00960f8a9df502f392fb5d7653a7646) - due to our discussion it now uses `schedule_timeout` for the control-message case.

chelmuth, 2023-01-03:
Removed *fixed* because of comment about memory leak.

cnuke, 2023-01-23:
Commit [8aa64c1](https://github.com/cnuke/genode/commit/8aa64c19fc7b91c90f1a15cc68145b4978707a36) addresses the apparent memory leak in the timeout case.

chelmuth, 2023-01-24:
@cnuke You added and later removed the fixed label. What is the current state of the fix?

cnuke, 2023-01-24:
@chelmuth my bad, the removal was accidental.

commit c4545f20 (2023-01-24): legacy/lx_emul/usb: handle saturated submit queue

Check if we are able to submit the Usb packet before attempting to
allocate it. In case the queue is already full we wait until another
pending packet is freed.

Fixes #4701.
commit e826095e (2023-02-27): legacy/lx_emul/usb: handle saturated submit queue

Check if we are able to submit the Usb packet before attempting to
allocate it. In case the queue is already full we wait until another
pending packet is freed.

Fixes #4701.
closed 2023-02-27 by chelmuth