genodelabs/genode #4704: hw: ipc node implementation faulty in destruction case

skalk, 2022-12-15:
Currently, we can observe that the bomb test fails repeatedly on hw on x86_64 hardware in our nightly tests. The symptom is that an already destructed thread gets scheduled.

When investigating this issue, I could see that a thread A, which is in a send operation to the destroyed thread B, was not informed correctly about its destruction, and thereby still tries to 'help' B whenever the timeslice aka Cpu_share of A gets scheduled. Further investigations showed that when B gets destroyed, A is not part of the FIFO in B anymore, although A points to B. With other words the IPC chain got broken already before B gets destroyed.

When deeply analyzing the code of the Ipc_node class, I've realized that the boolean variable '_help' not always gets re-set after a send operation, nor the '_state' variable gets updated in all cases.
However, it was hard to follow all inter-dependencies of the object's different state variables, which are '_state', '_help', pointers to current ipc nodes for send and receive, and a FIFO for all not currently handled ones.

The state of the object is fragmented in the wrong sense in my eyes. Therefore, I've consolidated it in several steps:
* first eliminated _help, and make it part of the _state variable (SEND and SEND_HELPING)
* remove the _caller pointer, instead leave the currently handled caller within the FIFO as head
* add a REPLY_NO_SENDER to the state, to register when a caller vanishs after the pointer got removed
* after the space of states exploded somehow, I split the view on the IPC relations into 'In' and 'Out', thereby making the implementation much more simple again
* finally, I've tried to simplify the terminology a bit by using 'to' and 'from', and 'in' and 'out', removed 'request', 'callee', and 'caller'

With the changes I can run the bomb test with 100 rounds successfully and repeatedly.

m-stein, 2022-12-15:
@skalk Thanks for solving this issue! Before I review the commit in detail I'd like to say that I'm a bit uneasy about the size of it and the fact that it seems to mix cosmetic with semantic changes. Could you split it up into smaller stand-alone commits?

m-stein, 2022-12-15:
It seems that you already determined a clear step-by-step recipe with the list you posted. Maybe this could be the blueprint for the commit chain also?

skalk, 2022-12-16:
@m-stein: the smallest step of putting '_help' into the state variable could theoretically be done as separate commit, however it is nearly nothing. The other steps alone lead to an either broken semantic, or lead to a highly complex state machine. That is why, I did not preserved it, but re-worked it again. Finally, I re-thought the design and came up with the idea (inspirations thanks to a discussion with @cnuke) to split the state into "in" and "out".
I can offer to split the commit into the change of the internals of Ipc_node, and the change of the visible class signature (cosmetics), everything else seems like a penality. Please, consider to review the new class implementation as a whole, and not the changes alone. It is much less, and (in my eyes) more easy-to-read code now.

skalk, 2022-12-16:
@m-stein I split it into three commits now.

m-stein, 2022-12-21:
I created a series of cosmetic fix-up's for your original commit "[hw: re-work the ipc node's internal state machine](https://github.com/skalk/genode/commit/f5201601575f7158dffd3d7c991bcf4b09d5fb58)", then reverted it and re-applied it in small steps. Afterwards, I added your other two commits with small cosmetic fix-up's. This is the result:

45af852b96 hw: rename functions of Ipc_node class signature
f53b745ce5 hw: mark threads as dead in case of ipc violations
3a2d00c6b7 base-hw ipc: dedicated in/out sub-structs
ccded3e653 base-hw ipc: skip dequeuing in destructor
600aac71a7 base-hw ipc: fix "can send request" method
2cc0e649a6 base-hw ipc: get rid of "caller" member variable
d6dafa8c48 base-hw ipc: no strange out-state change on send
93c0188d69 base-hw ipc: separated in and out state
bf6f97bfc8 base-hw ipc: rename state values
04642521e7 base-hw ipc: get rid of "help" member variable
0958e0bdc6 base-hw ipc: use "cancel send" in destructor
d8ab62b55e base-hw ipc: simplify "send request" method
df684fb459 base-hw ipc: rename member variables
cc995cccf1 base-hw ipc: introduce state "destruct"
8f6c710730 base-hw ipc: fix "cancel waiting" method
1958086970 base-hw ipc: methods whether channels are blocking
4b5146cb74 base-hw ipc: reduce number of private methods
a22bef23e9 base-hw ipc: extend "cancle send" method
8f9f2f5f0b base-hw ipc: rename methods
8737c1e6fc base-hw ipc: function const'ness

If you're interested in my fix-up's to your original commits, please let me know.

m-stein, 2023-01-05:
I have applied a review of @skalk and we now have these two versions of commit history with the exact same result:

A)
de4e78ec37 hw: rename functions of Ipc_node class signature
1dadf7b45d hw: mark threads as dead in case of ipc violations
291adfc5e4 fixup "hw: re-work the ipc node's internal state machine"
b056de5161 hw: re-work the ipc node's internal state machine

B)
7392a6851b hw: rename functions of Ipc_node class signature
0df2ee7b4a hw: mark threads as dead in case of ipc violations
cbfeb2607f base-hw ipc: dedicated in/out sub-structs
b088572156 base-hw ipc: use common noncopyable marker
9b3c253f1f base-hw ipc: skip dequeuing in destructor
2abccf9023 base-hw ipc: fix "can send request" method
b7e9a2b104 base-hw ipc: get rid of "caller" member variable
d6dafa8c48 base-hw ipc: no strange out-state change on send
93c0188d69 base-hw ipc: separated in and out state
bf6f97bfc8 base-hw ipc: rename state values
04642521e7 base-hw ipc: get rid of "help" member variable
0958e0bdc6 base-hw ipc: use "cancel send" in destructor
d8ab62b55e base-hw ipc: simplify "send request" method
df684fb459 base-hw ipc: rename member variables
cc995cccf1 base-hw ipc: introduce state "destruct"
8f6c710730 base-hw ipc: fix "cancel waiting" method
1958086970 base-hw ipc: methods whether channels are blocking
4b5146cb74 base-hw ipc: reduce number of private methods
a22bef23e9 base-hw ipc: extend "cancle send" method
8f9f2f5f0b base-hw ipc: rename methods
8737c1e6fc base-hw ipc: function const'ness

m-stein, 2023-01-09:
@skalk I pushed a commit series that should incorporate all the stuff we discussed. Do you agree?

7c2cf321d4 hw: rename functions of Ipc_node class signature
df36b08721 hw: mark threads as dead in case of ipc violations
ef34df0ec7 squash "hw: re-work the ipc node's internal state machine"
17b8d564f3 fixup "hw: re-work the ipc node's internal state machine" (cancel send doc)
01042a70a2 fixup "hw: re-work the ipc node's internal state machine" (enum style)
f4f6f0af0f fixup "hw: re-work the ipc node's internal state machine" (if brackets)
7ad9ca34e8 fixup "hw: re-work the ipc node's internal state machine"
f2beac0eb7 hw: re-work the ipc node's internal state machine

skalk, 2023-01-09:
@m-stein thank you, yes I agree. I've squashed all fixups etc. together to simplify life for @nfeske and @chelmuth. The following three commits on my merge_to_staging branch summarize the work: 40fa8af 6d199b6  b689126

nfeske, 2023-01-09:
Thanks for the notice. I have merged the three commits to staging.

closed 2023-01-24 by chelmuth