genodelabs/genode #4706: Simplify and modernize VFS

nfeske, 2022-12-16:
The VFS library carries quite a few historical legacies that are no longer needed, or that could benefit from an update to modern C++ practices.

m-stein, 2023-01-06:
@nfeske I found your commit series to be well divided into small steps and well documented. It's also nice to see that your changes in general reduce the code complexity. Beside the minor comments I added inline, I could find nothing wrong with your patches regarding the CBE-related VFS plugins. However, I must admit that I understand the mechanics behind the plugins way less than I'd wish for. I hope that this will change with my working at them the next weeks.

nfeske, 2023-02-20:
Commit removes the `file_size` type from the `File_io_service` interface and replaces the former C-style buffer arguments (pointer, len) by the `Byte_range_ptr` and `Const_byte_range_ptr` types.