genodelabs/genode #4707: trace session: fix double quota accounting

cproc, 2022-12-20:
Since the introduction of the `Constraint_ram_allocator` to the trace session component, the previous quota accounting code appears to be obsolete and causes a higher RAM quota demand. One location already got fixed by commit 99bdfbe, but there are more locations remaining.

cproc, 2022-12-20:
Fixed by 04192b7 if I understood the semantics correctly.

alex-ab, 2022-12-20:
In alloc_policy (https://github.com/genodelabs/genode/commit/04192b76200f473007979150a22e2a7630584149#diff-c8153d615a24995bc0e7b3939bf679812900f1f564b7bb4db3c211126edc8a94R63) 
```
Dataspace_capability ds_cap = _ram.alloc(size);
_policies.insert(*this, id, _policies_slab, ds_cap, size);
```
it seems that we may leak ram and a cap, if insert throws. The implementation of Policy_registry::insert (https://github.com/genodelabs/genode/blob/master/repos/base/src/core/include/trace/policy_registry.h#L109) uses new on an allocator, so this may trigger, or ?

cproc, 2022-12-20:
I think you are right. Also, `unload_policy()` did not free the dataspace. Fixed by fa89596.

cproc, 2022-12-20:
I also wonder about https://github.com/genodelabs/genode/blob/09efe915b3c167f18fd2fbe1cb066093ec7fe5a8/repos/base/src/core/include/trace/subject_registry.h#L298-L308

Maybe the buffer and policy should be freed regardless of the status of the source?

nfeske, 2022-12-20:
Thanks @alex-ab for the extra scrutiny. I merged fixup https://github.com/genodelabs/genode/commit/fa89596420ca682615d8f5ae0ed8fe8cd34c7b52 to staging.

> Maybe the buffer and policy should be freed regardless of the status of the source?

Intuitively, this sounds right. It looks like `release` is only called by `Subject_registry::_unsynchronized_destroy`, which destroys the buffers anyway. If so, the extra `release` method seems a bit superfluous. But I have not double-checked.

commit 09efe915 (2022-12-20): trace session: fix double quota accounting

Fixes #4707.
closed 2023-01-24 by chelmuth