genodelabs/genode #4709: fec_nic_drv/imx7d_sabre: warning about "ep tried to do cache maintainance at unallowed address"

chelmuth, 2022-12-21:
For example, when running depot_download.run the following log is generated.

```
Kernel: init -> drivers -> nic_drv -> ep tried to do cache maintainance at unallowed address 0xfff800
Kernel: init -> drivers -> nic_drv -> ep tried to do cache maintainance at unallowed address 0xfff940
Kernel: init -> drivers -> nic_drv -> ep tried to do cache maintainance at unallowed address 0xfffb40
Kernel: init -> drivers -> nic_drv -> ep tried to do cache maintainance at unallowed address 0xfff800
```

Other affected scenarios are netperf_lwip/lxip in all flavors while, e.g., fetchurl_lwip does not show this warning.

nfeske, 2023-08-24:
This problem no longer occurs with Genode 23.08.

chelmuth, 2024-02-05:
It's still there...

```
[2024-02-04 04:18:32] Kernel: init -> nic_drv -> nic_drv -> ep tried to do cache maintainance at unallowed address 0xfff800
```

chelmuth, 2025-04-29:
Since IMX-relevant changes in #5264 were merged, the issue no longer persists.

closed 2025-04-29 by chelmuth