genodelabs/genode #4712: not able to compile core for seL4 + arm_v8a

sid-agrawal, 2022-12-29:
Hi,
I am running into the following issuing when getting started on Genode with sel4 and arm_v8a. I was able to get it running for x86_64, so it seems like, I just missed a step somewhere.

Steps
```bash
git clone https://github.com/genodelabs/genode.git
cd genode
./tool/create_builddir arm_v8a
tool/ports/prepare_port sel4 grub2
```

Adjust build.conf
```diff
diff --git a/build.conf b/build.conf
index 3e8959817..af107662c 100644
--- a/build.conf
+++ b/build.conf
@@ -68,6 +68,7 @@ endif
 REPOSITORIES += $(GENODE_DIR)/repos/base
 REPOSITORIES += $(GENODE_DIR)/repos/os
 REPOSITORIES += $(GENODE_DIR)/repos/demo
+REPOSITORIES += $(GENODE_DIR)/repos/hello_tutorial
 
 
 ##
@@ -79,7 +80,7 @@ REPOSITORIES += $(GENODE_DIR)/repos/demo
 #
 # Make sure to execute 'make prepare' in 'libports' prior building.
 #
-#REPOSITORIES += $(GENODE_DIR)/repos/libports
+REPOSITORIES += $(GENODE_DIR)/repos/libports
 
 #
 # Ports of popular 3rd-party applications
```

Tried building just the `core` first, which I think should be possibile
```bash
$ make core BOARD=virt_qemu_arm_v8a KERNEL=sel4
checking library dependencies...
Library-description file core-sel4.mk is missing
```

Then try building everything, I think a path var needs fixing somewhere.
````bash
$ make  BOARD=virt_qemu_arm_v8a KERNEL=sel4
    COMPILE  avl_tree.o
    COMPILE  capability.o
    COMPILE  capability_raw.o
    COMPILE  child.o
    COMPILE  child_process.o
In file included from /home/siagraw/tmp/genode/repos/base-sel4/src/include/base/internal/kernel_debugger.h:22,
                 from /home/siagraw/tmp/genode/repos/base-sel4/src/include/base/internal/assert.h:24,
                 from /home/siagraw/tmp/genode/repos/base-sel4/src/include/base/internal/capability_space_sel4.h:24,
                 from /home/siagraw/tmp/genode/repos/base-sel4/src/include/base/internal/parent_cap.h:24,
                 from /home/siagraw/tmp/genode/repos/base/src/lib/base/child_process.cc:22:
/home/siagraw/tmp/genode/repos/base-sel4/src/include/base/internal/sel4.h:22:10: fatal error: sel4/sel4.h: No such file or directory
   22 | #include <sel4/sel4.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/home/siagraw/tmp/genode/repos/base/mk/generic.mk:62: child_process.o] Error 1
make[1]: *** [var/libdeps:109: base-sel4-common.lib] Error 2
make: *** [Makefile:336: gen_deps_and_build_targets] Error 2

alex-ab, 2023-01-03:
The support for arm_v8a was never enabled for seL4 with Genode, so this combination is not supported.

sid-agrawal, 2023-01-03:
I see, thanks Alex, is that true for `arm_v7a` too?

alex-ab, 2023-01-03:
No, if you look in repos/base-sel4 you will find hints for  imx7d_sabre and imx6q_sabrelite boards.

sid-agrawal, 2023-01-03:
Got it. I see those files.

But there is no support for arm_v7a qemu right?

alex-ab, 2023-01-03:
Yes, there is now "virt_qemu" board support for seL4 by now. It is probably not too hard to add, but there was no need for it.

sid-agrawal, 2023-01-03:
Got it. Thanks!

closed 2023-01-03 by sid-agrawal