genodelabs/genode #4713: gpu: change semantics from buffer objects to vram

ssumpf, 2023-01-02:
Genode's GPU session abstracts GPU objects as buffers, because both Linux and DRM operate on buffer objects. Experience has shown that DRM through the Mesa library tends to allocate a lot of small (one or two pages) buffers, which all have to be allocated and freed through GPU session calls. Therefore, I propose to change the session semantic to VRAM objects. This way, a GPU client (i.e., DRM) can allocate larger chunks of graphics memory that can be used to create *multiple* Mesa buffer objects at the client side. Note: A VRAM object can still be mapped to one buffer object at the client side.

ssumpf, 2023-01-02:
@nfeske: 3800c68 proposes the GPU session interface changes for VRAM (`old` function is followed by `replacement` function).

chelmuth, 2023-01-03:
I have some thoughts/questions after reading your proposal.

1. I like the notion of coarse-grained *VRAM ranges* for allocation, export, and import as this renders *buffers* a client-side object. Nevertheless, I expected buffers to be represented by a tuple of *( VRAM ID, offset, size )* on execute and gpu-side map/unmap/tiling. Could you explain why this is not the case in your proposal, esp., why *size* is no required on *execute*?

2. Isn't `query_vram_gpu()` just a convenience function that should be implemented on client side? I understand the semantics of *map* to be entirely client-driven.

3. Is the *tiling mode* completely opaque to the interface?

4. Would you mind to update the documentation of the interface according to the proposal?
   - clear distinction of VRAM and (MESA) buffers
   - avoidance of Intel-specific terms in generic functions resp. stating these as examples

ssumpf, 2023-01-05:
> I have some thoughts/questions after reading your proposal.
>
>     1. I like the notion of coarse-grained _VRAM ranges_ for allocation, export, and import as this renders _buffers_ a client-side [...]

`size` has never been used by any GPU multiplexer. The buffer just got executed. Now we additionally need tthe offset in video ram (= start of buffer/GPU instructions) and check if the offset is within the given vram range (id) that is known by the multiplexer. From there the Mesa generated GPU code can do anything to shot itself into the foot anyway.

>     2. Isn't `query_vram_gpu()` just a convenience function that should be implemented on client side? I understand the semantics of _map_ to be entirely client-driven.

This is required by Lima, where GPU virtual addresses are assigned by the GPU driver. Mesa has to retrieve these in order to write the correct addresses into it's batch buffers. On Intel, for example, this is handled by Mesa, so the function is not called.

>     3. Is the _tiling mode_ completely opaque to the interface?

Yes, mode is architecture dependent.

>     4. Would you mind to update the documentation of the interface according to the proposal?
>
>        * clear distinction of VRAM and (MESA) buffers
>        * avoidance of Intel-specific terms in generic functions resp. stating these as examples
Of course, and have already done so.

chelmuth, 2023-01-06:
> `size` has never been used by any GPU multiplexer. The buffer just got executed. Now we additionally need tthe offset in video ram (= start of buffer/GPU instructions) and check if the offset is [...]

Okay, got that.

> This is required by Lima, where GPU virtual addresses are assigned by the GPU driver. Mesa has to retrieve these in order to write the correct addresses into it's batch buffers. On Intel, for [...]

I suspected something like that. I bothers me that the current interface deludes the user that `alloc_vram` and `map_vram` are orthogonal functions resp. `map_vram` effectively defines the GPU-context-local address, which is not always the case.  So, let's approach this from the opposite direction: May we remove the `va` parameter from `map_vram` and let the multiplexer manage the GPU-local mappings in all cases?

ssumpf, 2023-01-06:
@chelmuth: For an explanation what we did now, here a short essay which I will put somewhere as a comment in code as well. Because of the different GPU driver architectures, functions may have different semantics. Here a short explanation of the difference of Intel and Lima.

Intel:
* `alloc_vram` is used by Iris to create an internal buffer object cache on the internal Mesa side. Because the allocated memory is RAM, Iris mmaps it directly into the clients address space (Genode::attach)
* Iris assigns Gpu virtual addresses to buffers, these addresses may change when buffers are reused later. On DRM side, Iris calls `DRM_MAP_PPGTT`, which we implement through `map_gpu` (establish a GPU mapping) on the client and multiplexer side.
* Graphics memory can also mapped through the aperture (for example, when tiling/untiling buffers, which we currently do not see, but have seen in the past). This is what `map_cpu` is used for on Intel. It adds a GTT mapping into the aperture (IOMEM) and returns a dataspace with the physical address within the aperture. This dataspace has a different physical address than the one returned by  `alloc_buffer` but in the end points to the same RAM through the GTT. This is now a nop because it is not used, which may change in newer Mesa versions again.

Lima:

The ported Lima driver assigns a GPU virtual address during buffer allocation. We found a way to at least tell the driver what virtual address to use, because on default it will use an arbitrary fitting GPU virtual address, which means unlike Iris, the GPU driver manages the virtual address space on Lima. This made it possible to remove `query_vram_gpu`. So on Lima the semantic is as follows now:

1. `alloc_vram` is a nop because we cannot hand a GPU virtual address over
2. `map_gpu` will actually allocate a buffer of given `size` at a given `Gpu::Virtual_address` in the GPU page tables. On the Mesa side, we implemented a virtual address-allocator, and therefore, handle the GPU address space ourselves as Iris does in contrib Mesa code.
3. `map_cpu` will return the `Dataspace_capability` of the allocated buffer that then can be directly attached by Mesa because the GPU memory is again ordinary RAM.

This changes can be found in [1db3c40](https://github.com/ssumpf/genode/commit/1db3c4053a203a6de68badf6f687253b8c585913) and [23f1ae5](https://github.com/ssumpf/genode-allwinner/commit/23f1ae575a5d01634998d5efb4f36b3ef0f55075)

I hope this explains the current situation a little better.

ssumpf, 2023-01-06:
@nfeske, @chelmuth: The commit series above implements the updated GPU session for Intel/etnaviv/Lima/black_hole and updates the respective `libdrm` implementations accordingly. Additionally, there are some missing parts for etnaviv because it did not work on `staging`.

nfeske, 2023-01-09:
Thank you @ssumpf for the excellent write-up. From my perspective, the clarity of the interface greatly benefits from the disposal of the `query_vram_gpu`. Thanks @ssumpf and @cnuke for the harmonization of the interface among the GPU types!

ssumpf, 2023-01-20:
0bd8a82 renames `Gpu::Virtual_address::va` to `Gpu::Virtual_address::value`
d9e1ceb adds the note above to the `gpu_session.h`

ssumpf, 2023-02-07:
@cnuke: As requested rebased and tested my issue branch to staging under: https://github.com/ssumpf/genode/tree/issue4713_gpu_vram_staging

cnuke, 2023-02-07:
@ssumpf thank you very much :+1: .

nfeske, 2023-04-26:
What is the status of this issue? Since the referenced commits entered the master branch, may we close it?

ssumpf, 2023-04-26:
Fixed in master.

closed 2023-04-26 by ssumpf