genodelabs/genode #4715: nvme_drv: add support for host-memory-buffer

cnuke, 2023-01-04:


cnuke, 2023-01-04:
The following commits introduce the feature and testing was done with a `Samsung 980` device:

* 1aecd23dd3 - nvme_drv: move PRP list helper construction
* 90173dcd86 - nvme_drv: allow for accessing admin cmd results
* 3f4d06d1b1 - nvme_drv: provide get/set feature command
* 59ae16d093 - nvme_drv: add support for host-memory-buffer

The Sculpt integration is ommitted as without more testing I am not sure which HMB size would be reasonable.

chelmuth, 2023-01-24:
I merged the preparatory commits to staging but skipped the final feature for the moment. Do you see any chance that it could break anything with Sculpt? What is the practical experience with *preferred* buffer size? Could we just use a small value like 8M for the integration? Does any of our machines on duty comes with an NVMe that supports HMB?

cnuke, 2023-01-26:
> I merged the preparatory commits to staging but skipped the final feature for the moment. Do you see any chance that it could break anything with Sculpt?

It should not break anything in Sculpt as the feature only gets active if availabe by the hardware and enabled in the drivers configuration.

> What is the practical experience with _preferred_ buffer size?

The devices I have access too have a preferred buffer size of _64 MiB_ - I have not checked if that is common or not.

> Could we just use a small value like 8M for the integration?

It depends on the device - in my case `hmmin` is *16 MiB*. Judging by the specs of the systems that feature NVMe devices this amount is neglectable so my answer would be yes.

> Does any of our machines on duty comes with an NVMe that supports HMB?

Not that I know of.

chelmuth, 2023-01-26:
> > Does any of our machines on duty comes with an NVMe that supports HMB?
> 
> Not that I know of.

So we should order a cheap one like WD SN750 **SE**/SN770 for testing, right?

cnuke, 2023-01-26:
@chelmuth commit [3c7ca1a](https://github.com/cnuke/genode/commit/3c7ca1a6be4ae73f62c3f6e2139281c0731ad162) addresses your remarks raised during review.

> So we should order a cheap one like WD SN750 **SE**/SN770 for testing, right?

Yes, although a WD *SN570* would be sufficient as well.

cnuke, 2023-03-09:
@chelmuth commit [1ef990c](https://github.com/cnuke/genode/commit/1ef990c29bf8bb41830e65cf4e61c02ef46b257c) address your remark regarding the preferred HMB size.

chelmuth, 2023-03-10:
Compile error on x86_32.

```
.../repos/os/src/drivers/nvme/main.cc: In constructor ‘Nvme::Set_hmb::Set_hmb(Genode::addr_t, Genode::addr_t, Genode::size_t, Genode::uint32_t)’:
.../repos/os/src/drivers/nvme/main.cc:450:29: error: right shift count >= width of type [-Werror=shift-count-overflow]
  450 |   write<Cdw14::Hmdlua>(hmdl >> 32u);
      |                        ~~~~~^~~~~~
```

It seems `hmdl` is not `addr_t` but `uint64_t`, right?

cnuke, 2023-03-10:
Yes, that is right, commit [a05e95d](https://github.com/cnuke/genode/commit/a05e95d7a6a56230112fbbe8b0985bb308d26834) addresses is problem.

commit 707783cd (2023-03-09): nvme_drv: add support for host-memory-buffer

This commit introduces support for the HMB feature and will setup the
buffer during start-up. The host-memory-buffer (HMB) feature is mostly
used on NVMe devices that do not make use of an DRAM cache to store its
translation tables amongst other operational data. Not using HMB can
impair the performance on such devices.

The memory is allocated in 2 MiB chunks of DMA-capable memory and its
total size in bytes is configurable via the 'hmb_size' config attribute.
The driver always checks the minimal and preferred size of the HMB and
issues a warning  [... cut]
closed 2023-03-13 by chelmuth