genodelabs/genode #4718: Enabling ccache for compilation of Linux sources in ported drivers

tomga, 2023-01-06:
Recently I found that compilation of ported drivers (using new DDE approach) does not use `ccache` even though it is enabled in `build.conf` and used for other compilations.

I made some checks following change (applied currently only in one case) enables `ccache`. I checked clean builds without parallel jobs and had about 3:45 without enabling `ccache` and about 1:45 with it enabled, so effect seems to not be small. Of course it was with cache populated already.

Before I start applying this in other places (I can see 7 places where such a change would be needed), I'd like to ask you if you find it valuable.

Note: to enable `ccache` it was enough to just add `CC=$(CC)` to flags. Rest of changes was to avoid cache misses.

skalk, 2023-01-09:
The ccache is not effective when building the Linux reference kernel only. This is typically needed once when porting new drivers resp. protocol stacks. It is not involved in the compilation of driver ports.
So I do not speak against @tomga changes, but do not wanted wrong expectations to be raised. I would wonder, where in the driver targets the ccache should not be effective, as it simply uses Genode's build system only (apart from the generation of Linux kernel headers).

tomga, 2023-01-09:
@nfeske, I'm sorry I forgot to add commit message. But you found the right one.

@skalk, I can't say much about effectiveness of using `ccache` for this compilation in normal work on porting drivers. I found it during my slowly progressing work on trying to build Genode with Scons where I frequently run clean builds for consistency checks. During that I found that two builds of parts of Linux in ported drivers seem to take more time then building everything else in repositories enabled by default. It may happen that it can be important only for your nightly builds but given that there is code to rebuild on any makefile change (`kernel_config.tag: $(MAKEFILE_LIST)`) it may have an impact on developer workflow too.

My lack of knowledge if it is known and if you find it important was the reason to ask you if you want this change.

I'll work on this not earlier than tomorrow evening, so let me know in case your final decision is to not want such modifications.

tomga, 2023-01-12:
I put my changes in [issue_4718](/tomga/genode/commits/issue_4718) branch.

They are split in two commits:
 * 8adcb22057824f01c86d45c80c6396b9197ed4fa - enables `ccache` where it really has some effect
 * 8ff881494ab1bfe59fb1029a24b50124b57a558b - adds similar options for rest occurences of `LX_MK_ARGS`
If you decide that consistency is not valuable in this case just ignore the latter.

I tested builds with verification that `ccache` has mostly hits for `x86_64`. For `arm_v8a` and `x64_32` I just tested builds. I also built one `virt_linux` depot if changes could break something there but it seems to be ok.

skalk, 2023-01-13:
@tomga thank you for providing your commit series. I think we would leave out the second one, as we do not see any benefits yet, but it increases the anyway complex makefile. To be able to merge the first commit, you would need to correct the commit message which is misleading now. You've written: "...build of code from Linux
used for ported device drivers...". As I've tried to explain in my last comment, the targets in `src/virt_linux` and `src/pc_linux` are not used for any Genode components nor drivers. They are simply convenience wrappers to build a reference Linux kernel during development. Thereby a developer is able to cross-correlate the behaviour of a ported driver to the original Linux kernel. With other words, you normally don't build `src/virt_linux` or `src/pc_linux` at all. I would be fine with something like: "enable ccache for Linux kernel reference" or something similar. Would you please re-phrase the commit message?

tomga, 2023-01-18:
@skalk, of course I don't have a problem with rewording. Hopefully 703b8c554a3e8e72ccc7140861f45e143c41d479 is ok.
I didn't understand properly you first comment. I'm sorry about that.

skalk, 2023-01-19:
@tomga thank you very much for the rewording! You do not have to apologize, its okay.

nfeske, 2023-01-19:
Thanks @tomga for the update. I merged the latest version to staging.

> I didn't understand properly you first comment.

I guess @skalk shared his doubts that the added complexity in `_linux_generated.inc` it really worth it. The `lib/mk/*_linux_generated.mk` pseudo libraries merely create a Linux kernel build directory, generating a bunch of headers along the way. It is needed as include-search location for building Linux driver code that depends on the generated headers.

The compiler of the Genode tool chain is not involved in creating those pseudo libraries. Hence, ccache is not expected to bring a benefit.

tomga, 2023-01-19:
@nfeske I think that adding options in `_generated.inc` may have benefit for `ccache` that is configured by Genode build system. I'm mentioning specifically those _Linux build stability flags_, not defining `CC`. I didn't verify this yet and therefore didn't mention until now, but your comment suggest that I may be right.

If I correctly understand, without those flags generated headers contain current timestamp causing `ccache` to have a cache miss on each compilation. I think that currently it is possible that:
 * Each Makefile change causes repeating tinyconfig and header generation.
 * Driver code has to be recompiled because of new headers. Due to one of headers containing timestamp of generation, `ccache` will not find result in cacche. I don't know if this specific _timestamp header_ is included in all compilation units though.

nfeske, 2023-01-23:
> I'm mentioning specifically those Linux build stability flags [...]

Thank for highlighting this aspect. This could actually make a difference in the following situation: When building depot binary archives for multiple DDE-Linux drivers that are using the same `pc_linux` API, a fresh Linux-build-directory is created once for each driver. So if the files generated by the Linux build system differ between each run (because on an absolute path or a date value ending up in a generated file), ccache may not match previously compiled files. This scenario could be tested by invoking the depot/create tool for building the binary packages for two PC drivers.

That said, this effect remains hypothetical at this point. Personally, I'm not eager to investigate because the creation of binary archives is not part of the work flow during the actual development. To recap, there are two important development phases to distinguish.

1. Finding a good Linux kernel configuration (see https://genode.org/documentation/genode-platforms-22-05.pdf#subsection.2.10). This step can require many rebuilds of the Linux kernel. Unfortunately, ccache does not help here because many kernel-config changes have globally visible effects in generated header files. (your first commit _potentially_ improve the situation but I think that most kernel-config changes render ccache ineffective anyway)
2. Iterating the development and testing of the driver embedded in a Genode component. Here, `*_generated` library is created only once (because the Linux kernel config remains constant at this development stage) and no depot archives are built (`--depot-auto-update` is not enabled). For this iterative building of the driver in this situation, ccache was already effective.

Since I'm unable to grasp a possible life improvement by the second commit, I'd personally not spend time on further investigation. If you decide to investigate nevertheless, please try to find a concrete example where the change has a measurable effect (like the situation outlined in my first paragraph above?). We should include the second commit only if we can validate that it serves a purpose.

The technical questions aside, the suggested changes make me worried about code duplicates, which must be maintained down the road. Your first commit contains the same code five times. Please don't mistake this as a blame - you merely followed the existing patterns, which already cried for a de-duplication effort. But the second commit would introduce further duplicates. So before going ahead, I'd appreciate a commit that de-duplicates the code.

tomga, 2023-01-24:
@nfeske, thank you for providing details about this process. I agree that further complication is not fruitful enough.

I see that this commit was already merged to master, so I believe it can be closed now. Thus I do it.

closed 2023-01-24 by tomga