genodelabs/genode #4724: base-hw: bootstrap fails on x86 qemu with more than 1024MB RAM

atopia, 2023-01-12:
Base-hw bootstrap on x86 qemu fails when the VM is given more than 1024MB RAM, probably because it maps only 1GB of RAM and the ACPI tables are outside of the mapped area (quick analysis by @alex-ab, this hasn't been debugged thoroughly).
Minimal example: running `repos/base/run/log.run` with `append qemu_args "-m 1024"` works, whereas for example with `append qemu_args "-m 1536"` it ends up in a boot loop.

chelmuth, 2023-01-16:
The boot loop happens for RAM sizes from 1025M to 2048M - up to 1024M and and from 2049M upwards it works as expected. I suspect the *mapping hole* from 1 to 2 GiB in *crt0_translation_table.s* is too pragmatic for your test case. Could we move the mapping hole to reasonably high addresses above usual RAM sizes?

atopia, 2023-01-16:
@chelmuth I moved the mapping hole above 4GiB in the hope that most hardware will use addresses below 4GiB. Fixes the initial qemu problem for 1536MiB RAM without introducing new regressions on qemu for 7GiB and 8GiB RAM. Do you think this is good enough for now, or should we move the mapping hole beyond 32 or 64GiB?

chelmuth, 2023-01-17:
I had an offline discussion with @alex-ab  after commenting here yesterday, in which it became apparent that bootstrap does not use a *mapping hole* for dynamic mappings. The current state was never designed and represents just the result of the development steps (some RAM beyond the ELF image is required in first GiB, access to APIC happens in fourth GiB, SMP enablement was done on a test machine that stored ACPI structures in third GiB). Therefore, we feel confident that it is best to just setup PDP0 and PD0-3 for identical mappings in a first step (except special handling of page0 and APIC ranges). 

All our hardware should be covered by this approach as even the locally accessible Framework and U7412 notebooks do not store ACPI data structures above 4 GiB. If this changes in the future, we agreed that bootstrap needs a semi-dynamic setup of PDs similar to the current asm code in *crt0_translation_table.s* before accessing dynamic addresses like ACPI tables.

Would you mind to adapt your commit accordingly?

atopia, 2023-01-17:
Thanks for the clarification! I'm not sure what you mean by "If this changes in the future, we agreed that bootstrap needs a semi-dynamic setup of PDs similar to the current asm code in crt0_translation_table.s before accessing dynamic addresses like ACPI tables." but otherwise this issue should be resolved.

alex-ab, 2023-01-17:
The comment in the beginning of the file is misleading, maybe it should be removed/adjusted.

atopia, 2023-01-17:
I just adjusted it to say that we identity-map up to 4GB, or did you have something else in mind?

alex-ab, 2023-01-17:
That was what I had in mind, :+1:

skalk, 2023-01-17:
Actually, the one-by-one mapping was used by bootstrap to have enough initial RAM available when starting with MMU enabled. Additional I/O memory for peripherals etc. have to be added to the page-table at runtime, the same way like on ARM. To me it seems more sustainable to change the ACPI code inside bootstrap to always map the table(s) before accessing it, instead of extending the 1:1 mapping in the crt0 code over and over again.
Anyway we can also keep the current change from @atopia, but change the ACPI code accordingly too.

chelmuth, 2023-01-17:
> To me it seems more sustainable to change the ACPI code inside bootstrap to always map the table(s) before accessing it, instead of extending the 1:1 mapping in the crt0 code over and over again.

Exactly what I tried to express in my second paragraph.

> Anyway we can also keep the current change from @atopia, but change the ACPI code accordingly too.

Merged the commit to staging.

commit 624c76a7 (2023-01-17): base-hw: remove memory hole in bootstrap page table

Tests on qemu would fail when started with RAM sizes from 1025MiB to
2048MiB, because the the mapping hole in the page table from 1GiB to
2GiB would interfere with qemu's mapping addresses for ACPI.

Identity-map the complete first 4GiB of memory to catch all early
memory accesses during bootstrap.

Fixes #4724.
closed 2023-01-24 by chelmuth