genodelabs/genode #4725: Error: incomplete or missing recipe

dogmode, 2023-01-14:
While following latest revision of Genode Foundations, after running `make run/demo KERNEL=linux BOARD=linux` the following error appears:
`Error: incomplete or missing recipe (genodelabs/api/base genodelabs/api/base-linux genodelabs/api/blit genodelabs/api/capture_session genodelabs/api/event_session genodelabs/api/framebuffer_session genodelabs/api/gui_session genodelabs/api/input_session genodelabs/api/nitpicker_gfx genodelabs/api/os genodelabs/api/report_session genodelabs/api/timer_session genodelabs/pkg/drivers_interactive-linux genodelabs/raw/drivers_interactive-linux genodelabs/src/base-linux genodelabs/src/demo genodelabs/src/event_filter genodelabs/src/fb_sdl genodelabs/src/global_keys_handler genodelabs/src/init genodelabs/src/nit_focus genodelabs/src/nitpicker genodelabs/src/report_rom genodelabs/src/rom_filter)`

I have all the needed tools installed, what can be the possible issue here?

dogmode, 2023-01-14:
I have been able to partially resolve the error by manually downloading packages with `tool/depot/download`, but it still requires some packages that aren't on [genodelabs depot](https://depot.genode.org/genodelabs):
`Error: incomplete or missing recipe (genodelabs/pkg/drivers_interactive-linu genodelabs/src/base-linux genodelabs/src/demo)`

dogmode, 2023-01-15:
As far as I understand from tool/depot/create help message, it should _create_ package archives from source tree, but instead it requires packages to be downloaded with tool/depot/download, because after this the issue is partially resolved:
`Error: incomplete or missing recipe (genodelabs/api/base-linux genodelabs/pkg/drivers_interactive-linux genodelabs/raw/drivers_interactive-linux genodelabs/src/base-linux genodelabs/src/demo)`

Though it still complains about some packages that I wasn't able to find in depot.genode.org.

ssumpf, 2023-01-15:
@dogmode: Please uncomment
`RUN_OPT += --depot-auto-update`
in `<build-dir>/etc/build.conf` and make sure all `REPOSITORIES` entries are uncommented as well.

dogmode, 2023-01-15:
> @dogmode: Please uncomment `RUN_OPT += --depot-auto-update` in `<build-dir>/etc/build.conf` and make sure all `REPOSITORIES` entries are uncommented as well.

I have all these options uncommented

ssumpf, 2023-01-15:
@dogmode: Can you run `make BOARD=linux KERNEL=linux VERBOSE= run/demo` and paste the complete output here?

dogmode, 2023-01-15:
@ssumpf
```
test -f "/home/slavko/genode.test/repos/os/run/demo.run" || (echo "Error: No run script for demo"; exit -1)
/home/slavko/genode.test/tool/run/run --genode-dir /home/slavko/genode.test \
                                     --name demo \
                                     --specs "linux x86_64 x86 64bit" \
                                     --board "linux" \
                                     --repositories "/home/slavko/genode.test/repos/base-linux /home/slavko/genode.test/repos/base /home/slavko/genode.test/repos/os /home/slavko/genode.test/repos/demo /home/slavko/genode.test/repos/libports /home/slavko/genode.test/repos/ports /home/slavko/genode.test/repos/dde_linux /home/slavko/genode.test/repos/dde_rump /home/slavko/genode.test/repos/gems /home/slavko/genode.test/repos/pc /home/slavko/genode.test/repos/dde_bsd /home/slavko/genode.test/repos/dde_ipxe" \
                                     --cross-dev-prefix "/usr/local/genode/tool/21.05/bin/genode-x86-" \
                                     --qemu-args "-no-kvm -display sdl " \
                                     --make "make SHELL=/usr/bin/bash" \
                                     --ccache \
                                     --depot-auto-update --include power_on/linux --include log/linux --include boot_dir/linux \
                                     --include /home/slavko/genode.test/repos/os/run/demo.run
including /home/slavko/genode.test/tool/run/power_on/linux
including /home/slavko/genode.test/tool/run/log/linux
including /home/slavko/genode.test/tool/run/boot_dir/linux
including /home/slavko/genode.test/repos/os/run/demo.run
update depot: /home/slavko/genode.test/tool/depot/create genodelabs/bin/x86_64/base-linux genodelabs/bin/x86_64/demo genodelabs/bin/x86_64/global_keys_handler genodelabs/bin/x86_64/init genodelabs/bin/x86_64/nit_focus genodelabs/bin/x86_64/nitpicker genodelabs/bin/x86_64/report_rom genodelabs/bin/x86_64/rom_filter genodelabs/pkg/x86_64/drivers_interactive-linux CROSS_DEV_PREFIX=/usr/local/genode/tool/21.05/bin/genode-x86- DEPOT_DIR=/home/slavko/genode.test/depot UPDATE_VERSIONS=1 FORCE=1 REBUILD= CCACHE=1  
make[1]: Entering directory '/home/slavko/genode.test/build/x86_64'
/usr/bin/make -f /home/slavko/genode.test/tool/depot/extract genodelabs/bin/x86_64/base-linux genodelabs/bin/x86_64/demo genodelabs/bin/x86_64/global_keys_handler genodelabs/bin/x86_64/init genodelabs/bin/x86_64/nit_focus genodelabs/bin/x86_64/nitpicker genodelabs/bin/x86_64/report_rom genodelabs/bin/x86_64/rom_filter genodelabs/pkg/x86_64/drivers_interactive-linux \
                     VERBOSE= FORCE=1 \
                     UPDATE_VERSIONS=1 \

make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
make[2]: bash: No such file or directory
Error: incomplete or missing recipe (genodelabs/api/base genodelabs/api/base-linux genodelabs/api/blit genodelabs/api/capture_session genodelabs/api/event_session genodelabs/api/framebuffer_session genodelabs/api/gui_session genodelabs/api/input_session genodelabs/api/nitpicker_gfx genodelabs/api/os genodelabs/api/report_session genodelabs/api/sandbox genodelabs/api/timer_session genodelabs/pkg/drivers_interactive-linux genodelabs/raw/drivers_interactive-linux genodelabs/src/base-linux genodelabs/src/demo genodelabs/src/event_filter genodelabs/src/fb_sdl genodelabs/src/global_keys_handler genodelabs/src/init genodelabs/src/nit_focus genodelabs/src/nitpicker genodelabs/src/report_rom genodelabs/src/rom_filter)
make[2]: *** [/home/slavko/genode.test/tool/depot/mk/extract_post_dependencies.inc:29: checked_versions_defined] Error 1
make[1]: *** [/home/slavko/genode.test/tool/depot/create:41: extract] Error 2
make[1]: Leaving directory '/home/slavko/genode.test/build/x86_64'
child process exited abnormally
    while executing
"exec {*}$cmd >@ stdout 2>@ stderr"
    (procedure "_depot_auto_update" line 39)
    invoked from within
"_depot_auto_update $archives "
    (procedure "_collect_from_depot" line 11)
    invoked from within
"_collect_from_depot [join $args " "]"
    (procedure "import_from_depot" line 3)
    invoked from within
"import_from_depot [depot_user]/src/[base_src] \
                  [depot_user]/pkg/[drivers_interactive_pkg] \
                  [depot_user]/src/rom_..."
    (file "/home/slavko/genode.test/repos/os/run/demo.run" line 10)
    invoked from within
"source $include_name"
    ("foreach" body line 6)
    invoked from within
"foreach include_name [get_cmd_arg --include ""] {
	# first check if the include name is absolute
	if {[string first "/" $include_name] == 0} {
		puts ..."
    (file "/home/slavko/genode.test/tool/run/run" line 1165)
make: *** [Makefile:431: run/demo] Error 1
```

chelmuth, 2023-01-16:
Could this be an issue with your Linux installation like in this [stackoverflow issue](https://stackoverflow.com/questions/35071872/bash-no-such-file-or-directory)?

As we do not publish the base-linux package, I wonder how you managed to get it. What happens if you run the following command (as proposed when running `make run/demo` without `--depot-auto-update`)?

```
./tool/depot/create genodelabs/pkg/x86_64/drivers_interactive-linux \
 genodelabs/bin/x86_64/demo genodelabs/bin/x86_64/global_keys_handler \
 genodelabs/bin/x86_64/nit_focus genodelabs/bin/x86_64/nitpicker genodelabs/bin/x86_64/rom_filter
```

rite, 2023-01-16:
I had the same issue after I updated my Arch Linux a week or so ago... It looks like it is related to the GNU Make version 4.4, which was released by the end of October 2022 [1]. @dogmode What is the output of `make --version`?

[1] https://lists.gnu.org/archive/html/info-gnu/2022-10/msg00008.html

For my purposes, I solved the problem with the following two commits:

- [b55c404](https://github.com/rite/genode/commit/b55c40406e0842daa186465629ac77c5bd1d39dc) makes sure the global environment variable `PATH` is not overwritten by make 4.4. This solves the issue that `bash` (or other binaries) is not found. But there seems to be yet another incompatibility I wasn't able to solve (or even identify, for that matter...). Therefore I added:
 
- [3a2e893](https://github.com/rite/genode/commit/3a2e893ad318149a0aff2dd98809370439893364) which allows me to use `nix-shell` and pin the toolchain to a certain version. I now have a file `shell.nix` and I enter `nix-shell` when I want to build Genode and I doesn't have to deal with toolchain incompatibilities and fear Arch Linux updates anymore :)

ssumpf, 2023-01-16:
@rite: I can confirm that the issue is introduced by make 4.4 and does not happen in 4.3. Thanks!

ssumpf, 2023-01-17:
We still need to fix this within Genode's build system.

ssumpf, 2023-01-17:
GNU Make 4.4 also breaks parallel depot builds for API and PKG targets. 099cb15 fixes this.

nfeske, 2023-01-17:
Thanks a lot @ssumpf! I just merged your commit to staging.

nfeske, 2023-01-17:
@rite thank you very much for fix regarding the PATH variable name, which I just merged to staging. Well, this special treatment of the variable name was a surprise to me.

ssumpf, 2023-01-17:
@delay-lama: Thanks for reporting, since not many of us run on an up to date Linux distro any more.

ssumpf, 2023-01-18:
When building Qt5/Mesa with normal `make` calls I observed errors like:
`/bin/bash: line 1: echo: write error: Broken pipe`
with Make 4.4. This also needs to be addressed.

ssumpf, 2023-01-19:
32ea660 fixes the `Broken pipe` error.

nfeske, 2023-01-19:
Thanks @ssumpf, merged to staging.

nfeske, 2023-01-20:
With https://github.com/genodelabs/genode/commit/b5cb13350771ff7fe270971ed34a76f483463599, the preceding comment no longer applies. Fixup https://github.com/genodelabs/genode/commit/5950392af156bef4c7525e88baa68638efbefb77 removes the comment.

closed 2023-05-30 by ssumpf

ssumpf, 2023-06-12:
5d48c1a resolves an issue I had when running `tool_chain_qt5` with GNU Make 4.4. The `shell` that spawned `tool/ports/current` returned make's operation like `Entering directory` or `Leaving directory`, leading to multiple targets because of `CONTRIB_DIR`. This does not happen with Make 4.3.

nfeske, 2023-06-12:
Thanks @ssumpf. Merged https://github.com/genodelabs/genode/commit/5d48c1acc0fc5aa83be82a3c627254c973d43a55 to staging.