genodelabs/genode #4728: nic_router: fix updating UDP/TCP links

m-stein, 2023-01-17:
There is a bug in the updating of UDP/TCP links on reconfiguration that was introduced by this earlier commit:

0f6714c6d7 nic_router: find forward rules w/o exceptions

The NIC router falsely dissolves TCP/UDP connection states while re-configuring although the connection states are still legal according to the new config. The reason is that the above mention commit introduced nested lambdas but missed to return from the top-most nesting level when having found a configuration that legitimates the connection state.

m-stein, 2023-01-17:
Fixed:
a7b1f7b639 nic_router: simplify updating of TCP/UDP links
223e8a810f nic_router: fix updating UDP/TCP links

chelmuth, 2023-01-18:
Merged fixes to staging.

m-stein, 2023-01-18:
@chelmuth After careful re-thinking of the matter I had to add a fixup commit:
9cf4c20370 Squash "nic_router: simplify updating of TCP/UDP links"

m-stein, 2023-01-18:
Again without hashes:
df7b4e4109 Squash "nic_router: simplify updating of TCP/UDP links"

commit a8449160 (2023-01-18): nic_router: simplify updating of TCP/UDP links

* Don't use good-case exceptions (Dismiss_link) for updating TCP/UDP links on
  re-configuration of the router.

* Make conditions when to dismiss a forward rule easier to read.
  * Introduces != operator to the public Port class in the net library.

* Fix unnecessary log message that a link was dismissed when only a potentially
  matching forward rule turned out to be not matching.
* Apply Genode coding style to if statements with a single body statement.

Fix #4728
commit 2c55ce09 (2023-01-19): Squash "nic_router: simplify updating of TCP/UDP links"

Please rewrite commit message:

nic_router: rework updating of TCP/UDP links

* Update links from forward rules only with forward rules and links from
  transport-routing rules only with transport-routing rules. Besides raising
  the performance of the code, this also fixes a former bug that allowed
  forward-rule links to falsely stay active because of a transport-routing
  rule that matched the client destination ip and port.

* Don't use good-case exceptions for updating TCP/UDP links on re-configuration
  of the router.

* Make condi [... cut]
closed 2023-01-24 by chelmuth