genodelabs/genode #4730: run: unify role of var/run/<scenario>/boot/ directory between ARM and x86

nfeske, 2023-01-18:
The file structure of the platform-specific `var/run/<scenario>/` directories was chosen in a pragmatic way that used to considered only the target platform and boot strategy at hand.

On NOVA/x86 it the boot/ directory contains the GRUB boot-loader config + the hypervisor + the bender pre-boot loader + the banner image + the Genode system image. This structure corresponds 1:1 to the boot/ directory as found on the 3rd partition of the Sculpt system, which is very nice. A system update comes down to replacing these files.

I'd like to handle the boot image for the PinePhone in a similar manner, hosting the uImage in the boot/ directory at the root of the booted SD-card. Following the pattern of NOVA, one would expect to find the uImage at `var/run/<scenario>/boot/uImage`. However, currently, the boot directory contains an image.elf file whereas the uImage resides at `var/run/<scenario>/uImage`.

To foster consistency, I'd like tweak the run tool as follows:

- Create the uImage under boot/ when using the image/uboot plugin
- Remove the boot/image.elf file after the conversion step to the uImage
- Change the `load_tftp_inst_cmd` argument (given by the boot_dir plugins) to point to boot/uImage instead of uImage
- Change the board-specific U-Boot configurations that boot from SD-card to fetch the uImage from the boot/ subdirectory

As an extra nicety, U-Boot could fetch its environment config from /boot/uboot.env instead of /uboot.env, which would keep the root of the SD-card more tidy. Following the pattern of the NOVA version, the U-Boot environment intuitively belongs to boot/ as well.

nfeske, 2023-01-18:
I just noticed that the image.elf is passed as argument to `run_image`, which generates the uImage file. Having this function remove the image.elf as side effect would be quite unexpected. Hence, I'd host the image.elf file at `var/run/<scenario>/` (besides the boot/) directory and keep it.

nfeske, 2023-01-18:
Upon closer inspection - looking at run/boot_dir/hw where the image.elf file vanishes in the process on PCs - we can actually change `run_image` to dispose the `boot/image.elf`file. However, instead of passing this file as argument, we would consistently apply the convention that an boot/image.elf file has to exist when the function is called. This makes sense because only run/image/uboot evaluates this argument anyway, and the argument is always `boot/image.elf`. With this change, the official semantics of `run_image` would be: "replace the boot/image.elf file by platform-specific file(s) at boot/ that can actually be booted".

The name `run_image` is quite misleading because "run_" is just a prefix, not the action performed by the function. Are there any ideas for a better name?

nfeske, 2023-01-18:
Commit https://github.com/genodelabs/genode/commit/86265d40f2cd21918412dc351359af0d3a0868f1 removes the argument of the `run_image` function.

nfeske, 2023-01-18:
Commit https://github.com/nfeske/genode/commit/2559b7bf453484b2ddb176fc16da89023ee6182d moves the uImage file to boot/, removes the image.elf file, and adjusts the TFTP paths accordingly.

jschlatow, 2023-01-19:
> @jschlatow would that line of reasoning apply to image.itb files as well?

Yes, it also applies to the image.itb.

nfeske, 2023-01-19:
Thanks @jschlatow for the confirmation. My take at https://github.com/genodelabs/genode/commit/54469bc7d8bdc43012dfe0f3d256cd66857caf8c

nfeske, 2023-01-19:
Commits https://github.com/genodelabs/genode-allwinner/commit/c2856b30887e3a0d298158b7aa339c459170be49 and https://github.com/genodelabs/genode-allwinner/commit/ab9c59d619684e6d569560e32957006ee323f462 wrap up the issue for the PinePhone.

jschlatow, 2023-01-19:
Commit [1764372](https://github.com/genodelabs/genode-zynq/commit/1764372c56579e99de78ffd21fbdd823ef476020) applies the necessary adaptations for the Zynq-SoC.

nfeske, 2023-01-19:
Thanks a lot @jschlatow!

nfeske, 2023-01-20:
Commit https://github.com/genodelabs/genode/commit/b258d09d0a25f5b81f26a96efd66f0619cf5bc7e should make tftp boot work again.

nfeske, 2023-01-24:
Fixed in master.

closed 2023-01-24 by nfeske