genodelabs/genode #4736: sculpt: expansion of ext2 partition is unreasonably slow

nfeske, 2023-01-20:
The official system images of the PinePhone will contain only the bare-bones system on a tiny partition (16 MiB). As the second step after copying the image to an SD-card and booting into the system, the user will be required to expand the partition to the actual size of the user's SD-card. The Sculpt system supports this via the "Expand..." function that appears when selecting a partition in the storage dialog.

However, currently, the expansion takes an enormous amount of time. For example, when expanding a 32 GiB Sandisk Extreme SD-card on the PinePhone, this operation takes almost 5 minutes! Since this is one of the first operations every single user is expected to do, this excessive waiting time may spoil the first impression. It is also intuitively not plausible.

@cnuke could you please analyze and hopefully improve this situation? A first low-effort idea would be to look at possible options of the resize2fs program. But it can't hurt to get hold of the actual sequence of block operations, and check them for plausibility (e.g., is the trace dominated by intermediate sync requests?). Or may the problem be related to some kind of timeout/timing issue, a code path that could be avoided? Or is the performance severely impeded by an I/O buffer size?

I think this issue must be solved before publishing the first system image for the PinePhone. Or, if it cannot be solved, we should have a solid explanation why that is.

cnuke, 2023-01-20:
@nfeske  I'll take a look at it.

nfeske, 2023-01-25:
Commit https://github.com/genodelabs/genode-allwinner/commit/a193d30366d546c44052a71c5d27c6767bcf0493 fixes this issue for the PinePhone. Thanks to @chelmuth for the aid.

chelmuth, 2023-01-25:
I used the following Makefile to examine which properties mkfs.ext2 generates. In my opinion, the most expensive property is the different block size, which compels the resize to act down to the block level.

```make
FS_TYPES := allwinner small default big
PROPS    := $(FS_TYPES:=.props)

PARAMS = -q -F -U clear -t ext2 -L $*

PARAMS(allwinner) := -N 0 -m 5 -r 1
PARAMS(small)     := -T small
PARAMS(default)   := -T default
PARAMS(big)       := -T big

default: $(PROPS)

generate_img = dd if=/dev/urandom of=$@ bs=1M count=32

%.img: $(MAKEFILE_LIST)
	$(generate_img)
	mkfs.ext2 $(PARAMS) $(PARAMS($*)) $@

%.props: %.img
	tune2fs -l $< > $@

clean:
	rm -f $(PROPS)
```

cnuke, 2023-01-26:
> I used the following Makefile to examine which properties mkfs.ext2 generates. In my opinion, the most expensive property is the different block size, which compels the resize to act down to the block level.

These are my findings as well and were confirmed by testing the various presets on the PinePhone.

chelmuth, 2023-02-08:
Issue was fixed in genode/genode-allwinner repo.

closed 2023-02-08 by chelmuth