genodelabs/genode #4738: nic_router: no denied sessions with dynamic config

m-stein, 2023-01-23:
A user reported this bug and provided a corresponding patch that I merely adapted slightly.

The NIC router used to announce its services directly on startup. This has a negative side effect in scenarios where the NIC-router configuration is dynamically generated and therefore potentially empty at the beginning. If a session request of a client reaches the NIC router before the configuration has turned non-empty and valid, the request is denied. It is common among network client components such as NIC drivers to treat this as a fatal error and terminate. This can be prevented by deferring the announcement of the NIC router services to the point in time where the NIC router receives its first valid configuration.

chelmuth, 2023-01-23:
Could you please give more information about the error case? What I do not understand is why a session request is denied on empty config but not with a config that just misses a matching policy node. I'd expect the NIC router to behave similar in both cases, which means the session is *not* denied.

m-stein, 2023-01-23:
After the comment by @chelmuth, I tried to reproduce the described issue myself and found that the router actually doesn't deny sessions with an invalid configuration. I a successive mail discussion with the user, we found that the issue didn't come from the initially invalid configuration but an unofficial router quickfix.

closed 2023-01-23 by m-stein